Edzelf/ESP32-Radio #170: Some oddities using a  2.8"TFT

JeanRenier, 2018-08-24:
Hi,

Let me first thank you for sharing this really amazing piece of software engineering.  

I made a little PCB combining the ESP32 VROOM, the VS1053 module, a 2.8"TFT module based on the ILI9341 chip and a few other bits to make the whole work.  It seems to work quite all right.  
I modified your software a very little bit (starting from the August 6 release) in order to accommodate the XTP2046 touch screen hardware, just a few simple things for now: setting the volume and up/down preset.   I also enlarged the text size and the bargraphs for the volume and the battery charge, to improve the visibility. 

A few oddities were noted, all of them have to do with the ILI9341 (or so I believe):
- whenever a file is selected from the SD card the TFT screen becomes blank (and stays also even if an Internet station is selected afterwards) until a hard reset, note that the back light is still working.  
- the right part of the bargraph for the battery charge (the one that is written last I think) seems to be gradually (in less than 1s) overwritten by random pixels, as if the TFT would not haven been deselected.  
Temporarily eliminating the touch screen lines in the code do no seem to change anything.  Any thoughts on this to help me further would be greatly appreciated.

Many thanks and all the best,  Jean Renier.

Edzelf, 2018-08-24:
It looks like the display is not deselected correctly.  This problem is reported more than once.  I was not able to reproduce it however.  Sometimes it helps if CS of the display was connected to a different GPIO pin.

JeanRenier, 2018-08-29:
Hi, 
Back home now and having had the opportunity to put an oscilloscope probe to the tft_cs_pin (GPIO15).  
This issue is very much the same as reported in issue 100.  The tft_cs_pin is driven low for very much all the time, leaving the TFT wide open.  Every second the TOD, battery level, ... updates will end by drive the pin high as it should, but after a short while (anything from 1ms ...200ms) the pin will be driven low again without apparent reason.  The phenomenon starts about immediately after the message with the IP address is displayed, i.e. there is no problems until an IP connection exists.  This points indeed to some obscure issue in Espressif's WiFi / IP handling software (ancient remains of a debugging aid ?).  
Contrary as was reported, upgrading to the newest version of the ESP32 board manager (1.0.0) in the Arduino IDE didn't solve the problem for me.  I have included a write rectangle routine of just one pixel in a corner, this seems to eliminate at least the visual effect of this issue.  I may change the assignment of the tft_cs_pin in future, but this is not an easy thing on a PCB.
Since this issue has nothing to do with the ESP32-Radio software I propose to close it.
Many thanks all the best,  JeanRenier

closed 2018-08-29 by JeanRenier