Edzelf/ESP32-Radio #173: Strange 6 µs spikes on GPIO 26

fletsche, 2018-09-19:
I made my own PCB and changed the GPIO 26 from rotary encoder CLK to rotary encoder sw resulting in stuttering of the playback. I noticed that this is because of strange 6 µs negative spikes with a frequency of 10 Hz (see attached oscilloscope image, yellow=GPIO26=CLK, blue=GPIO27=DT). I went back to my breadboard where GPIO 26 was still CLK and found the same spikes there, although not causing any symptoms as turning of the rotary encoder gets detected correctly nonetheless (see second image).
Interestingly on boot up these spikes aren't present. They start with around the end of the boot up process and stay even when playback is stopped. They also appear when I disconnect the VS1053. When I change stations there is always one "lost" spike.
Could you check whether you also get these spikes? I thought they probably come from the 100 ms ISR routine, but changing that to 110 ms didn't affect the frequency of the spikes. Do you have any idea where these might come from?

[image]
[image]

Edzelf, 2018-09-19:
I do see spikes of 6 microseconds on DT and CLK, but the amplitude is only 1 volt in my case.  With a 1000 pF capacitor to ground, I could suppress the spikes.

Edzelf, 2019-01-18:
Debounce is not necessary for the signals that are coming from the decoder.  Other signals can surely have strange effects.

SimonMerrett, 2019-01-18:
> Debounce is not necessary for the signals that are coming from the decoder. Other signals can surely have strange effects.

And I wasted my own time by assuming the two were the same thing!