Edzelf/ESP32-Radio #174: How to connect the encoder

student1704, 2018-09-21:
is there schematic available how to connect the encoder?
I don't own the proposed modul.
Any help is appreciated

Rainer-G, 2018-09-23:
Do you have a specification for your encoder model?
It would help to know the connections and their function on your model.

Normally, you will have a data (DT) and a Clock (CLK) pin for the encoder function and a switch (SW) for the pushbutton and a ground (GND) connection. 
If the model has pullup resistors, a VCC connector will also be provided. 
The documentation ( both in the code and in a file ) shows which pins to use on the ESP32 but you are free to use different pins. You only need to change the preferences for it in the WEB Interface:

[image]

student1704, 2018-09-24:
Thanks Rainer,
I just have a single ALPS component not soldered on a pcb. I thought there is something special with the modul.  I have to check the wiring.
BR

closed 2018-09-24 by student1704