Edzelf/ESP32-Radio #175: memory volume and station number

avulyanichev, 2018-09-29:
Ed. Tell me where to turn off the recording in memory, volume and station numbers as in EspRadio?  in the afternoon I listen loudly, and when I turn on in the evening I want to make the sound as in the settings

Edzelf, 2018-09-29:
Look for "handleSaveReq()".

avulyanichev, 2018-09-30:
Thanks it turned out.
savetime = millis() ;                                   // Set time of last save
//  nvssetstr ( "preset", String ( currentpreset )  ) ;     // Save current preset
//  nvssetstr ( "volume", String ( ini_block.reqvol ) );    // Save current volue**
  nvssetstr ( "toneha", String ( ini_block.rtone[0] ) ) ; // Save current toneha
  nvssetstr ( "tonehf", String ( ini_block.rtone[1] ) ) ; // Save current tonehf
  nvssetstr ( "tonela", String ( ini_block.rtone[2] ) ) ; // Save current tonela
  nvssetstr ( "tonelf", String ( ini_block.rtone[3] ) ) ; // Save current tonelf
}

With IR the problem has remained.

if ( ( intval > 300 ) && ( intval < 800 ) ) // Short pulse?
и
else if ( ( intval > 1400 ) && ( intval < 1900 ) ) // Long pulse?

I cheated on mine, but it didn't get any better. remembers the last press and when you press another button pulls the last of the memory.