Edzelf/ESP32-Radio #179: OLED Screen... some info is missing on the screen

gpeter62, 2018-10-09:
It is a fantastic project, everythink works fine! Thank you!

But on the OLED screen there is no information about the Mute or Stop state.  And after switching OFF and ON, the MUTE is remembered. It is a problem for simple users.

The volume bar is not visible after station change.

letarg0, 2019-01-18:
is possible to run out of box Wemos® TTGO TS V1.0 esp32 1.44 TFT MicroSD Card Slot Speakers Bluetooth Wifi Module
for example https://www.banggood.com/Wemos-TTGO-TS-V1_0-esp32-1_44-TFT-MicroSD-Card-Slot-Speakers-Bluetooth-Wifi-Module-p-1273383.html

?

Edzelf, 2019-01-18:
No, the module has no MP3 decoder.

Edzelf, 2019-01-24:
A lovely search engine exists.  It is called "google".

letarg0, 2019-01-24:
i searched but not found

Edzelf, 2019-01-24:
Search for Lenco DIR-150

letarg0, 2019-01-27:
I need hackable esp32 working radio. I'm programmer. I would like to create my own software but I'm not electroic

Edzelf, 2019-01-27:
Use Harms print.  See issue 198.

letarg0, 2019-01-27:
What is **Harms print** ?
any link? please tell me where i can bu working device

Edzelf, 2019-01-28:
Read the document, page 10.