Edzelf/ESP32-Radio #180: Enkoder logic.

Victorbins, 2018-10-10:
Hello Ed.
Your project is amazing. I make it with encoder and LCD1602 and it work fine. Only one thing.
I want to use  preset mode by default, not volume mode. How to do it?

Victorbins, 2018-10-11:
Yes, I already did this, but afte i chenge VOLUME mode and PRESET mode LCD screen was not work properly. It becames blank after 4 seconds past change preset.

Edzelf, 2018-10-11:
Yeah, programming is sometimes a struggle...

Victorbins, 2018-10-12:
May be you may help me by advice?  ))

Edzelf, 2018-10-12:
The 4 seconds is the time that is used to see if there is no activity on the encoder.  It switches to volume mode.  You have to change this to preset mode.

Victorbins, 2018-10-13:
I have changed it to preset mode and appears LED screen problem, it becomes blank.

Edzelf, 2018-10-13:
Note that in 1602.h there is also a reference to "enc_menu_mode" (line 393).  It will display different lines depending on its state.

Rainer-G, 2018-10-17:
I'm working on a similar problem, my board has a potentiometer with switch to control volume and turn on/off the radio. The encoder only should be used to change from Radio to mp3 and vice versa. 
Other than above, I have one board with bluetft and a second one with a Waveshare Touch display ( no encoder, touch pads simulate the encoder activities ).
I found that it was not very easy to change the standard logic but Ed's comments helped to make it work, many thanks to Ed!
If have added a file : [volume_default.txt](https://github.com/Edzelf/ESP32-Radio/files/2486174/volume_default.txt) with some code sniplets to log the changes. I used the most recent files from GitHub.

It is working with both displays, the only issue is a short diplay refresh if there is a single click without any station/track change as a result of mute/unmute not working any more.

The 1602 display should work as well as tftset takes care of changing the text and 1602.h changes the text to the station/track text if mode is not VOLUME so no code change is needed ??

Good luck, Rainer

Victorbins, 2018-10-24:
Thanks Rainer.