Edzelf/ESP32-Radio #183: next station dosn't work

student1704, 2018-11-10:


student1704, 2018-11-11:
here it is!
faulty logs with encoder and web keys and a good one created with drop down list.
[debugLog20181111.txt](https://github.com/Edzelf/ESP32-Radio/files/2569215/debugLog20181111.txt)

Edzelf, 2018-11-11:
No, I want to see the logs during startup.

student1704, 2018-11-11:
here it is!
I detected: Pin 35 is used for IR (which is not connected) and ENC. But that should not be a problem , or?
BR

[StartupLog20181111.txt](https://github.com/Edzelf/ESP32-Radio/files/2569278/StartupLog20181111.txt)

Edzelf, 2018-11-11:
Yes, thera are conflicting settings in your preferences for pin 35.  Remove the setting for IR.

student1704, 2018-11-13:
I did. 
But without success.

fletsche, 2019-01-28:
I am having the same problem. The reason is that your (and my) stations are provided as m3u playlists. As I understand the code it tries to play the next track from that m3u playlist (but there is no next track) instead of changing to the next preset. You could add `playlist_num = 0 ;` after `ini_block.newpreset += ivalue ;`
However if there would be a playlist with multiple tracks you wouldn't be able to switch between those.

p1mike, 2019-09-20:
Same for me. The issue comes up with *.m3u stations only. If you select a *m3u station you can change station with "Preset:" list in Web-interface only. :-(
Most of my favorite stations are listed as *.m3u. You can reproduce it for example with:
mp3-live.swr.de/swr1bw_m.m3u

p1mike, 2019-09-26:
> Try "swr-swr1-bw.cast.addradio.de/swr/swr1/bw/mp3/128/stream.mp3" instead.

Thank you for this hint. For now I'm already using the content of the m3u file or the resolved station like "http://swr-edge-2023.dus-lg.cdn.addradio.net/swr/swr1/bw/mp3/128/stream.mp3". But in the past, they have changed the link and I got silence some days later. :-(

If it's to difficult to take a one-song-list.m3u list as a station instead of playlist so it's ok for me.

Thank you for this great project and also for your fast answer

breti, 2021-08-29:
Unfortunately this is still an issue when using m3u files. :-(