Edzelf/ESP32-Radio #189: Output enable not working as expected

fletsche, 2018-11-26:
I did some testing with the shutdown/shutdownx pins and found out that they do not work as expected. It seems that there are at least two issues:
1.) In VS1053::startSong output was enabled regardless of the current volume (I fixed this in pull request #188).

2.) When issuing the "stop" command the shutdown pin is activated but only for about one second, then it somehow gets deactivated and I could not yet figure out where or why. I hope Edzelf can help with that one!

Pavulon87, 2018-12-20:
I see this problem too. Its because handle_spec() sets volume immediately / infinitely.
`vs1053player->setVolume ( ini_block.reqvol ) ;            // Unmute`
Quick fix. In VS1053::setVolume I added:
`	if ( vol && ( datamode & ( STOPREQD | STOPPED ) ) )
	{
		return;
	}`

fletsche, 2019-01-02:
I tried your quick fix. It has the side effect that the volume cannot be changed, when playback is stopped. I took a deeper look into the volume controll at it seems to me, that muting is a bit of a hack anyway. I think for a real solution it would be necessary to strictly differentiate between desired volume, mute state and amplifier shutdown.
Shutdown is actually a little more complex as it depends on the recovery of the amp from shutdown. I. e. the PAM8403 has separate mute and shutdown pins. While both pins react instantaneous to mute the output at least in my experience it took a few seconds after enabling the amp via the shutdown pin until output was audible.