Edzelf/ESP32-Radio #190: SLOW SPI

paysan18, 2018-11-26:
I'm building this very promissing Internet radio.
The ESP32 is a ESP32-DEVKITV1 with 30 pins.
And I use the VS1053B.

During the startup I  got the message SLOW SPI and VS1053 not properly installed.
I double checked the wiring and all looks ok.

Who has an idea how to fix this.
[debug.txt](https://github.com/Edzelf/ESP32-Radio/files/2616682/debug.txt)

[image]

Edzelf, 2018-11-26:
There is no setting for pin_vs_cs, pin_vs_dcs and pin_vs_dreq.  Did you run the Esp-radio-int sketch?

Edzelf, 2018-11-26:
Start the webinterface and enter the proper settings for the VS1053 pins.

paysan18, 2018-11-27:
I did, see my next issue #191. 
Sofar no luck....

paysan18, 2018-11-30:
This issue can be closed. It was 1 typo in the Esp-radio-int sketch.

closed 2018-11-30 by paysan18