Edzelf/ESP32-Radio #191: Esp32_radio_init compile error

paysan18, 2018-11-26:
Running Linux-Mint and Arduino IDE 1.8.5
The preferences.txt file is located in :   ~/.arduino15/

When I try to compile the Esp32_radio_init.ino file I got this error

[image]

this file is attached (renamed from .ino to .txt)
[Esp32_radio_init.txt](https://github.com/Edzelf/ESP32-Radio/files/2616714/Esp32_radio_init.txt)

Edzelf, 2018-11-26:
Just remove the variable "Str".  It is not used and you are treating warnings as an error.  This wiil probable cause more trouble....

paysan18, 2018-11-27:
I looked for a variable named "Str"
Nothing found.  Just an object declaration "String  str"
Anyway for some reason I suddenly was able to compile and flash the init file.

But still no luck.

I went into the sketch Esp32_radio to line 2746 and replaced the -1 values with 22, 5 and 32.
See attachement. But still no luck.
[trace.txt](https://github.com/Edzelf/ESP32-Radio/files/2619955/trace.txt)

[image]

Please advice........

Edzelf, 2018-11-27:
Your trace looks good.  There are now no SPI errors.  You should have sound on your vs1053 output.
What is the problem?

paysan18, 2018-11-27:
Sometimes I see the SPI error again during reboot.
And the sound is very “hakkelig”. I can not recognize what it is. Music of voice.
During playing with the web interface the system hangs. I placed some capacitors at the power lines.
But the problem remains.... i think this will be extreemly  difficult to fix.....

Edzelf, 2018-11-27:
The web interface should work even if you run it on a naked ESP32 (no wires attached).  If it does not run, there is something wrong with you ESP32 or your power supply.

paysan18, 2018-11-28:
The ESP32 runs fine in all disconnected mode. I think there is a problem with my VS1053.
I tried it by connecting it to a Arduino UNO with the Adafruit setup. It sees the board but it dus not produce any sound. I ordered a new one and will update this issue afterward....

Edzelf, 2018-11-29:
Are you sure that your VS1053 board contains a VS1053 chip.  Some Chinese boards have a VS1051 chip on it...

paysan18, 2018-11-29:
Yes, the chip on my board is a VS1053B.

paysan18, 2018-11-29:
It's almost working now. Powersupply (over USB) was 4 volt!
But the output sound is very intermittend.
5 seconds sound 3 seconds no sound (random timing).
I will first have a look at the debug trace.

paysan18, 2018-11-29:
Every time the sound stops I see Single Click in the trace.
Looks like the pin of the rotary push button is a bit noisy or floating.
I will check in the sketch if the pullup has been enabled.

Edzelf, 2018-11-29:
Yes, single click has a mute function.  You can remove the encoder from the preferences to switch it off temporarily.

paysan18, 2018-11-29:
After a restart I did not see the SingleClicks any more and the sound goes on...
What I see with the scope on the push button pin is a negative going puls of 10uS and goes from 3.3 to 1.8 volt. So not all the way down to zero.  If fluctuates in amplitude. Maybe at this moment it stays above the threshold of logic zero. Do you think an extra pullup resistor of 10K will be better?

Here another problem. Maybe I should open a new issue?
After selecting the MP3 Player I got this:

[image]

Rainer-G, 2018-11-29:
There was a similar issue ( 173 ) with GPIO26.
 I also found these on GPIO13 and GPI36 in a different setup with the Adafruit HUZZA32 board.
Looks as if there is some cross-talk from other pins. Pull-Up resistors with smaller values might help but may also cause other problems.

Good luck,
Rainer 

Von: Willem Nijntjes
Gesendet: Donnerstag, 29. November 2018 19:29
An: Edzelf/ESP32-Radio
Cc: Subscribed
Betreff: Re: [Edzelf/ESP32-Radio] Esp32_radio_init compile error (#191)

After a restart I did not see the SingleClicks any more and the sound goes on...
What I see with the scope on the push button pin is a negative going puls of 10uS and goes from 3.3 to 1.8 volt. So not all the way down to zero. If fluctuates in amplitude. Maybe at this moment it stays above the threshold of logic zero. Do you think an extra pullup resistor of 10K will be better?
Here another problem. Maybe I should open a new issue?
After selecting the MP3 Player I got this:

paysan18, 2018-11-30:
question to Ed:
What does the trace out say: SPI semaphore not taken within ..... when I start the mp3 player?

Edzelf, 2018-11-30:
It means that the SPI bus is very busy.  Probably caused by reading many files from SD card.  Just remove the SD during testing, it saves a lot of time.