Edzelf/ESP32-Radio #192: Rotary encoder question.

paysan18, 2018-11-30:
My rotary encoder does not work. However the push button works.
I can see the action of the push button in the debug serial output.
But when I rotate the encoder nothing comes out in the trace.

I tried a little standalone prog and all looks ok with this encoder. Same pins used.
By looking at the code I have some questions.

1. Pins are not defined in the Esp-radio-init sketch. Why not?

2. If I press the config button I see  them configured as pin 25, 26 and 27. So this looks good.
    Where does this come from? From the progpin_struct?

3. What do the different fields mean in the struct progpin_struct?

Rainer-G, 2018-11-30:
Do you still have the interrupts occuring without moving the encoder ? 
I had the same problem and found that if the interrupts come very fast, a rotation will not be detected.
I put this :  
      tftset( 3, dbgprint( "Timer %5d %3d", enc_inactivity, rotationcount  )) ;      

at the start of chk_enc for a test and found that enc_inactivity never was above 3 or 4 so it was reset by the interrupt routine                  

Von: Willem Nijntjes
Gesendet: Freitag, 30. November 2018 11:57
An: Edzelf/ESP32-Radio
Cc: Subscribed
Betreff: [Edzelf/ESP32-Radio] Rotary encoder question. (#192)

My rotary encoder does not work. However the push button works.
I can see the action of the push button in the debug serial output.
But when I rotate the encoder nothing comes out in the trace.
I tried a little standalone prog and all looks ok with this encoder. Same pins used.
By looking at the code I have some questions.
1. Pins are not defined in the Esp-radio-init sketch. Why not?
2. If I press the config button I see them configured as pin 25, 26 and 27. So this looks good.
Where does this come from? From the progpin_struct?
3. What do the different fields mean in the struct progpin_struct?

paysan18, 2018-11-30:
Rainer, the interrupts did not come from the rotary encoder pins but from the pushbutton pin. I solderred a capacitor to GND on this pin and all works fine now.
But the rotary is still dead.
Wim

Rainer-G, 2018-11-30:
Wim,
for my information :
when you do a double or triple click and then no further action, will the display return to the text it was before the clicks ?
If it does, no unexpected interrupts ocurr on enc data and clock.
Rainer

Von: Willem Nijntjes
Gesendet: Freitag, 30. November 2018 12:45
An: Edzelf/ESP32-Radio
Cc: Rainer-G; Comment
Betreff: Re: [Edzelf/ESP32-Radio] Rotary encoder question. (#192)

Rainer, the interrupts did not come from the rotary encoder pins but from the pushbutton pin. I solderred a capacitor to GND on this pin and all works fine now.
But the rotary is still dead.
Wim

paysan18, 2018-11-30:
Rainer, I have no display attached. But the messages came in the serial debug trace without I touched anything. But like I told you, just from the switch.
The rotary is still dead. 
During the bootup trace I see the line Rotary encoder enabled.

Rainer-G, 2018-11-30:
Hello Wim,

the trace show a single click even on very short clicks, that’s you can see random interrupts without any action. 
Rotary counts are only shown at stable situations ( debouncing, look at the interrupt routine ) so spurious interrupts may not be seen.

Just for a test, try  

      dbgprint( "Timer %5d ", enc_inactivity  ) ;  in chk_enc, this works without display using the serial monitor.
      
If the counter goes above 40, there are definitely no interrupts on data and clock!
Or, if not done so far, add capacitors similar to the enc sw!

Cheers, Rainer

Von: Willem Nijntjes
Gesendet: Freitag, 30. November 2018 13:54
An: Edzelf/ESP32-Radio
Cc: Rainer-G; Comment
Betreff: Re: [Edzelf/ESP32-Radio] Rotary encoder question. (#192)

Rainer, I have no display attached. But the messages came in the serial debug trace without I touched anything. But like I told you, just from the switch.
The rotary is still dead.
During the bootup trace I see the line Rotary encoder enabled.

paysan18, 2018-11-30:
Rainer, I placed the line just after the curly bracket "{" in the chk_enc.
Now when I watch the debug output I see the counter increasing. When I turn the knob the counter resets and starts from zero. This means at least that the rotary is working. But is your line at the right place?

paysan18, 2018-11-30:
> Hi Wim,
>
> the place is ok! It also can be somewhere in the main loop, it only needs to processed each loop.
>
> If the counter goes above 40 without any activity, it looks ok.
> It should go to zero only when you turn the knob or press the button.
> But if you still do not see any rotationcount in the trace and the encoder works in a different program I cannot see a reason for this.
> In the trace you sent Nov. 27th, there were numerous -1 in a row, is this sill the case ?
>
> Rainer
>
>
> Gesendet von Mail für Windows 10

It’s strange that the counter reacts to the rotation of the encoder but the program does not.
Indication in the trace file and no in or decrease of the volume.

When the button is pressed 3 times the system hangs because the SD card will be opened. This does not work. The only way to get out of this is a reboot.

Groeten, Wim

Edzelf, 2018-11-30:
Maybe you should use Rainer's PCB....

paysan18, 2018-11-30:
Yes, good idea!

Groeten, Wim

paysan18, 2018-11-30:
Rainer, can I see a picture of your print for this radio please?

Rainer-G, 2018-12-01:
Here it is, Wim!

It uses different boards, I selected these for my tests as they were breadboard friendly!

Rainer

Von: Willem Nijntjes
Gesendet: Freitag, 30. November 2018 23:38
An: Edzelf/ESP32-Radio
Cc: Rainer-G; Comment
Betreff: Re: [Edzelf/ESP32-Radio] Rotary encoder question. (#192)

Rainer, can I see a picture of your print for this radio please?

Rainer-G, 2018-12-01:
Here is a picture of the PCB

Von: Willem Nijntjes
Gesendet: Freitag, 30. November 2018 23:38
An: Edzelf/ESP32-Radio
Cc: Rainer-G; Comment
Betreff: Re: [Edzelf/ESP32-Radio] Rotary encoder question. (#192)

Rainer, can I see a picture of your print for this radio please?

paysan18, 2018-12-01:
Problem found and case can be closed.

Problem was a solder bridge between the clk pin and ground of the rotary encoder.
Found it by looking at the bootup debug trace. GPIO-25 was LOW and remark was possible no pullup.

paysan18, 2018-12-01:
Rainer, I don't buy your PCB. I have a different type of ESP32 and VS1053.

closed 2018-12-01 by paysan18

paysan18, 2018-12-01:
I don’t see any picture. Is this the same pcb as sold by Naked Ninja in Roterdam?

paysan18, 2018-12-01:
> Rainer, can I see a picture of your print for this radio please?

Is this the pcb as mentioned in issue #181 ?