Edzelf/ESP32-Radio #193: Task watchdog got triggered

fanfanlatulipe26, 2018-11-30:
Hi Ed,

First of all congratulation for your work !
I try to run the radio on [this board](https://www.banggood.com/Wemos-ESP32-OLED-Module-For-Arduino-ESP32-OLED-WiFi-Bluetooth-Dual-ESP-32-ESP-32S-ESP8266-p-1148119.html?rmmds=myorder&cur_warehouse=CN). 
As far as I understand the log it seems all good until I get 👍
 E (13360) wifi: pp.c 3379

D: Duration mp3loop 802
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: loopTask

I haven't time to get a sound.
See attached full log.
I have the same problem if I remove the VS1053 board (of course with some errors in the log)
The board seems OK and works with other wifi test programs.
Thank you for your help.
[debug trace.txt](https://github.com/Edzelf/ESP32-Radio/files/2633856/debug.trace.txt)

Edzelf, 2018-11-30:
Looks like there is something wrong with the WiFi.  Hard to see from here what the problem is.  I cannot find the file pp.c on my PC...
I am using these libraries:
Using library PubSubClient at version 2.7 in folder: C:\Users\Ed\Documents\Arduino\libraries\PubSubClient 
Using library WiFi at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WiFi 
Using library ESPmDNS at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\ESPmDNS 
Using library FS at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\FS 
Using library SD at version 1.0.5 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\SD 
Using library SPI at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\SPI 
Using library ArduinoOTA at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\ArduinoOTA 
Using library Update at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\Update 
Using library Adafruit_ST7735_Library at version 1.0.8 in folder: C:\Users\Ed\Documents\Arduino\libraries\Adafruit_ST7735_Library 
Using library Adafruit_GFX_Library at version 1.3.4 in folder: C:\Users\Ed\Documents\Arduino\libraries\Adafruit_GFX_Library

fanfanlatulipe26, 2018-11-30:
Hi Ed,

Sorry for bothering you. 
I just partly solved the problem. Digging in the web I only found one useful   link https://www.esp32.com/viewtopic.php?t=6304  
The guy reported that there was a problem if a WEP key is used and it was the case in my network. I switched to WPA(TKIP+AES) and now the board runs better. It looks like this bug is still pending.
 Don't kow if it is worth to report it again. I don't get this trouble with ESP8266. But it is really a trap and the behaviour of the board is very strange, some times working, sometimes craching.

I still don't have sound with the radio but the debug log seems better. Just some messages like "D: SPI semaphore not taken within xxx ticks by CPU 1, id hspec" when I give a Next/Prrev/Stop/Unstop ... command.

I also have a the line "D: Switch to DATA, bitrate is 0, metaint is 0" when a station is selected. Correct ??
 I will investigate the VS1053 part of my wiring.  The "EN" pin which should be connected to the pin of XRST of the VS1053 is not available on my board and so XRST is left opened. Do you think it could be a problem ?
Thank you for your help.
Francis

  ----- Original Message ----- 
  From: Ed Smallenburg 
  To: Edzelf/ESP32-Radio 
  Cc: fanfanlatulipe26 ; Author 
  Sent: Friday, November 30, 2018 5:01 PM
  Subject: Re: [Edzelf/ESP32-Radio] Task watchdog got triggered (#193)

  Looks like there is something wrong with the WiFi. Hard to see from here what the problem is. I cannot find the file pp.c on my PC...
  I am using these libraries:
  Using library PubSubClient at version 2.7 in folder: C:\Users\Ed\Documents\Arduino\libraries\PubSubClient
  Using library WiFi at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WiFi
  Using library ESPmDNS at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\ESPmDNS
  Using library FS at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\FS
  Using library SD at version 1.0.5 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\SD
  Using library SPI at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\SPI
  Using library ArduinoOTA at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\ArduinoOTA
  Using library Update at version 1.0 in folder: C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\Update
  Using library Adafruit_ST7735_Library at version 1.0.8 in folder: C:\Users\Ed\Documents\Arduino\libraries\Adafruit_ST7735_Library
  Using library Adafruit_GFX_Library at version 1.3.4 in folder: C:\Users\Ed\Documents\Arduino\libraries\Adafruit_GFX_Library

Edzelf, 2018-11-30:
Concentrate on the line with "bitrate =0".  In your first logging I saw 128.  Please show your logging.

fanfanlatulipe26, 2018-12-01:
Here is the log
[debug trace 2.txt](https://github.com/Edzelf/ESP32-Radio/files/2635723/debug.trace.2.txt)