Edzelf/ESP32-Radio #195: Replace LCD 1602 with 2004

h1aji, 2018-12-05:
I think it would be nice to have bigger display that would be more informative and will show volume level bitrate and time with date.

Since there is not much difference in price.

mrohner, 2018-12-07:
I included this for that purpose. You also have to modify the main.
//***************************************************************************************************
//*  LCD2004.h -- Driver for LCD 2004 display with I2C backpack.                                    *
//***************************************************************************************************
// The backpack communicates with the I2C bus and converts the serial data to parallel for the      *
// 2004 board.  In the serial data, the 8 bits are assigned as follows:                             *
// Bit   Destination  Description                                                                   *
// ---   -----------  ------------------------------------                                          *
//  0    RS           H=data, L=command                                                             *
//  1    RW           H=read, L=write.  Only write is used.                                         *
//  2    E            Enable                                                                        *
//  3    BL           Backlight, H=on, L=off.  Always on.                                           *
//  4    D4           Data bit 4                                                                    *
//  5    D5           Data bit 5                                                                    *
//  6    D6           Data bit 6                                                                    *
//  7    D7           Data bit 7                                                                    *
//***************************************************************************************************
//
// Note that the display function are limited due to the minimal available space.

#include <driver/i2c.h>

#define I2C_ADDRESS 0x27                                     // Adjust for your display
#define ACKENA      true                                     // Enable ACK for I2C communication
// Color definitions for the TFT screen (if used)
#define BLACK   0
#define BLUE    1
#define RED     1
#define GREEN   0
#define CYAN    GREEN | BLUE
#define MAGENTA RED | BLUE
#define YELLOW  RED | GREEN
#define WHITE   RED | GREEN | BLUE

#define DELAY_ENABLE_PULSE_SETTLE    50      // Command requires > 37us to settle
#define FLAG_BACKLIGHT_ON    0b00001000      // Bit 3, backlight enabled (disabled if clear)
#define FLAG_ENABLE          0b00000100      // Bit 2, Enable
#define FLAG_RS_DATA         0b00000001      // Bit 0, RS=data (command if clear)
#define FLAG_RS_COMMAND      0b00000000      // Command

#define COMMAND_CLEAR_DISPLAY               0x01
#define COMMAND_RETURN_HOME                 0x02
#define COMMAND_ENTRY_MODE_SET              0x04
#define COMMAND_DISPLAY_CONTROL             0x08
#define COMMAND_FUNCTION_SET                0x20
#define COMMAND_SET_DDRAM_ADDR              0x80
//
#define FLAG_DISPLAY_CONTROL_DISPLAY_ON     0x04
#define FLAG_DISPLAY_CONTROL_CURSOR_ON      0x02
//
#define FLAG_FUNCTION_SET_MODE_4BIT         0x00
#define FLAG_FUNCTION_SET_LINES_2           0x08
#define FLAG_FUNCTION_SET_DOTS_5X8          0x00
//
#define FLAG_ENTRY_MODE_SET_ENTRY_INCREMENT 0x02
#define FLAG_ENTRY_MODE_SET_ENTRY_SHIFT_ON  0x01
//
// Various macro's to mimic the ST7735 version of display functions
#define dsp_setRotation()                                          // Use standard landscape format
#define dsp_print(a)                                               // Print a string 
#define dsp_println(a)                                             // Print string plus newline
#define dsp_fillRect(a,b,c,d,e)                                    // Fill a rectange
#define dsp_setTextSize(a)                                         // Set the text size
#define dsp_setTextColor(a)                              
#define dsp_setCursor(a,b)                                         // Position the cursor
#define dsp_erase()         tft->sclear()                          // Clear the screen
#define dsp_getwidth()      20                                     // Get width of screen
#define dsp_getheight()     4                                      // Get height of screen
#define dsp_usesSPI()       false                                  // Does not use SPI

[... 372 lines cut ...]
//                                                         *
//**************************************************************************************************
void displaytime ( const char* str, uint16_t color ){
  const char* WDAYS [] ={"Mon","Tue","Wed","Thu","Fri","Sat","Sun"};
  char        datetxt[21] ;                           
  sprintf ( datetxt, "%03s %02d.%02d.  %08s",                // Format new time to a string
                  WDAYS[timeinfo.tm_wday-1],
                  timeinfo.tm_mday,
                  timeinfo.tm_mon,
                  str) ;
  char oldstr[21] = "...................." ;             // For compare
  uint8_t     i ;                                  // Index in strings
  if ( datetxt[14] == '0' ) {                           // Empty string?
    for ( int i = 0 ; i < 21 ; i++ )                    // Set oldstr to dots
       oldstr[i] = '.' ;
    return ;                                       // No actual display yet
  }
  
    if ( datetxt != oldstr ) {                  // Difference?
       dline[0].str = datetxt;          
       dsp_update_line(0);
       for ( int i = 0 ; i < 21 ; i++ )                    // Set oldstr to dots
       oldstr[i] = datetxt[i] ;
      }
}

mrohner, 2018-12-07:
I added this as a file. You have to also update the main, change the include.
```
//***************************************************************************************************
//*  LCD2004.h -- Driver for LCD 2004 display with I2C backpack.                                    *
//***************************************************************************************************
// The backpack communicates with the I2C bus and converts the serial data to parallel for the      *
// 2004 board.  In the serial data, the 8 bits are assigned as follows:                             *
// Bit   Destination  Description                                                                   *
// ---   -----------  ------------------------------------                                          *
//  0    RS           H=data, L=command                                                             *
//  1    RW           H=read, L=write.  Only write is used.                                         *
//  2    E            Enable                                                                        *
//  3    BL           Backlight, H=on, L=off.  Always on.                                           *
//  4    D4           Data bit 4                                                                    *
//  5    D5           Data bit 5                                                                    *
//  6    D6           Data bit 6                                                                    *
//  7    D7           Data bit 7                                                                    *
//***************************************************************************************************
//
// Note that the display function are limited due to the minimal available space.

#include <driver/i2c.h>

#define I2C_ADDRESS 0x27                                     // Adjust for your display
#define ACKENA      true                                     // Enable ACK for I2C communication
// Color definitions for the TFT screen (if used)
#define BLACK   0
#define BLUE    1
#define RED     1
#define GREEN   0
#define CYAN    GREEN | BLUE
#define MAGENTA RED | BLUE
#define YELLOW  RED | GREEN
#define WHITE   RED | GREEN | BLUE

#define DELAY_ENABLE_PULSE_SETTLE    50      // Command requires > 37us to settle
#define FLAG_BACKLIGHT_ON    0b00001000      // Bit 3, backlight enabled (disabled if clear)
#define FLAG_ENABLE          0b00000100      // Bit 2, Enable
#define FLAG_RS_DATA         0b00000001      // Bit 0, RS=data (command if clear)
#define FLAG_RS_COMMAND      0b00000000      // Command

#define COMMAND_CLEAR_DISPLAY               0x01
#define COMMAND_RETURN_HOME                 0x02
#define COMMAND_ENTRY_MODE_SET              0x04
#define COMMAND_DISPLAY_CONTROL             0x08
#define COMMAND_FUNCTION_SET                0x20
#define COMMAND_SET_DDRAM_ADDR              0x80
//
#define FLAG_DISPLAY_CONTROL_DISPLAY_ON     0x04
#define FLAG_DISPLAY_CONTROL_CURSOR_ON      0x02
//
#define FLAG_FUNCTION_SET_MODE_4BIT         0x00
#define FLAG_FUNCTION_SET_LINES_2           0x08
#define FLAG_FUNCTION_SET_DOTS_5X8          0x00
//
#define FLAG_ENTRY_MODE_SET_ENTRY_INCREMENT 0x02
#define FLAG_ENTRY_MODE_SET_ENTRY_SHIFT_ON  0x01
//
// Various macro's to mimic the ST7735 version of display functions
#define dsp_setRotation()                                          // Use standard landscape format
#define dsp_print(a)                                               // Print a string 
#define dsp_println(a)                                             // Print string plus newline
#define dsp_fillRect(a,b,c,d,e)                                    // Fill a rectange
#define dsp_setTextSize(a)                                         // Set the text size
#define dsp_setTextColor(a)                              
#define dsp_setCursor(a,b)                                         // Position the cursor
#define dsp_erase()         tft->sclear()                          // Clear the screen
#define dsp_getwidth()      20                                     // Get width of screen
#define dsp_getheight()     4                                      // Get height of screen
#define dsp_usesSPI()       false                                  // Does not use SPI
[... 374 lines cut ...]
//**************************************************************************************************
void displaytime ( const char* str, uint16_t color ){
  const char* WDAYS [] ={"Mon","Tue","Wed","Thu","Fri","Sat","Sun"};
  char        datetxt[21] ;                           
  sprintf ( datetxt, "%03s %02d.%02d.  %08s",                // Format new time to a string
                  WDAYS[timeinfo.tm_wday-1],
                  timeinfo.tm_mday,
                  timeinfo.tm_mon,
                  str) ;
  char oldstr[21] = "...................." ;             // For compare
  uint8_t     i ;                                  // Index in strings
  if ( datetxt[14] == '0' ) {                           // Empty string?
    for ( int i = 0 ; i < 21 ; i++ )                    // Set oldstr to dots
       oldstr[i] = '.' ;
    return ;                                       // No actual display yet
  }
  
    if ( datetxt != oldstr ) {                  // Difference?
       dline[0].str = datetxt;          
       dsp_update_line(0);
       for ( int i = 0 ; i < 21 ; i++ )                    // Set oldstr to dots
       oldstr[i] = datetxt[i] ;
      }
}
```

ememem13, 2019-04-11:
Can scrolling be accelerated?

h1aji, 2020-07-12:
@Edzelf Any chance to add LCD2004 code. I have tested it and itworks ok

Edzelf, 2020-07-13:
I 've ordered an LCD2004.  I will add support as soon as I have tested it.

avulyanichev, 2020-07-22:
Здравствуйте Эд. Экран 1602 с поддержкой кирилицы. Успешно встроен в 1602.h  русский язык. Могу выложить файл для вашего тестирования,  английский язык поддерживается полностью.
[image]

avulyanichev, 2020-07-22:
Эд, пожайлуста помогите сделать вывод на экран название станций из config файла. Это очень улучшит изображение названия станций, так как в config файле можно самому редактировать название станций. Спасибо.

h1aji, 2020-08-10:
@mrohner is it possible to implement volume progress bar like here: https://www.electronicsblog.net/arduino-lcd-horizontal-progress-bar-using-custom-characters/

h1aji, 2020-09-21:
Hi @Edzelf thanks for adding it in the latest release. The only thing i wanted to improve is volume progress bar. So instead of having minus have a full square
This line:
`        dline[3].str += "-" ;                         // Add minus sign`

is it possible to create custom character like 
```
byte customChar[] = {
  0x1F,
  0x1F,
  0x1F,
  0x1F,
  0x1F,
  0x1F,
  0x1F,
  0x1F
};
```
and print it then?

Edzelf, 2020-09-22:
Maybe I'm wrong, but could you try:
`dline[3].str += "\x7F" ;                      // Add block character`

h1aji, 2020-09-23:
@Edzelf didnt work, shows nothing.
Found alternative solution :)        
 `dline[3].str += "#" ;`