Edzelf/ESP32-Radio #196: Stuck on boot

toblum, 2018-12-09:
Hi @Edzelf,

I'm trying to get the sketch to work but wasn't very successful.

I was able to compile everything without problems. I just had to use PubSub 2.6 instead of the latest version, because that breaks compilation (undefined reference to `vtable for PubSubClient').

Init script was applied with wifi credentials. When I start the board I get the following output:
D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 04 Oct 2018 07:22:32 GMT.  Free memory 171036
D: Display type is DUMMYTFT
D: Partition nvs found, 20480 bytes

Nothing else happens after that. I also tried with nothing connected to the ESP32 DOIT v1 board. Same result.

Is there anything else that I missed? Do I have to use special versions for the libraries?

Regards
Tobias

Edzelf, 2018-12-09:
Looks like there is something wrong with the NVS.  After the detection of nvs, the debug output should list all the entries in the nvs.  Did you run the init-script without errors?

toblum, 2018-12-09:
Hi @Edzelf,

yes, I saw how the output should look in other posts, but in my case it didsn't print out anything.

Today I gave it another try and after reflashing and initializing several times, it works now! I didn't change anything in the code in comparison to yesterday but something else must have changed. I may never find out what. But as it works now I'm fine with it. :-)

Nevertheless: Thank you for your support. I will now hook up the other parts and see if the rest works as expected.

Best regards
Tobias

closed 2018-12-09 by toblum