Edzelf/ESP32-Radio #197: Installmof Nextion display

paysan18, 2018-12-09:
Hello all, I want to connect a Nextion display to the radio.
When I supply power to the display it starts running a kind of Demo.
So it seems I have to install new formware to it.
I found inside the folder NEXTION 2 files. A .HMI of about 35kB and a .tft file of 2 bytes.
After copying the .HMI file to a fresh formatted SD card I inserted this card into the display and connnected the power. On the screen comes: SD update...... and all remains like this.
What do I do wrong?
Regards, Wim

Edzelf, 2018-12-09:
Looks like the .tft file is damaged.  You need a .tft file to upload the data to the nextion.  You may use the nextion compiler to create the .tft file.  I will upload a new .tft file tomorrow.

paysan18, 2018-12-09:
Ok, but what to do with the .HMI file?

Edzelf, 2018-12-10:
.hmi is the source file.  It is the input for the Nextion compiler.  You may edit the .hmi file to adapt the  nextion interface to your needs.

Edzelf, 2018-12-10:
.tft file is now complete (139 kB).

paysan18, 2018-12-10:
Great, thank a lot. I will try it this afternoon.

paysan18, 2018-12-10:
Ik probeer het van,iddag. Moet nu even weg.

Groeten, Wim

paysan18, 2018-12-10:
Update succeeded. But during te restart of the radio the display shows:

Read SD card
27 tracks on SD

and this loops every second.

There is no card inserted.

paysan18, 2018-12-10:
Everything is OK now. It seemed I also had problems with the powersupply.
Ed, thanks for uploading the .tft file.

paysan18, 2018-12-10:
Issue can be closed.

closed 2018-12-10 by paysan18