Edzelf/ESP32-Radio #198: PCB Webradio available

certeza, 2018-12-14:
Dear friends,

It took some time because I was moving my workshop, but I can now offer the webradio PCB with USB connector, diode and electrolytic capacitor already soldered onto the board, see 
[image]
This is a new version of the PCB, the potmeter switch of the PAM aplifier will now switch the power of the radio on and off.

Also included is a Bill of Material with links to components sold on AliExpress. We have no affiliation with AliExpress, so if can you find them cheaper or better elsewhere, be my guest. 

Price for the PCB with the 3 components is 15 euros, shipping is 3.50 euros. So combined makes 18.50 euros. Please send an email with your shipping details to info@certeza.nl, and transfer 18.50 euros to Certeza BV, NL11 KNAB 0256 5662 59 or to PayPal account info@certeza.nl with description "webradio pcb" and add your email address.

Best Regards,
Harm

Edzelf, 2018-12-18:
I've build the radio with your PCB.  It worked without problems.  The pins used are the same as my default settings.  I'm still waiting for the PAM amplifier to arrive from China, but with an external amplifier it plays.
Good to hear that the USB connector is already mounted, because that's a little bit hard to do.

MakeySQ, 2019-01-04:
That's cool!!

certeza, 2019-01-12:
All,

The PCB uses GPIO 16 for the DCS pin of the VS1053, so set this accordingly in defaultprefs.h:
`pin_vs_dcs = 16                                      # GPIO Pin number for VS1053 "DCS"`

Edzelf, 2019-01-12:
Note that changing defaultprefs.h does NOT have direct effect on the settings.  Use the config page of the webinterface to change all settings including pin_vs_dcs.

trombose009, 2019-11-20:
I have build the radio too. 
I have two principal questions: 
1.) To do the upload of the sketch, you have to plug into the esp32 directly, right? If I do so, it ends up in a boot loop. I can give more information if neccesary. 
2.) If i plug in a USB-power supply at the onboard usb connector, nothing happens. the led of the esp32 does not switch on.

What could be wrong?
Has anyone any idea?

Edzelf, 2019-11-20:
Use the bare ESP32, without any wires or connections.  Then use a USB cable to connect the ESP32 to your PC running the Arduino IDE.  Upload the sketch.  Replace the ESP32 is no success.

trombose009, 2019-11-20:
thank you for answering that fast. I gonna try that. thanks.

Rainer-G, 2019-11-20:
Looking at the PCB picture, I cannot find the PCB's USB connector, on December's picture it is on top side of the board???
Did you check the voltage ?

Rainer

trombose009, 2019-11-20:
https://user-images.githubusercontent.com/23176307/49999283-b4e6c500-ff96-11e8-9ae9-41a038121670.jpg
it's the same PCB. top left you can see the micro usb connector.  I will check the voltage and try to remove the esp32 from the board this evening.

nakedninja-design, 2019-11-20:
Don't forget that the potmeter switch of the PAM module is used to switch the power on and off for the entire board, including the ESP32.
You should use the usb connector of the ESP32 itself to program the ESP32.

Harm

trombose009, 2019-11-20:
I have removed the ESP32-WROOM-32 from the PCB. If I try to upload any sketch, it still goes into a boot loop...

`rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
ets Jun  8 2016 00:22:57
`

is it corrupt?

uploading of the sketch seems to be fine. But then the boot loop.

Rainer-G, 2019-11-21:
Before putting the ESP32 into the bin, try with a differenz ( shorter ) USB cable and/or reduce the upload speed ...

Edzelf, 2019-11-21:
As I understand, the upload succeeds, but the sketch does not start.  Try the blink example first.

trombose009, 2019-11-21:
I did so. Same boot loop. I ordered a new one.

trombose009, 2019-11-22:
OK, I moved on and get the blinking thing working. but not by using Arduino IDE for uploading. I did following with success: 
1.)
esptool.exe --chip esp32 --port COM3 --baud 115200 --before default_reset --after hard_reset erase_flash
2.)
esptool.exe --chip esp32 --port COM3 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x8000 ESPEasy.ino.partitions.bin 0xe000 boot_app0.bin 0x1000 bootloader.bin
3.)
esptool.exe --chip esp32 --port COM3 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x10000 sketch_nov20a.ino.esp32.bin

sketch_nov20a.ino.esp32.bin is exported from Arduino IDE.

If I try upload the ino-file with IDE I get the boot loop. Is there any configuration wrong inside Arduino IDE?

[image]

trombose009, 2019-11-22:
OMG! suddenly it is working! I change nothing. just played around with the configuration of IDE. But With this setup shown in the screenshot, it is working now. 
Thank you for helping me. 
I know, it was a little bit off-topic but maybe it helps other newbies getting things working well. 
Thank you.

Rainer-G, 2019-11-24:
Just for my curiosity:
if you change Flash Frequency to 80 MHz and Upload Speed to 921600, will it still work ? 
Please try one after the other, not both changes at the same time.
When I select the ESP32 Dev. Module, these values are the default ones, all other are like the ones on you screenshot.
Many thanks, Rainer

trombose009, 2019-11-25:
I am not sure if i want to do that. I am happy to get this thing working now. :-)I tried different speeds, especially slower baud rates. always also adepting the driver baud rate in windows of course. but i am not sure if i have tried the suggested speed. finally (at the moment) it's also not 921600 which is working well now.

trombose009, 2019-11-26:
It drives me crazy. I can visit the site of the radio, the nextion display also shows me the station. But the speaker only rustles and at the display I can see the radio is more or less permanently rebooting. during the reboot it tells me that the SD Card mount fails; Hopefully that'S normal becaus i dont have any SD card reader attached (except that on, which is included in the nextion display. But I dont think you could use this reader?!).

Edzelf, 2019-11-26:
The answer is in the logging at the serial output.

trombose009, 2019-11-27:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

D: Starting ESP32-radio running on CPU 1 at 160 MHz.  Version Thu, 16 May 2019 12:10:00 GMT.  Free memory 294204
D: Display type is NEXTION
D: Partition nvs found, 20480 bytes
D: Read 43 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 21
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Enable pin 35 for IR
D: Start display
D: Init Nextion, pins 16,17
D: Pin 16 is already reserved!
D: Nextion command cls BLACK
D: Nextion command page 0
D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added dampfmaschine2 to list of networks
D: Added SSID2 to list of networks
D: End adding networks
[... 199 lines cut ...]
D: SPI semaphore not taken within 320 ticks by CPU 1, id hspec
D: SPI semaphore not taken within 330 ticks by CPU 1, id hspec
D: SPI semaphore not taken within 120 ticks by CPU 1, id hspec
D: SPI semaphore not taken within 120 ticks by CPU 1, id hspec
D: SPI semaphore not taken within 130 ticks by CPU 1, id hspec
D: SPI semaphore not taken within 140 ticks by CPU 1, id hspec
D: SPI semaphore not taken within 150 ticks by CPU 1, id hspec
D: SPI semaphore not taken within 160 ticks by CPU 1, id hspec
D: SPI semaphore not taken within 170 ticks by CPU 1, id hspec
D: SPI semaphore not taken within 180 ticks by CPU 1, id hspec
D: SPI semaphore not taken within 190 ticks by CPU 1, id hspec
D: SPI semaphore not taken within 200 ticks by CPU 1, id hspec
D: SPI semaphore not taken within 210 ticks by CPU 1, id hspec
D: SPI semaphore not taken within 220 ticks by CPU 1, id hspec
D: (Re)connecting number 2 to MQTT 192.168.178.30
D: MQTT connection failed, rc=-2
E (15864) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (15864) task_wdt:  - IDLE0 (CPU 0)
E (15864) task_wdt: Tasks currently running:
E (15864) task_wdt: CPU 0: Playtask
E (15864) task_wdt: CPU 1: loopTask
E (15864) task_wdt: Aborting.
abort() was called at PC 0x400e5563 on core 0

Backtrace: 0x4008cc58:0x3ffbe170 0x4008ce89:0x3ffbe190 0x400e5563:0x3ffbe1b0 0x40084f25:0x3ffbe1d0 0x400e29ce:0x3ffd78d0 0x400df569:0x3ffd78f0 0x400d2624:0x3ffd7910 0x40089369:0x3ffd7930

trombose009, 2019-11-27:
> Just for my curiosity:
> if you change Flash Frequency to 80 MHz and Upload Speed to 921600, will it still work ?
> Please try one after the other, not both changes at the same time.
> When I select the [...]

I tried to do that. but it is not possible to switch the driver inside windows device manager to 921600. so it's not possible.

certeza, 2019-11-27:
The semaphore errors are indication of insufficient power or poor wiring.

Regards,
Harm

Rainer-G, 2019-11-27:
Both changes are done inside the Arduino IDE, look at your screenshot, there are select option on the right of upload speed and flash frequency, 
( but this does not solve your problem with the radio station …)

Von: trombose009
Gesendet: Mittwoch, 27. November 2019 20:21
An: Edzelf/ESP32-Radio
Cc: Rainer-G; Comment
Betreff: Re: [Edzelf/ESP32-Radio] PCB Webradio available (#198)

Just for my curiosity:
if you change Flash Frequency to 80 MHz and Upload Speed to 921600, will it still work ?
Please try one after the other, not both changes at the same time.
When I select the ESP32 Dev. Module, these values are the default ones, all other are like the ones on you screenshot.
Many thanks, Rainer
I tried to do that. but it is not possible to switch the driver inside windows device manager to 921600. so it's not possible.

Edzelf, 2019-11-27:
Note the error: VS1053 not properly installed.

trombose009, 2019-12-03:
ok, i found two cold soldered points. since i have corrected this, there are no reboots anymore. 
but i still can'T hear any music. just rustles, maybe rustles with some kind of rhythm. 
the logging:

Davet998, 2021-07-17:
Will you be getting more stock of these boards? Regards Dave

iMike78, 2022-02-28:
Hi! I can see that your webshop is closed. Is there any chance to share the gerber files to order the board for myself? Cheers

z34pec, 2023-05-11:
Hello. Regarding this post, anyone managed to get gerber or dip files for this pcb because it is not sold anymore and ninja's site is closed ? Also anyone tried to contact ninja so get info how to get this pcb board ?

Edzelf, 2023-05-11:
There is a .dip file on [https://github.com/Edzelf/ESP32Radio-V2/blob/main/doc/ESP32-Radio.dip](url).
You can open it with the free version of "Dip Trace PCB", see [https://diptrace.com/download/download-diptrace/](url), change it if you like and generate gerber files.

z34pec, 2023-05-11:
> There is a .dip file on [https://github.com/Edzelf/ESP32Radio-V2/blob/main/doc/ESP32-Radio.dip](url). You can open it with the free version of "Dip Trace PCB", see [https://diptrace.com/download/download-diptrace/](url), change it if you like and generate gerber files.

Thank you for this information. I already did it, and upload it on JLCPCB. Because I am a newbie to this I wonder, due to two layer PCB, where soldering is made? Do we solder on only bottom side or on both sides ? Any kind of wiring diagram is welcome. :)

Edzelf, 2023-05-11:
The holes are plated-through.  So you just solder the backside.  I did not design the print, so the wiring requires some reverse engineering.