Edzelf/ESP32-Radio #203: feature request

airasz, 2018-12-19:
Can you add feature to send stream title to web client when web client connected and when stream title change because play next song? I try write httpheader ( String ( "text/html" ) ) ;  cmdclient.print("test send"); .But no luck. Thanks

perlix, 2018-12-20:
If I understand your request correctly, it can be done by changing the index_html.c file to make it send an AJAX request to the server, say every 2 seconds. This is what I did:

1. In Esp32-Radio.ino, change the line
`sprintf ( reply, "%s - %s", icyname.c_str(),`
to
`sprintf ( reply, "%s ~ %s", icyname.c_str(),`

2. Change the content of index_html.c so it will refresh station name and song title every 2 seconds. 
As an example, the content of my modified (and slightly stripped) index_html.c file can be found [here](http://dellascolto.com/temp/index_html.h.txt). 

After these changes, the web interface of my radio looks like [this](http://dellascolto.com/temp/esp32-Radio.jpg) and it automatically refreshes station name and/or song title when they change.

airasz, 2018-12-22:
Thank for reply. 
For information, I used own native android app as web client. There using android volley library to make HTTP request for string radio information and send some command. 
And refreshing radio information also can be done by periodically send requests for radio status. But what I want is esp send information to client just when audio title change not by request from client. Are we can do that?

koskee, 2019-01-02:
Of course it is possible, but the mechanism behind a standard http request doesn't allow this type of communication. The page currently uses AJAX and a similar concept to REST (although REST technically has a few subtle differences). Basically this is essentially a "hack" that enables updating of a webpage dynamically that somebody figured out at some point, but JavaScript and http web requests were never designed or intended to be used this way, which is why the protocol falls short in this sense. 

To do it you'd need to get into web sockets, long/short polling (short polling was already discussed above), server-sent events, or I believe an embedded Java applet would also have this capability. There's some newer technology based on javascript and HTML5 as well, called "service workers" if my memory serves me.

Here is a page that does a decent job of explaining some of these concepts. 
https://www.smashingmagazine.com/2018/02/sse-websockets-data-flow-http2/

Personally, I went the short polling route as well due to  performance considerations and for simplicity, rather than trying to maintain an open session.

blotfi, 2020-06-06:
Control the Web Radio with a remote control on Android:
https://play.google.com/store/apps/details?id=com.embesystems.esp_radio_rc

No Ads, totally free, no authorisation (only internet access to send/receive data from ESP32 web radio)

I made it for myself, for my web radios, and I share the App with all the happy users of these ESP32 / 8266 web radios.

Try my mods:
https://github.com/blotfi/ESP32-Radio-with-Spectrum-analyzer
and
https://github.com/blotfi/ESP32-Radio-with-SalatTime
Enjoy