Edzelf/ESP32-Radio #204: No sound ....

cherob67, 2018-12-19:
Hello
I an trying to make the ESP32-RADIO. I succeed in wifi connection and radio server connection. I also succeed in sdcard connection. But i have error in fast SPI testing VS1053.
I use a ESP-Wroom-32 and VS1053 bought on internet like this one ([https://www.amazon.fr/VS1053B-Arduino-Breakout-Real-Time-Recording/dp/B07J63G1RX/ref=sr_1_52?ie=UTF8&qid=1545249974&sr=8-52&keywords=vs1053](https://www.amazon.fr/VS1053B-Arduino-Breakout-Real-Time-Recording/dp/B07J63G1RX/ref=sr_1_52?ie=UTF8&qid=1545249974&sr=8-52&keywords=vs1053)).
The wiring has verified at many time and now I need some help to find a solution.
Thanks

I join the serial response at starting:
D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 04 Oct 2018 07:22:32 GMT.  Free memory 233728
D: Display type is OLED
D: Partition nvs found, 20480 bytes
D: Read 29 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 22
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Locate mp3 files on SD, may take a while...
D: 11 tracks on SD
D: Create list with acceptable WiFi networks
D: Added domicile_rc to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 5
D:  1 - Bbox-B476B850             Signal: -79 dBm, Encryption WPA_WPA2_PSK, 
D:  2 - domicile_rc               Signal: -84 dBm, Encryption WPA_WPA2_PSK, Acceptable
D:  3 - XXXX-1698                 Signal: -85 dBm, Encryption WPA_WPA2_PSK, 
D:  4 - rodgers25                 Signal: -89 dBm, Encryption WPA_WPA2_PSK, 
D:  5 - freebox_CeliaSXB          Signal: -91 dBm, Encryption WPA_PSK, 
D: End of list
[... 158 lines cut ...]
D: Command client available
D: handlehttp started
D: Get command is: status
D: Send reply for status
D: Command: status with parameter 0
D: Command client available
D: handlehttp started
D: Get command is: test
D: Send reply for test
D: Command: test with parameter 0
D: Stack maintask is 4732
D: Stack playtask is 1096
D: Stack spftask  is 660
D: ADC reading is 145
D: scaniocount is 10
D: Max. mp3_loop duration is 191
D: Command client available
D: handlehttp started
D: HTTP input shorter than expected
D: Command client available
D: handlehttp started
D: Get command is: stop
D: Send reply for stop
D: Command: stop with parameter 0
D: Command client available

Edzelf, 2018-12-20:
The debug logging looks normal.  But there are still problems with the SPI bus.
You may try to remove the lines (around line 923):
```
    //SPI Clock to 4 MHz. Now you can set high speed SPI clock.
    VS1053_SPI = SPISettings ( 5000000, MSBFIRST, SPI_MODE0 ) ;
```
The SPI will run on a slower rate.

cherob67, 2018-12-20:
Thanks for your answer, I an going to try your solution. 
After asking my question yesterday, by chance in removing the sdcard, the sound came. I do not know why ??? and if I put the sdcard the sound is cut off immediately.
strange....

Edzelf, 2018-12-20:
I did not realize that you had an sd card connected.  There are some known problems with some sd card interfaces.

cherob67, 2018-12-20:
OK I have tried your solution without and with the sdcard. Without there is no difference but i think it's better to keep the code. With the sdcard, it works but the sound is bad.
Thank you

closed 2018-12-21 by cherob67