Edzelf/ESP32-Radio #208: Slow SPI, Testing VS1053 read/write registers / VS1053 not properly installed

HaViGit, 2019-01-03:
Hi,

First thanks for sharing this project! Sorry to bore you with my problem but hope you can help with a VS1053 issue. After uploading the sketch I can open the web interface without any problem via WiFi. I did not yet connect my blue 128x160 TFT but the VS1053 is connected as described in your docs. Using the same ESP32 and VS1053 board (with VS1053B chip on board) as you did I’ve checked the wiring severtal times:

pin 1 > D32	pin 4 > D4	 pin 7 > D19
pin 2 > D5      pin 5 > D18     pin 8 > GND
pin 3 > EN      pin 6 > D23     pin 9 > 5V (VIN)

And I have also run the ESP-radio-int sketch and defined one display via #define BLUETFT in the main sketch. No further editing of the original sketch.

The serial monitor shows:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1324
load:0x40078000,len:7788
ho 0 tail 12 room 4
load:0x40080400,len:6448
entry 0x400806e8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 04 Oct 2018 07:22:32 GMT.  Free memory 295208
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 31 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to -1
D: pin_vs_dcs set to -1
D: pin_vs_dreq set to -1
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is HIGH
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is HIGH
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Create list with acceptable WiFi networks
D: Added TIS to list of networks
D: Added TISMK to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 5
D:  1 - T**              Signal: -67 dBm, Encryption WPA_PSK, Acceptable
D:  2 - T**              Signal: -79 dBm, Encryption WPA_WPA2_PSK, Acceptable
D:  3 - T**              Signal: -83 dBm, Encryption WPA2_PSK, Acceptable
D:  4 - V**             Signal: -85 dBm, Encryption WPA2_PSK, 
D:  5 - M**            Signal: -88 dBm, Encryption WPA2_PSK, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqtttpreset with parameter none
D: Command: mqttuser with parameter none
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 72
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 not properly installed!
D: Connect to WiFi
D: Connected to T**
D: IP = 192.168.2.43

As you can see there is a VS1053 issue: 

D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 not properly installed!

There also other warnings:

D: GPIO2/34/39 is LOW, probably no PULL-UP

Can you give me some advice to solve this?

Best regards, Hans

3zra, 2019-01-03:
Hi HaViGit,

I'm currently having a similar problem, so it may be useful to share my experience.

I guess, your problem is due to the fact that some VS1053 pins are not defined:

> D: pin_vs_cs set to -1
> D: pin_vs_dcs set to -1
> D: pin_vs_dreq set to -1

Set the pin definitions according to your wiring in the config panel of the web interface - reboot.

Best, 3sra

HaViGit, 2019-01-03:
Hi 3sra,

Thanks for helping! I tried your suggestion on several ways but I'm afraid not using the right way. Can you help wit the exact code to enter in the config screen? CS is connected to D5 (GPIO5), DCS to D32 (GPIO32) and DREQ to D4 (GPIO4). These connections are according Ed's docs.

Best regards, Hans.

MakeySQ, 2019-01-03:
Hi 
In my case
If connect esp32 via webinterface, I recommand default setting -> save -> restart in config page.

ShinSongSup, 2019-01-03:
Hi
good working is good news and it is my pleasure.
It took me a long time to succeed

paysan18, 2019-01-09:
I have exactly the same problem. 
A previous version runs fine. 
Tomorrow I will compare the configuration of HaViGit to what I have made sofar.