Edzelf/ESP32-Radio #209: VS1053 module or SPI problem

3zra, 2019-01-03:
Hi,

thanks for the great project with a sound documentation. I tried to implement a very basic setup: [ESP32 DEVKIT V1](https://www.ebay.de/itm/ESP32-ESP32S-CP2102-Development-Board-2-4GHz-Dual-Mode-WiFi-Bluetooth-Antenna/201853283621?hash=item2eff64ad25:g:~HEAAOSwUoNaJ15-:rk:1:pf:0) (30 GPIOs) + [VS1053b module](https://www.ebay.de/itm/VS1053-MP3-Module-Development-Board-w-On-Board-Recording-Function-SPI-Interface/310645105092?hash=item4853e421c4:g:wqwAAOSwL-xcGv2W:rk:1:pf:0) (no jumper wire shorting the line-in Pins) - no display, no encoder.

I use the standard wiring, which I have double-checked several times:

- ESP32 board PIN 5 <> VS1053b module CS
- ESP32 board PIN 16 <> VS1053b module DCS
- ESP32 board PIN 4 <> VS1053b module DREQ
- ESP32 board PIN 18 <> VS1053b module SCK
- ESP32 board PIN 19 <> VS1053b module MISO
- ESP32 board PIN 23 <> VS1053b module MOSI

I'm getting errors related to the VS1053b module (see Log 11:04:21.176 ff.) and SPI errors (,e.g. see Log 11:04:31.967 ff.) and don't get any sound from the headphones connected to the audio jack.

I'm lost - help is very much appreciated.

Thanks in advance, 3zra

Log (w/ timestamps):
> 11:04:16.927 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 04 Oct 2018 07:22:32 GMT.  Free memory 290520
11:04:16.960 -> D: Display type is NEXTION
11:04:16.960 -> D: Partition nvs found, 20480 bytes
11:04:16.960 -> D: Read 46 keys from NVS
11:04:16.960 -> D: pin_ir set to 35
11:04:16.960 -> D: pin_enc_clk set to 25
11:04:16.960 -> D: pin_enc_dt set to 26
11:04:16.960 -> D: pin_enc_sw set to 27
11:04:16.960 -> D: pin_tft_cs set to 15
11:04:16.960 -> D: pin_tft_dc set to 2
11:04:16.960 -> D: pin_tft_scl set to -1
11:04:16.960 -> D: pin_tft_sda set to -1
11:04:16.960 -> D: pin_tft_bl set to -1
11:04:16.960 -> D: pin_tft_blx set to -1
11:04:17.056 -> D: pin_sd_cs set to 21
11:04:17.056 -> D: pin_vs_cs set to 5
11:04:17.056 -> D: pin_vs_dcs set to 32
11:04:17.056 -> D: pin_vs_dreq set to 4
11:04:17.056 -> D: pin_shutdown set to -1
11:04:17.056 -> D: pin_shutdownx set to -1
11:04:17.056 -> D: pin_spi_sck set to 18
11:04:17.056 -> D: pin_spi_miso set to 19
11:04:17.056 -> D: pin_spi_mosi set to 23
11:04:17.056 -> D: GPIO0 is HIGH
11:04:17.056 -> D: GPIO2 is LOW, probably no PULL-UP
11:04:17.056 -> D: GPIO4 is HIGH
11:04:17.056 -> D: GPIO5 is HIGH
11:04:17.056 -> D: GPIO12 is HIGH
11:04:17.056 -> D: GPIO13 is HIGH
11:04:17.056 -> D: GPIO14 is HIGH
11:04:17.056 -> D: GPIO15 is HIGH
11:04:17.122 -> D: GPIO16 is HIGH
11:04:17.122 -> D: GPIO17 is HIGH
11:04:17.122 -> D: GPIO18 is HIGH
11:04:17.122 -> D: GPIO19 is HIGH
11:04:17.122 -> D: GPIO21 is HIGH
11:04:17.122 -> D: GPIO22 is HIGH
11:04:17.122 -> D: GPIO23 is HIGH
11:04:17.146 -> D: GPIO25 is HIGH
11:04:17.146 -> D: GPIO26 is HIGH
11:04:17.176 -> D: GPIO27 is HIGH
11:04:17.176 -> D: GPIO32 is HIGH
11:04:17.176 -> D: GPIO33 is HIGH
11:04:17.199 -> D: GPIO34 is LOW, probably no PULL-UP
11:04:17.199 -> D: GPIO35 is LOW, probably no PULL-UP
11:04:17.229 -> D: GPIO39 is LOW, probably no PULL-UP
11:04:17.229 -> D: gpio_00 will execute uppreset = 1
11:04:17.229 -> D: gpio_12 will execute upvolume = 2
11:04:17.229 -> D: gpio_13 will execute downvolume = 2
11:04:17.229 -> D: gpio_14 will execute stop
[... 351 lines cut ...]
11:04:34.161 -> D: SPI semaphore not taken within 2250 ticks by CPU 0, id hspec
11:04:34.161 -> D:   B -  F8F8
11:04:34.194 -> D:   C -     0
11:04:34.194 -> D: SPI semaphore not taken within 2260 ticks by CPU 0, id hspec
11:04:34.194 -> D:   D -     0
11:04:34.194 -> D:   E -     0
11:04:34.194 -> D: SPI semaphore not taken within 2270 ticks by CPU 0, id hspec
11:04:34.194 -> D:   F -     0
11:04:34.293 -> D: Nextion t2.txt="Magic Oldies Florida"
11:04:34.393 -> D: Rotation count -1
11:04:34.692 -> D: Metadata block 48 bytes
11:04:34.692 -> D: Streamtitle found, 44 bytes
11:04:34.692 -> D: StreamTitle='Beatles - A Hard Day's Night ';
11:04:34.791 -> D: Nextion t1.txt="Beatles\rA Hard Day's Night "
11:04:35.089 -> D: Rotation count -1
11:04:35.985 -> D: Rotation count -1
11:04:36.383 -> D: Rotation count -1
11:04:36.781 -> D: Rotation count -1
11:04:37.710 -> D: Rotation count -1
11:04:37.909 -> D: Rotation count -1
11:04:38.108 -> D: Rotation count -1
11:04:38.607 -> D: Rotation count -1
11:04:39.204 -> D: Rotation count -1
11:04:39.603 -> D: Rotation count -1
> ...

Edzelf, 2019-01-03:
Nextion is probably interfering with DCS of the VS1053 (both use pin 16).  Use DUMMYTFT if you have no dispaly.

MakeySQ, 2019-01-03:
In my case
 - ESP32 board PIN 32 <> VS1053b module DCS
 - Insert capacitor(470uF). VIN - GND

Edzelf, 2019-01-03:
Switching to DUMMYTFT should change the log.

3zra, 2019-01-03:
OK, sorry for being a bit sloppy. Of course, switching to DUMMYFIT changed the log in that respect (e.g. see log 19:03:35.278). 

But it didn't affect the initial SPI issues 
1. when initializing the VS1053b module (see log 19:03:39.098 ff.) and 
2. when playing a radio stream (e.g. see log 19:03:49.647 ff. or after automatically switching to the next station see 19:04:19.035 ff.). 

Thanks for your support.
3zra

Here is an updated log:
> 19:03:35.278 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 04 Oct 2018 07:22:32 GMT.  Free memory 295812
19:03:35.278 -> D: Display type is DUMMYTFT
19:03:35.278 -> D: Partition nvs found, 20480 bytes
19:03:35.278 -> D: Read 41 keys from NVS
19:03:35.278 -> D: pin_ir set to -1
19:03:35.278 -> D: pin_enc_clk set to -1
19:03:35.278 -> D: pin_enc_dt set to -1
19:03:35.278 -> D: pin_enc_sw set to -1
19:03:35.278 -> D: pin_tft_cs set to 15
19:03:35.278 -> D: pin_tft_dc set to 2
19:03:35.278 -> D: pin_tft_scl set to -1
19:03:35.278 -> D: pin_tft_sda set to -1
19:03:35.278 -> D: pin_tft_bl set to -1
19:03:35.278 -> D: pin_tft_blx set to -1
19:03:35.278 -> D: pin_sd_cs set to -1
19:03:35.278 -> D: pin_vs_cs set to 5
19:03:35.278 -> D: pin_vs_dcs set to 32
19:03:35.278 -> D: pin_vs_dreq set to 4
19:03:35.278 -> D: pin_shutdown set to -1
19:03:35.278 -> D: pin_shutdownx set to -1
19:03:35.278 -> D: pin_spi_sck set to 18
19:03:35.278 -> D: pin_spi_miso set to 19
19:03:35.278 -> D: pin_spi_mosi set to 23
19:03:35.278 -> D: GPIO0 is HIGH
19:03:35.278 -> D: GPIO2 is LOW, probably no PULL-UP
19:03:35.520 -> D: GPIO4 is HIGH
19:03:35.520 -> D: GPIO5 is HIGH
19:03:35.520 -> D: GPIO12 is HIGH
19:03:35.520 -> D: GPIO13 is HIGH
19:03:35.520 -> D: GPIO14 is HIGH
19:03:35.520 -> D: GPIO15 is HIGH
19:03:35.520 -> D: GPIO16 is HIGH
19:03:35.520 -> D: GPIO17 is HIGH
19:03:35.520 -> D: GPIO18 is HIGH
19:03:35.520 -> D: GPIO19 is HIGH
19:03:35.520 -> D: GPIO21 is HIGH
19:03:35.520 -> D: GPIO22 is HIGH
19:03:35.520 -> D: GPIO23 is HIGH
19:03:35.520 -> D: GPIO25 is HIGH
19:03:35.520 -> D: GPIO26 is HIGH
19:03:35.520 -> D: GPIO27 is HIGH
19:03:35.520 -> D: GPIO32 is HIGH
19:03:35.520 -> D: GPIO33 is HIGH
19:03:35.520 -> D: GPIO34 is LOW, probably no PULL-UP
19:03:35.520 -> D: GPIO35 is LOW, probably no PULL-UP
19:03:35.520 -> D: GPIO39 is LOW, probably no PULL-UP
19:03:35.520 -> D: gpio_00 will execute uppreset = 1
19:03:35.520 -> D: gpio_12 will execute upvolume = 2
19:03:35.520 -> D: gpio_13 will execute downvolume = 2
19:03:35.520 -> D: gpio_14 will execute stop
19:03:35.520 -> D: gpio_17 will execute resume
19:03:35.520 -> D: gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
19:03:35.520 -> D: Start display
19:03:35.520 -> D: Create list with acceptable WiFi networks
19:03:35.520 -> D: Added my-WLAN67 to list of networks
19:03:35.520 -> D: Added SSID2 to list of networks
19:03:35.585 -> D: End adding networks
19:03:35.713 -> D: Scan Networks
19:03:38.699 -> D: Scan completed
19:03:38.699 -> D: Number of available networks: 15
[... 587 lines cut ...]
19:04:21.228 -> D: SPI semaphore not taken within 2280 ticks by CPU 0, id hspec
19:04:21.228 -> D:   0 -   808
19:04:21.228 -> D:   1 -    40
19:04:21.228 -> D: SPI semaphore not taken within 2290 ticks by CPU 0, id hspec
19:04:21.261 -> D:   2 -     0
19:04:21.261 -> D:   3 -     0
19:04:21.261 -> D: SPI semaphore not taken within 2300 ticks by CPU 0, id hspec
19:04:21.261 -> D:   4 -    2E
19:04:21.261 -> D:   5 -  AC45
19:04:21.261 -> D: SPI semaphore not taken within 2310 ticks by CPU 0, id hspec
19:04:21.261 -> D:   6 -     0
19:04:21.261 -> D:   7 -     0
19:04:21.261 -> D: SPI semaphore not taken within 2320 ticks by CPU 0, id hspec
19:04:21.294 -> D:   8 -     0
19:04:21.294 -> D:   9 -     0
19:04:21.294 -> D: SPI semaphore not taken within 2330 ticks by CPU 0, id hspec
19:04:21.294 -> D:   A -     0
19:04:21.294 -> D:   B -  F8F8
19:04:21.294 -> D: SPI semaphore not taken within 2340 ticks by CPU 0, id hspec
19:04:21.294 -> D:   C -     0
19:04:21.294 -> D:   D -     0
19:04:21.294 -> D: SPI semaphore not taken within 2350 ticks by CPU 0, id hspec
19:04:21.294 -> D:   E -     0
19:04:21.294 -> D:   F -     0
> ...

3zra, 2019-01-04:
Hi,

No progress yet. 

Is there a simple way to test the hardware, e.g. the VS1053b module? 

Best, 3zra

Edzelf, 2019-01-04:
If uyou have a simple logic analyzer, you may check the SPI signals.

3zra, 2019-01-16:
Thank you for your reply and support.

I tried to redo your measurement:
[image]
It seems, that the Master signals SCK(red), MOSI(green) and XCS(yellow) match yours, resulting in the same decoded data from MOSI. Only MISO signal coming from the slave differs from yours. But if I zoom in (1V/div), it still looks pretty similar to yours, but at a much lower voltage scale. 

If I low-pass filter (200 kHz) the signal to get rid of the spikes, I even can decode the same MISO data:
[image]

Any idea, where the voltage scaling problem of MISO may come from?

I again checked the wiring to confirm that the Pins are connected to the correct Pins of the VS1053b as listed in the datasheet. I also updated the software via the web interface - no changes.

Cheers, 3zra.

Edzelf, 2019-01-16:
Looks like the internal pull-up does not work on MISO pin 19.  You may try to add an external pull-up.

3zra, 2019-01-17:
I tried the pull-up resistor - still no success.

Here are the first traces acquired when booting-up the ESP32 with pull-up at MISO (SCK-red, MOSI-green, MISO-blue XCS-yellow):
[image]
[image]
[image]

Now, MISO trace has changed, but still having the SPI errors in the log (see below) and no sound output. 

Is there anything else I may try?

Thanks in advance, 
3zra

> 20:23:40.690 -> ets Jun  8 2016 00:22:57
20:23:40.690 -> 
20:23:40.690 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:23:40.690 -> configsip: 0, SPIWP:0xee
20:23:40.690 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:23:40.690 -> mode:DIO, clock div:1
20:23:40.690 -> load:0x3fff0018,len:4
20:23:40.690 -> load:0x3fff001c,len:928
20:23:40.724 -> ho 0 tail 12 room 4
20:23:40.724 -> load:0x40078000,len:9280
20:23:40.724 -> load:0x40080400,len:5848
20:23:40.724 -> entry 0x40080698
20:23:41.056 -> 
20:23:41.056 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 04 Oct 2018 07:22:32 GMT.  Free memory 290820
20:23:41.056 -> D: Display type is DUMMYTFT
20:23:41.056 -> D: Partition nvs found, 20480 bytes
20:23:41.089 -> D: Read 41 keys from NVS
20:23:41.089 -> D: pin_ir set to -1
20:23:41.089 -> D: pin_enc_clk set to -1
20:23:41.089 -> D: pin_enc_dt set to -1
20:23:41.089 -> D: pin_enc_sw set to -1
20:23:41.089 -> D: pin_tft_cs set to 15
20:23:41.089 -> D: pin_tft_dc set to 2
20:23:41.089 -> D: pin_tft_scl set to -1
20:23:41.089 -> D: pin_tft_sda set to -1
20:23:41.089 -> D: pin_tft_bl set to -1
20:23:41.089 -> D: pin_tft_blx set to -1
20:23:41.089 -> D: pin_sd_cs set to -1
20:23:41.089 -> D: pin_vs_cs set to 5
20:23:41.089 -> D: pin_vs_dcs set to 32
20:23:41.089 -> D: pin_vs_dreq set to 4
20:23:41.089 -> D: pin_shutdown set to -1
20:23:41.123 -> D: pin_shutdownx set to -1
20:23:41.123 -> D: pin_spi_sck set to 18
20:23:41.123 -> D: pin_spi_miso set to 19
20:23:41.123 -> D: pin_spi_mosi set to 23
20:23:41.123 -> D: GPIO0 is HIGH
20:23:41.123 -> D: GPIO2 is LOW, probably no PULL-UP
20:23:41.123 -> D: GPIO4 is HIGH
20:23:41.155 -> D: GPIO5 is HIGH
20:23:41.155 -> D: GPIO12 is HIGH
20:23:41.155 -> D: GPIO13 is HIGH
20:23:41.198 -> D: GPIO14 is HIGH
20:23:41.198 -> D: GPIO15 is HIGH
20:23:41.198 -> D: GPIO16 is HIGH
20:23:41.253 -> D: GPIO17 is HIGH
20:23:41.253 -> D: GPIO18 is HIGH
20:23:41.253 -> D: GPIO19 is HIGH
20:23:41.253 -> D: GPIO21 is HIGH
20:23:41.253 -> D: GPIO22 is HIGH
20:23:41.253 -> D: GPIO23 is HIGH
20:23:41.253 -> D: GPIO25 is HIGH
20:23:41.292 -> D: GPIO26 is HIGH
20:23:41.292 -> D: GPIO27 is HIGH
20:23:41.292 -> D: GPIO32 is HIGH
20:23:41.292 -> D: GPIO33 is HIGH
[... 105 lines cut ...]
20:23:54.233 -> D: Connected to server
20:23:54.233 -> D: Duration mp3loop 1
20:23:54.266 -> D: SPI semaphore not taken within 120 ticks by CPU 0, id hspec
20:23:54.266 -> D: SPI semaphore not taken within 130 ticks by CPU 0, id hspec
20:23:54.266 -> D: SPI semaphore not taken within 140 ticks by CPU 1, id hspec
20:23:54.266 -> D: SPI semaphore not taken within 150 ticks by CPU 1, id hspec
20:23:54.300 -> D: SPI semaphore not taken within 160 ticks by CPU 1, id hspec
20:23:54.300 -> D: SPI semaphore not taken within 170 ticks by CPU 1, id hspec
20:23:54.300 -> D: SPI semaphore not taken within 180 ticks by CPU 1, id hspec
20:23:54.300 -> D: Switch to HEADER
20:23:54.338 -> D: Headerline: Content-Type: audio/mpeg
20:23:54.338 -> D: audio/mpeg seen.
20:23:54.338 -> D: Headerline: icy-br:192
20:23:54.338 -> D: Headerline: ice-audio-info: samplerate=48000;channels=2;bitrate=192D: SPI semaphore not taken within 190 ticks by CPU 1, id hspec
20:23:54.338 -> D: Headerline: icy-br:192
20:23:54.338 -> 
20:23:54.338 -> D: Headerline: icy-genre:Talk
20:23:54.338 -> D: Headerline: icy-metadata:1
20:23:54.338 -> D: Headerline: icy-name:NPO Radio1
20:23:54.338 -> D: Headerline: e not takD: SPI semaphore not taken within 200 ticks by CPU 1, id hspec
20:23:54.436 -> D: Headerline: icy-url:http://www.radio1.nl
20:23:54.436 -> 
20:23:54.436 -> D: Headerline: Server: Icecast 2.4.0-kh8
20:23:54.436 -> D: Headerline: Cache-Control: no-cache, no-store
>20:23:54.436 -> D: Headerline: Accet taken within 210 ticks bD: SPI semaphore not taken within 210 ticks by CPU 1, id hspec

3zra, 2019-01-20:
Hi Ed,

Sorry for bothering you again. Since my SPI signals resemble the ones you provided from your working system I'm wondering why I'm still getting SPI errors in the log and no audio output. 

To exclude that the similar similar signals are just a lucky coincidence it would be helpful to see your logic analyzer traces of the first non-zero MISO response. So, if its not a too big issue for you ... .

Best, 3zra

Edzelf, 2019-01-21:
When I trigger on XCS going low, I see after about 120 msec:
[image]
After that you should see more data during the SPI test:
[image]
You should see the same data for write and read:
[image]

3zra, 2019-01-21:
Hi Ed,

Thanks a lot for your measurement. 

I tried the same set-up (SCK-red, MOSI-green, MISO-blue XCS-yellow, triggering on falling edge of XCS) but, unfortunately, I get different results in several attempts: Either no MISO-response, ...
[image]
or a wrong response (the bits read don't mirror the bits written before)...
[image]
or even a nonsense response (a high response during the instruction and address section of a word) ... .
[image]

Moreover, communication stops after 6x32bits and starts again ~10 ms later:
[image]
I guess, this is related to the failure of the single SPI read/write tests.

Somehow the feeling manifests that my VS1053 module is broken. Do you agree or do the measurements provide you with some insight that is hidden to me?

Cheers, 3zra

Edzelf, 2019-01-22:
The wrong MISO data is reflected in the logging by:
`20:23:44.980 -> D: VS1053 error retry SB:0708 R1:FFFE R2:85FF`
It means: date Should Be 0708, Read 1 returns FFFE and Read 2 returns 85FF.
The test stops after 20 errors.  The pauses in the SPI communication (10 msec) you see is caused by the time necessary to print the error message.
So it looks like a faulty VS1053 or ESP32 or a wiring error.

3zra, 2019-01-22:
Ok, if we exclude the wiring why do you think also a faulty ESP32 could explain this behavior? To me all the signals coming from the ESP32 look reasonable.

3zra

Edzelf, 2019-01-22:
Simple, it's about a WIRE, connected between an ESP32 and a VS1053.  So there are 3 possible points of failure....

t-c-o-d-e, 2019-04-29:
@3zra did you ever figured out? I'm having similar problem with mine.

Robiv8, 2019-05-03:
So I have the same Problem!
I also do not understand why in the ino file stay this:
// GPIO32 pin 1 XDCS
// GPIO5 pin 2 XCS
// GPIO4 pin 4 DREQ
// GPIO18 pin 5 SCK
// GPIO19 pin 7 MISO
// GPIO23 pin 6 MOSI
// EN pin 3 XRST

And in the config file of the Webpage only this:
pin_vs_cs = 5 # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 32 # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4 # GPIO Pin number for VS1053 "DREQ"

Edzelf, 2019-05-04:
The other 4 signals  ( sck, miso, mosi and reset) are fixed.

3zra, 2019-06-21:
Sorry for the late reply. Finally, I wanted to share what solved my SPI communication problems.

The problems disappeared when switching from a jumper wired setup for initial testing to a soldered circuit board setup. From that I derive that the problems must have been due to a faulty jumper wire. 

Thank you for your support.

closed 2019-06-21 by 3zra