Edzelf/ESP32-Radio #212: Rotary encoder and volume

Milka-Cow, 2019-01-05:
Tks Ed for this nice project.

My son copied the software to the ESP (had some problem at beginning) and I did the hardware on a breadboard.
Now our prototype works fine and stable.

But there is one issue with the rotary encoder.
Normally to increase volume at each amplifier or radio cassette recorder rotary encoder must be turned to the right side.

But at this project we have to turn rotary to the left side that volume increase!
This is a little bit strange.
Is there a possibilty to change it?

Tks for help
Joe

Rainer-G, 2019-01-05:
Hi Joe,

it may help if you swap data and clock, either on the breadboard or in the pin definitions!

Good luck, Rainer

Milka-Cow, 2019-01-05:
Hi Rainer
Yes this worked!
I never used that switch before, so I'm not familar with the function.
Tks again
Joe

closed 2019-01-05 by Milka-Cow