Edzelf/ESP32-Radio #213: LCD --> at station name last character missing

Milka-Cow, 2019-01-05:
Hi Edzelf community,

the rotary switch problem about volume I have solved, but closed issue by mistake. No idea whether I deleted it. Maybe the admin can undelete it.

I have a further question.

LCD --> At station name last character is missing. Only "Easy Hits Florid" in display is shown. --> "a" is missing.
I copy paste this from Ed's pdf to webservers config for test.
preset_01 = airspectrum.cdnstream1.com:8114/1648_128 # 1 - Easy Hits Florida 128k

I have this problem also with other stations. Sometimes the last charcter is missing or I have a B there!
preset = 3
preset_00 = swr-swr3-live.cast.addradio.de:80/swr/swr3/live/mp3/128/stream.mp3 # 0 - SWR3
preset_01 = swr-swr1-bw.cast.addradio.de:80/swr/swr1/bw/mp3/128/stream.mp3 # 1 - SWR1
preset_02 = br-br3-live.cast.addradio.de:80/br/br3/live/mp3/128/stream.mp3 # 2 - BR3
preset_03 = swr-swr4-bw.cast.addradio.de:80/swr/swr4/bw/mp3/128/stream.mp3 # 3 - SWR4
preset_04 = mp3channels.webradio.antenne.de:80/antenne # 4 - ANTENNE

Any idea?
Tks for help
Jo

[image]
[image]
[image]

Edzelf, 2019-01-05:
Maybe there are too much characters in the string.  Try to remove the ":80", for port 80 is the default.

Milka-Cow, 2019-01-06:
Good morning

It also doesn't work if I remove :80

>>Station button on pcb
gpio_12 = station = swr-swr3-live.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3
>>Station button at control tool
preset_00 = swr-swr3-live.cast.addradio.de/swr/swr3/live/mp3/128/stream.mp3 # 0 - SWR3

But please look on the display.
Also on top left side at --> ESP32-Radi --> th "o" is missing too

Do you think the BUS is to long from ESP32 to the TFT on the right side?
It is about 140mm!

Edzelf, 2019-01-06:
I use a cable of 20 cm to connect the display without problems. So it cannot be the length of the SPI bus wiring.
Could you try to do a software update from the webinterface config page?  Then we can make sure that it is not a compile problem.

Milka-Cow, 2019-01-06:
Cool!
Problem solved
Thank you!!

Edzelf, 2019-01-06:
Well, not solved completely...