Edzelf/ESP32-Radio #214: How to play .m3u in google drive or dropbox?

MakeySQ, 2019-01-09:
Dear
Thanks for great project!!

I want to play .m3u in google drive or dropbox?
How to ?

Edzelf, 2019-01-09:
I am not familiar with this stuff, but If the drive is accessible via an URL it should work.
The radio can not connect to a Windows drive.

tokio14-dot, 2025-10-30:
subiendo la película a Dropbox 
y ese mismo link ponerlos en una estructura en github 

estructura:

#EXTINF:-1 tvg-logo="IMAGEN DE PELICULA"group-title="GENERO DE LA PELICULA",NOMBRE DE PELÍCULA 
https://dl.dropbox.com/scl/fi/fpxuhx3d58q6p6b1t57ba/nadie-nos-vio-pertir-2025.txt?rlkey=7mcoi4foau4emm8vllcp987mg&st=mpsagcp5&dl=0

a los enlaces de Dropbox se le cambia el www. por dl.

en github tenés que ir a raw y copiar el link ponerlo en REPRODUCTOR VLC

o otros REPRODUCTORES