Edzelf/ESP32-Radio #218: no as smooth as esp8266 ...something has gone wrong..?

SurenBono, 2019-01-10:
HI Ed,
Im Suren from Malaysia, recently i studied some of your work on webradio... i have done working on the D1 R1 on VS1053B and it works well....but later i managed to get a ESP32 R32 board with  another VS1053B for  testing the esp32 version.. but it is not as smooth as the esp8266.... the R32 is well compatible with the Esp32 Dev Board ....yesterday i managed to read some of the issues before seeking githubs assistance where the DEFAULTED #define NEXTION was cousing constant sys reboot..well after undefining it and choosing the #define OLED worked but it was choppy sounding... i thought mybe undefining #define TFTFILE would leech less and  but it turn out error on upload...please help

[image]

SurenBono, 2019-01-10:
> Undefine TFTFILE should not have any effect. You may switch of the display
> by defining DUMMYTFT.

Do i need to define the dummytft for the OLED display option..?

Edzelf, 2019-01-10:
No, but if there are problems, you may switch off the display in order to find the problem.

Edzelf, 2019-01-10:
Please show the debug output.

SurenBono, 2019-01-10:
i tried switching the display to OLED ... also without  ...the result same...here is my debug output... i cant copy & paste the output on the serial so i took a snapshot... please check Doctor Ed..

[image]

Edzelf, 2019-01-10:
Looks like there is something wring with gpio19.  It should be high during the initial test.

SurenBono, 2019-01-10:
.. the SPI pins...? what is the open close state..?...what should i do to fix it..?

SurenBono, 2019-01-10:
..do i need to pullup with 3v3 manualy or ..is it a board problem/or coding error.?

SurenBono, 2019-01-10:
...or do i need to increase the buffer size..?

Edzelf, 2019-01-11:
Could you check the behaviour of gpio19 (miso) with a scope or a logic analyzer?  It looks like the signal is inverted.  Maybe you can use a different pin for miso.
You also might try to do a software update from the comfig page of the webinterface.  This may rule out compiling errors.

SurenBono, 2019-01-11:
..i had similar choppy problems on esp8266 but fixed it with the v2higher bandwidth from the options...but esp32 didnt have similar options for higher bandwidth..probably in future release....regarding gpio19 ...i do not have a logic analyser..and the miso on vs1053b is directly connected to the esp32 board...sometimes it is smooth on one preset..sometimes not...and it skips selected presets automaticaly sometimes...some preset persist the same choppy playback as in esp8266..probably the way it is streamed...but still esp8266 were smoother/cheaper/easier than the esp32... after a week of observation....

Edzelf, 2019-01-11:
In my setup the ESP32 is faster and more stable than the 8266.  Did you try one of my sugestions?

SurenBono, 2019-01-11:
I did try flashing undefining any display...but it still lagged...When you flash into the eeprom...does it include the firmware flashing...or is the firmware separated from the sketch..

Edzelf, 2019-01-11:
The firmware is in ROM.

Edzelf, 2019-01-11:
It is impossible to erase ROM memory.  Please try thesoftware update from the config page in the webinterface.  Maybe this will give some information.  It only takes one minute....

SurenBono, 2019-01-12:
..than why did espressif release the AT firmware and the esp32 flasher..the documentation for the adress is not there

SurenBono, 2019-01-12:
Its official esp32 sucks....im discontinuing purchasing it or recommending it in the future..at least esp8266 can be installed on xp

SurenBono, 2019-01-12:
..sorry edzelf on esp32 failed..its cumbersome and still failed

SurenBono, 2019-01-14:
...recently i re-checked the project, i found that the freertos is not loaded up entirely before it is played...and when esp32 is playing it is not stocking on to the buffer but it played earlier than the stable esp8266 and pauses because the buffer is emptied,.refills a bit..and play ..and pause ..i doubt the 2 core is working as it should..and i never came across 32bit with 2core..the other core cant work simultaneously...regarding the gpio19.. it was high without the vs1053 connected....but i really do appreciate your programming skills...i wish i can write something like this someday ...

[image]

SurenBono, 2019-01-14:
...I found your old more simpler version...this one lesser choppier than the current ones...the oled worked to....RINGBUFFER STILL ROCKS

[image]

SurenBono, 2019-01-14:
[image]

SurenBono, 2019-01-24:
.....i have solved a little of the problem for choppy play on esp32 by switching to a fiberoptic isp ......it has solved most of those usual choppy stream on copper wire on certain higher bitrate preset ...but i have not test it to run 24/7 like i did on esp8266 yet...but still i cant get the 128x32OLED to display the ip (blankscreen)..it is supposed to be a 128x64 like advertised when i purchased it...its different from the broadband ip .....only could get ip through the IDE port...i also tried to upload the patches for vs1053b from the VSLI HOMEPAGE... (vs1053b-patches260) BY #include vs1053b-patches.c ..not sure if it did what it supposed to...and modified the web interface... and the mp3 file played at last but couldnt display a proper filename yet (fat32)

[image]

SurenBono, 2019-01-25:
...Edzelf ESP32 Webradio Sroto&Gargees Version is now tested on offgrid PWM Solar Charge Controller...it will run until 12v15a battery low power disconnects it at 11.4v as safety for battery longevity unless i get a higher Amp hour battery and higher Wattage Solar panel (car lead acid)..to make it run forever ...HiFi Amp powered by grid tied...trying to get em run both offgrid in the near future... Thanks again Edzelf for this awesome project that you gave to this world ...there was time that i was frustrated for some reason, i slept it off...  zzzzzzzzzz

[image]

closed 2019-02-09 by SurenBono