Edzelf/ESP32-Radio #222: Problem connecting TFT SPI 1.8 128x160

cherob67, 2019-01-15:
I need some help. Thanks for your answer.

I am using the PCB. The ESP32-Radio working fine without the TFT module. When I connect the TFT, all is good on the beginning (VS1053 testing, Wifi connexion, MQTT initialisation and synchro TOD) and informations are correctly displayed on the TFT. 

Immediatly after, there is an error and the ESP reboot :
 

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 04 Oct 2018 07:22:32 GMT.  Free memory 238504
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 44 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to 18
D: pin_tft_sda set to 23
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: Start display
D: Create list with acceptable WiFi networks
D: Added Prc to list of networks
D: Added domicile_rc to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 6
D:  1 - domicile_rc               Signal: -78 dBm, Encryption WPA_WPA2_PSK, Acceptable
D:  2 - Bbox-B476B850             Signal: -83 dBm, Encryption WPA_WPA2_PSK, 
D:  3 - freebox_CeliaSXB          Signal: -91 dBm, Encryption WPA_PSK, 
D:  4 - XXXX-1698                 Signal: -92 dBm, Encryption WPA_WPA2_PSK, 
D:  5 - Bbox-0CFE9D25             Signal: -94 dBm, Encryption WPA_WPA2_PSK, 
D:  6 - rodgers25                 Signal: -94 dBm, Encryption WPA_WPA2_PSK, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter europe.pool.ntp.org
D: Command: mqttbroker with parameter broker.hivemq.com
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttpubtopic with parameter P_espradioIP
D: Command: mqtttopic with parameter P_espradio
D: Command: mqtttpreset with parameter none
D: Command: mqttuser with parameter none
D: Command: preset with parameter 0
D: Command: preset_00 with parameter chai5she.cdn.dvmr.fr:80/franceinter-midfi.mp3
D: Command: preset_01 with parameter chai5she.cdn.dvmr.fr:80/francemusique-midfi.mp3
D: Command: preset_02 with parameter chai5she.lb.vip.cdn.dvmr.fr/francemusiqueclassiqueplus-hifi.mp3
D: Command: preset_03 with parameter chai5she.lb.vip.cdn.dvmr.fr/francemusiqueconcertsradiofrance-hifi.mp3
D: Command: preset_04 with parameter direct.francemusique.fr/live/francemusiquelevenementielle-hifi.mp3
D: Command: preset_05 with parameter chai5she.lb.vip.cdn.dvmr.fr/francemusiquelacontemporaine-hifi.mp3
D: Command: preset_06 with parameter chai5she.lb.vip.cdn.dvmr.fr/francemusiquelajazz-hifi.mp3
D: Command: preset_07 with parameter chai5she.lb.vip.cdn.dvmr.fr/francemusiqueocoramonde-hifi.mp3
D: Command: preset_08 with parameter chai5she.cdn.dvmr.fr:80/franceculture-midfi.mp3
D: Command: preset_09 with parameter live02.rfi.fr/rfimonde-96k.mp3
D: Command: preset_10 with parameter cdn.nrjaudio.fm/audio1/fr/40043/aac_64.mp3
D: Command: preset_11 with parameter rfm-live-mp3-128.scdn.arkena.com/rfm.mp3
D: Command: preset_12 with parameter str0.creacast.com/frequence_verte_2
D: Command: preset_13 with parameter str0.creacast.com/rec
D: Command: preset_14 with parameter www.maxifrance.fr/stream128.m3u
D: Command: preset_15 with parameter mp3lg4.tdf-cdn.com/9240/lag_180945.mp3
D: Command: preset_16 with parameter radioclassique.ice.infomaniak.ch/radioclassique-high.mp3
D: Command: preset_17 with parameter streaming.radio.rtl.fr/rtl-1-48-192
D: Command: preset_18 with parameter direct.fipradio.fr/live/fipstrasbourg-midfi.mp3
D: Command: preset_19 with parameter 149.202.74.135:8078/live
D: Command: preset_20 with parameter jazzradio.ice.infomaniak.ch/jazzradio-high.mp3
D: Command: preset_21 with parameter jazz-wr02.ice.infomaniak.ch/jazz-wr02-128.mp3
D: Command: preset_22 with parameter jazzblues.ice.infomaniak.ch/jazzblues-high.mp3
D: Command: preset_23 with parameter radiojudaicastrasbourg.ice.infomaniak.ch/radiojudaicastrasbourg-128.mp3
D: Command: preset_24 with parameter ecoutez.radio-libertaire.org:8080/radiolib.m3u
D: Command: preset_25 with parameter www.classicfm.nl/player/classicfm.m3u
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 15
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 88
D: Slow SPI, Testing VS1053 read/write registers...
D: endFillByte is 0
D: Connect to WiFi
D: Connected to domicile_rc
D: IP = 192.168.1.11
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MQTT uses prefix P25A4AE30
D: Init MQTT
D: MDNS responder started
D: Rotary encoder is disabled (-1/-1/-1)
D: Sync TOD
D: Sync TOD, new value is 15:00:119-21:37:40
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert faileD: d!
abort() was called at PC 0x4008ca29 on core 0

Backtrace: 0x4008fcd8:0x3ffbd800 0x4008fedb:0x3ffbd820 0x4008ca29:0x3ffbd840 0x400d2df5:0x3ffbd880 0x400d4721:0x3ffbd8a0 0x400d47be:0x3ffbd8c0

Rebooting...
ets Jun  8 2016 00:22:57

Edzelf, 2019-01-15:
There is something wrong with the TOD (time of day) setting. Try to remove the settings for the time server.

cherob67, 2019-01-16:
That 's right. I don't understand why...perhaps it is a library problem in my configuration ?

closed 2019-01-16 by cherob67

Edzelf, 2019-01-16:
Try a different time server.

cherob67, 2019-01-18:
Now it's working fine. 
Best regards