Edzelf/ESP32-Radio #223: Suggestion: support for the ESP32-A1S

henkiejan1, 2019-01-16:
Hi Ed,

I didn't know if you knowing there is a special ESP32 for audio functions. It has a AC101 audio decoder inside. There are also very complete development boards like the AI Thinker audio development kit and it cost arround the €15,-. In my opinion a very good price for a real complete product.

I have ordered a AI Thinker audio developement kit with ESP32-A1S. It has a lot of function. But not standard a display. But you can easily add one. :)

See for more info [this link. ](https://blog.hackster.io/seeed-drops-new-esp32-audio-development-kit-for-audio-related-iot-projects-ad38d1f02637)

Edzelf, 2019-01-16:
Yes, looks good.  I hope there is some documentation on how to use the AC101 for mp3 decoding.

Edzelf, 2019-01-17:
Reading the documentation I found out that the audiochip does not decode MP3.  This is done in software, so high bitrates may cause a problem.

arlaor, 2019-02-05:
> Reading the documentation I found out that the audiochip does not decode MP3. This is done in software, so high bitrates may cause a problem.

Hi @Edzelf 

then what would be the function of the AC101 chip?

Edzelf, 2019-02-05:
Digital to analog conversion, volume control, filtering.
I just received the esp32 audio development module.  No success in putting it to work...

henkiejan1, 2019-02-10:
Little bit sad that it could not be work. The board has all the hardware needs what i see integrated. 

But what i also see is that Espressif also have there own equalboard but with other components. The Espressif LyraT board. [Here is al link to the official Espressif site. ](https://www.espressif.com/en/products/hardware/esp32-lyrat) One of the difference is that it used a external ES8388 codec chip. And that there is more documentation what i see than about the ESP32-A1S. [Here is a lot information. ](https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/board-esp32-lyrat-v4.3.html) i do not own this board but maybe also a idea?

Edzelf, 2019-02-10:
As far as I know: the ES8388 is not an MP3 decoder.

aguynamedben, 2019-03-08:
There's an abandoned PR from a year ago to get the AC101 working with esp-adf: https://github.com/espressif/esp-adf/pull/11

So sad 😭could have been the perfect audio dev board. Are y'all using LyraT? Does it "just work" with esp-adf?

Edzelf, 2019-03-08:
I finally got my A1S board to work with the  play_mp3 demo.  I followed https://github.com/phkehl/esp32-a1s-audio_hal.

arlaor, 2019-03-08:
Hi @Edzelf
So is possible implement Esp32 Radio in a1s module?

Edzelf, 2019-03-08:
Maybe, but I think that software MP3 decoding is too slow for the ESP32-Radio project.
There is a sample program that reads a MP3 file from https://dl.espressif.com/dl/audio/ff-16b-2c-44100hz.mp3 (128 kbps) and plays it.  That works correctly.  Sound is good.
Note that the codec used for Internet Radio is not always MP3, but sometimes AAC.
I would say: give it a try!

humberto-trujillo, 2019-03-08:
@Edzelf I was able to get things working fine. Thank you!
Just one observation though. When compiling there was a conflict inside the 'dueros_service' under the components directory in esp-adf because of the use of the AUDIO_HAL_ES8388_DEFAULT macro. I just had to add that in your audio_hal.h file, then it compiled just fine.

Edzelf, 2019-03-09:
You have to replace "AUDIO_HAL_ES8388_DEFAULT" by "AUDIO_HAL_AC101_DEFAULT".

henkiejan1, 2019-03-27:
Very interesting! :) My experience with the ESP32-A1S with mp3 on a SD card and the default demo software is that the sound quality was really good! You could download also a sample firmware from the AI Thinker site. https://wiki.ai-thinker.com/esp32-a1s on the bottem of the site. Tip: use google chrome because the whole site is in Chinese.

henkiejan1, 2019-06-11:
I has a little bit time so i started to experimented with my AI Thinker board. I have now for a long time the AAC stream of Q-music Netherlands playing without interrupts with the example ai_examples/pipeline_living_stream . Off course with the default example so i cant tune up the volume. But sound quality is very good. I have configured in make menuconfig for now the CPU on 240MHz. I used the modified ESP-ADF from https://github.com/donny681/esp-adf

I like to try more :)

henkiejan1, 2019-06-11:
And BTW id didn;t know if you saw it. But there is a datasheet in your link for the AC101 decoder. https://github.com/phkehl/esp32-a1s-audio_hal/blob/master/doc/x-powers-AC101-codec-datasheet.pdf

henkiejan1, 2019-06-25:
And it even run on 80MHz without problems. 

`rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5540
load:0x40078000,len:0
load:0x40078000,len:13752
entry 0x40078fb0
I (28) boot: ESP-IDF v3.0.1 2nd stage bootloader
I (28) boot: compile time 23:00:04
I (28) boot: Enabling RNG early entropy source...
I (33) boot: SPI Speed      : 80MHz
I (37) boot: SPI Mode       : DIO
I (41) boot: SPI Flash Size : 4MB
I (45) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (82) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1de9c (122524) map
I (127) esp_image: segment 1: paddr=0x0002dec4 vaddr=0x3ffb0000 size=0x0214c (  8524) load
I (130) esp_image: segment 2: paddr=0x00030018 vaddr=0x400d0018 size=0x88b88 (560008) map
I (296) esp_image: segment 3: paddr=0x000b8ba8 vaddr=0x3ffb214c size=0x01104 (  4356) load
I (298) esp_image: segment 4: paddr=0x000b9cb4 vaddr=0x40080000 size=0x00400 (  1024) load
I (303) esp_image: segment 5: paddr=0x000ba0bc vaddr=0x40080400 size=0x0beac ( 48812) load
I (329) esp_image: segment 6: paddr=0x000c5f70 vaddr=0x400c0000 size=0x00000 (     0) load
I (337) boot: Loaded app from partition at offset 0x10000
I (337) boot: Disabling RNG early entropy source...
I (339) cpu_start: Pro cpu up.
I (342) cpu_start: Starting app cpu, entry point is 0x40081228
I (0) cpu_start: App cpu up.
I (353) heap_init: Initializing. RAM available for dynamic allocation:
I (360) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (366) heap_init: At 3FFB91A0 len 00026E60 (155 KiB): DRAM
I (372) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (378) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (385) heap_init: At 4008C2AC len 00013D54 (79 KiB): IRAM
I (391) cpu_start: Pro cpu start user code
I (184) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (286) HTTP_MP3_EXAMPLE: [ 1 ] Start audio codec chip
I (1406) HTTP_MP3_EXAMPLE: [2.0] Create audio pipeline for playback
I (1406) HTTP_MP3_EXAMPLE: [2.1] Create http stream to read data
I (1406) HTTP_MP3_EXAMPLE: [2.2] Create i2s stream to write data to codec chip
I (1416) HTTP_MP3_EXAMPLE: [2.3] Create mp3 decoder to decode mp3 file
I (1416) HTTP_MP3_EXAMPLE: [2.4] Register all elements to audio pipeline
I (1426) HTTP_MP3_EXAMPLE: [2.5] Link it together http_stream-->mp3_decoder-->i2s_stream-->[codec_chip]
I (1436) HTTP_MP3_EXAMPLE: [2.6] Setup uri (http as http_stream, mp3 as mp3 decoder, and default output is i2s)
I (1446) HTTP_MP3_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network
I (1496) phy: phy_version: 3662, 711a97c, May  9 2018, 14:29:06, 0, 0
W (3756) PERIPH_WIFI: Wi-Fi disconnected from SSID MijnWifi, auto-reconnect enabled, reconnect after 1000 ms
I (5726) HTTP_MP3_EXAMPLE: [ 4 ] Setup event listener
I (5726) HTTP_MP3_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (5726) HTTP_MP3_EXAMPLE: [4.2] Listening event from peripherals
I (5736) HTTP_MP3_EXAMPLE: [ 5 ] Start audio_pipeline
I (6286) HTTP_MP3_EXAMPLE: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2`

Technikfreak2002, 2019-07-13:
@henkiejan1 How did you compile this? I have set up everything yesterday on a Linux Mint VM and could not compile any example of the modified adf without errors....

Is there any progress anyway?

Thank You!

henkiejan1, 2019-07-13:
Just with the ESP-ADF framework and the examples. Not with ESP32-Radio. You just need to read every step carefully.