Edzelf/ESP32-Radio #224: Switching of Channels

gcharles81, 2019-01-27:
Hello I would like to know if there is any part of the code that do automatically switch to next saved channels if current playing channel for any reason stop playing.

I am experiencing channel changes , first I pointed out that it might be related to wrong pullups defined for the  pre defined buttons , but I still experienced the same channel switching after I disabled all defined inputs.

Right now I have only encoder defined but still channels switch by themselves below is the current printout of my defined IO:

clk_dst = 1                                          # Offset during daylight saving time (hours)
clk_offset = 1                                       # Offset with respect to UTC in hours
clk_server = pool.ntp.org                            # Time server to be used
#
ir_40BF = upvolume = 2
ir_C03F = downvolume = 2
#
mqqprefix = none
mqttbroker = none
mqttpasswd = *******
mqttport = 1883
mqttuser = none
#
pin_enc_clk = 26                                    # GPIO Pin number for rotary encoder "CLK"//10 new chga
pin_enc_dt = 25                                      # GPIO Pin number for rotary encoder "DT"//9new chga
pin_enc_sw = 27                                      # GPIO Pin number for rotary encoder "SW"//11new chga
pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
pin_sd_cs = 22                                       # GPIO Pin number for SD card "CS"
pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"
pin_vs_cs = 17                                        # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 16                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"
#
preset = 10
preset_00 = rpclassichits.ice.infomaniak.ch/rpclassichits-128.mp3  #  0 - Pilatus Classic
preset_01 = Playing rpinthemix.ice.infomaniak.ch/rpinthemix-128.mp3  #  1 - Pilatus Mix
preset_02 = radiopilatus.ice.infomaniak.ch/pilatus64.mp3  #  2 - Radio Pilatus
preset_03 = stream.srg-ssr.ch/m/rsp/mp3_128          #  3 - Radio Swiss Pop
preset_04 = sunshineradio.ice.infomaniak.ch/sunshineradio-128.mp3 #  4 - Radio Sunshine Zug
preset_05 = rpcharts.ice.infomaniak.ch/rpcharts-128.mp3  #  5 - Pilatus Charts
preset_06 = s10.voscast.com:8344                     #  6 - Kiss FM Malta
preset_07 = stream.srg-ssr.ch/m/reteuno/mp3_128      #  7 - RSI Rete Uno
preset_08 = nr5.newradio.it:8520                     #  8 - Radio Studio Star
preset_09 = 94.23.66.155:8106                        #  9 - *ILR CHILL and GROOVE
preset_10 = 198.178.123.23:8128                      # 10 - iHeartRadio IHR_IEDM
preset_11 = 198.178.123.23:8128                      # 11 - iHeartRadio IHR_TRAN
#
toneha = 2
tonehf = 15
tonela = 2
tonelf = 4
#
volume = 60

Edzelf, 2019-01-27:
The radio switches to the next channels if the input stream stops for some reason.  You can change this behaviour in one of the timer functions.

Edzelf, 2019-01-27:
Yes, but it will probably not start as the stream was interrupted.  You may change the number of seconds that define the time-out.

gcharles81, 2019-02-16:
thanks infact on some occasions the stream will remain stopped since it was interrupted 
would the datamode set to INIT  restart the stream if it is added instead of the  ini_block.newpreset++ ;

Edzelf, 2019-02-16:
You can try the INIT trick.  But it will stop the Radio if there is a problem with the stream.

Edzelf, 2019-02-16:
You may also increase the 10 seconds time-out.

gcharles81, 2019-02-16:
That did the trick it works flawlessly now i did the below - Thanks 

      {
       // datamode = STOPREQD ;                    
       // ini_block.newpreset++ ;                 
           datamode = INIT;		//added for same channel reset
      }

gcharles81, 2019-02-24:
Hello I wanted to clarify with you where exactly can the 10sec  timeout be extended because I can see it is only call from the 100ms timer every 30 seconds according to the below - Is that right ?

 if ( ++count10sec == 300  )                     // 30 seconds passed? 
  {
    timer10sec() ;                                // Yes, do 10 second procedure
    count10sec = 0 ;                              // Reset count
  }

Edzelf, 2019-02-24:
Yes, that's the one.