Edzelf/ESP32-Radio #225: Testing VS1053 read/write reigsters creates errors

netrosec, 2019-01-30:
@Edzelf I'm running with minimal hardware (No TFT or SD module) everything thing compiles and loads fine until it starts testing read/write registers. I checked and triple d checked my connections, changed wires and so on but still no change. am i missing something?
```
0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 06 Aug 2018 12:12:32 GMT.  Free memory 233176
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 45 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 21
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: gpio_14 will execute stop
D: gpio_17 will execute resume
D: gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
D: Enable pin 35 for IR
D: Start display
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:471] ff_sd_initialize(): GO_IDLE_STATE failed
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added (:) to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 2
D:  1 - (:)                       Signal: -69 dBm, Encryption WPA2_PSK, Acceptable
D:  2 - Go and sub you dirty pig  Signal: -87 dBm, Encryption WPA2_PSK, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: gpio_14 with parameter stop
D: Command: gpio_17 with parameter resume
D: Command: gpio_34 with parameter station = icecast.omroep.nl:80/radio1-bb-mp3
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: mqqprefix with parameter none
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter 21
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 16
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 7
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 0
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 error retry SB:1518 R1:13EC R2:13EC
D: VS1053 error retry SB:4D58 R1:4D20 R2:4D58
D: VS1053 error retry SB:9E34 R1:9E34 R2:B400
D: VS1053 error retry SB:AD70 R1:E400 R2:AD70
D: VS1053 error retry SB:C864 R1:C738 R2:C738
D: VS1053 error retry SB:D9F8 R1:D9F8 R2:AE00
D: VS1053 error retry SB:F168 R1:F168 R2:C600
D: Connect to WiFi
D: Try WiFi (:)
D: Connected to (:)
D: IP = 192.168.43.170
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 03:15:24
D: STOP requested
D: New preset/file requested (7/0) from 205.164.62.15:10032
D: Connect to new host 205.164.62.15:10032
D: Connect to 205.164.62.15 on port 10032, extension /
D: Connected to server
D: Duration mp3loop 1
D: Switch to HEADER
D: Headerline: icy-br: 32
D: Headerline: icy-pub: 1
D: Headerline: icy-description: 1.FM - Radio Gaia
D: Headerline: icy-audio-info: bitrate=32;samplerate=44100;channels=2
D: Headerline: icy-url: http://1.fm
D: Headerline: Instance-id: 6a2790434342f31ec41656a33c34b2d2
D: Headerline: Cache-Control: no-cache
D: Headerline: Server: AIS Streaming Server 7.7.2
D: Headerline: icy-genre: Chill
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint: 8192
D: Headerline: Pragma: no-cache
D: Headerline: icy-name: 1.FM - Radio Gaia
D: Headerline: Connection: close
D: Headerline: Content-Type: audio/aacp
D: audio/aacp seen.
D: Switch to DATA, bitrate is 32, metaint is 8192
D: Duration mp3loop 53
D: Metadata block 48 bytes
D: Streamtitle found, 33 bytes
D: StreamTitle='DEEP FOREST - TIKO';
D: Song stopped incorrectly!
D: REG   Contents
D: ---   -----
D:   0 -   808
D:   1 -    40
D:   2 -     0
D:   3 -     0
D:   4 -     0
D:   5 -  1F40
D:   6 -     0
D:   7 -     0
D:   8 -     0
D:   9 -     0
D:   A -     0
D:   B -  F8F8
D:   C -     0
D:   D -     0
D:   E -     0
D:   F -     0
D: Duration mp3loop 885
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
Metadata block 32 bytes
D: Streamtitle found, 28 bytes
D: StreamTitle='';StreamUrl='';
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Rotation count -1
D: Metadata block 64 bytes
D: Streamtitle found, 48 bytes
D: StreamTitle='ENYA - STORMS IN AFRICA (PART II)';
D: Rotation count -1
D: Rotation count -1

```

Edzelf, 2019-01-30:
See issue 100.  People reported the inversion of pin 15.  You may use a different pin for CS.  You may also try to do a software update from the config page of the webinterface.  This will rule out compiling- and library errors.

netrosec, 2019-01-30:
@Edzelf I'm afraid i do not fully understand "the inversion of pin 15". I'm not using pin 15 at all. What pin could i use for CS?

Edzelf, 2019-01-30:
Sorry, I mean pin 5.  Maybe you can measure this pin.  It should be HIGH most of the time.La  Same for pin 16.  Did you try the software update?

netrosec, 2019-01-30:
@Edzelf Debug output shows gpio 5 and 16 are both high. I ran the software update and now when it gets to the  read/write registers it says vs1053 not properly installed..

Robiv8, 2019-05-05:
i have the same issue!
is there a solution?

D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 32
D: Command: pin_vs_dreq with parameter 4

``` D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 SPI error. SB:012C R1:0000 R2:0000
D: VS1053 SPI error. SB:0258 R1:0000 R2:0000
D: VS1053 SPI error. SB:0384 R1:0000 R2:0000
D: VS1053 SPI error. SB:04B0 R1:0000 R2:0000
D: VS1053 SPI error. SB:05DC R1:0000 R2:0000
D: VS1053 SPI error. SB:0708 R1:0000 R2:0000
D: VS1053 SPI error. SB:0834 R1:0000 R2:0000
D: VS1053 SPI error. SB:0960 R1:0000 R2:0000
D: VS1053 SPI error. SB:0A8C R1:0000 R2:0000
D: VS1053 SPI error. SB:0BB8 R1:0000 R2:0000
D: VS1053 SPI error. SB:0CE4 R1:0000 R2:0000
D: VS1053 SPI error. SB:0E10 R1:0000 R2:0000
D: VS1053 SPI error. SB:0F3C R1:0000 R2:0000
D: VS1053 SPI error. SB:1068 R1:0000 R2:0000
D: VS1053 SPI error. SB:1194 R1:0000 R2:0000
D: VS1053 SPI error. SB:12C0 R1:0000 R2:0000
D: VS1053 SPI error. SB:13EC R1:0000 R2:0000
D: VS1053 SPI error. SB:1518 R1:0000 R2:0000
D: VS1053 SPI error. SB:1644 R1:0000 R2:0000
D: VS1053 SPI error. SB:1770 R1:0000 R2:0000
```

Tony763, 2019-05-23:
For me, gpio 5 for VS is running OK, but TFT cs I have to use gpio 27 instead of 15 and for SD cs gpio 22 instead of 21. So You can try these two.

Robiv8, 2019-05-26:
Sorry, but my ili9341 TFT is working great, only the VS1053 have the error.

Tony763, 2019-05-26:
Instead of 5 try pin 27 or 22 as cs pin for VS.

Robiv8, 2019-05-26:
Aaah! OK Thank you very much.
Sorry for the stupid question, and where do I have to change that in the code? That's so many :-(

Tony763, 2019-05-26:
Add `pin_vs_cs = 27` to your configuration via web interface  or in ESP32_radio.ino  find `struct iosetting klist[] = {                            // List of I/O related keys` on line 2762  - there you can change all pins.

Beware! Settings in configuration override settings in code.

Robiv8, 2019-05-26:
Great! Many thanks. I will try

Robiv8, 2019-05-26:
YOU are a GENIUS
Many thanks, it works on pin 27

xtrinch, 2021-01-31:
> Instead of 5 try pin 27 or 22 as cs pin for VS.

Thanks. Do you guys think it would make sense to add this to the docs? I am using an ILI9341 and it was not functioning when CS for VS1053 was pin 5. On pin 22 it works perfectly.

xP373Rx, 2021-02-05:
pins 15 and 5   outputs PWM signal at boot