Edzelf/ESP32-Radio #227: Suggestion: integrate bitluni's AM radio transmitter driver

bootrino, 2019-01-30:
bitluni wrote code to transmit AM radio from an ESP32 without any additional components at all.

http://bitluni.net/am-radio-transmitter/

Any chance you could inetgrate bitluni's driver into this project?

Edzelf, 2019-01-31:
I don't think it's gonna work:
1 - The signal to modulate the carrier is not available in the radio.
2 - The ESP32 is used for handling the WiFi, the MP3 decoder and the display.  That will result in interruptions of the modulation.
3 - It is not allowed to broadcast on the AM radio band.
4 - You may attach a FM transmitter to the output of the VS1053 for the same, if not better, results.

bootrino, 2019-01-31:
Thanks for the update.  That's a pity - bitluni's code works well but not being able to play mp3 or WAV is very limiting.

Can you explain more this comment please?
>>> 1 - The signal to modulate the carrier is not available in the radio.

Edzelf, 2019-01-31:
The radio receives a compressed stream with MP3, AAC or other codec.  This is decoded by the VS1053, not by the ESP32.  You need an uncompressed stream to modulate a AM transmitter.