Edzelf/ESP32-Radio #230: Request: Will there be support for the ESP32-Wrover Module and/or SPIRAM Flash 32 Mbit

SinglWolf, 2019-02-21:
Hi!
Will there be support for the ESP32-Wrover Module and/or SPIRAM Flash 32 Mbit?
This ESP32-Wrover Module (or a separate SPIRAM chip) allows you to organize a playback buffer for up to 8 seconds. Very useful for bad internet connection. If you plan to enable SPIRAM support, let us know.
Give thanks.

Edzelf, 2019-02-24:
I do not have a module with SPRAM, so I can't test it.  However, there is normally some 200 KB feer memory available, so why not increase the queue.  See line 185:
#define QSIZ 400
You may try a setting of 5000.

SinglWolf, 2019-02-24:
Thanks for the answer. I ordered a module with additional memory on Aliexpress. I will check your recommendation and be sure to report the results.

SinglWolf, 2019-05-18:
Good day!
Fulfilled your recommendations
```
#define QSIZ 5000
.....
dataqueue = xQueueCreate(QSIZ, // Create queue for communication
                           sizeof(qdata_struct));
......
```

and got a cyclic reboot
```

........
D: Sync TOD
D: Sync TOD, new value is 08:52:02
top/Et() /ESP32/esp-idf-p 0l4c/cdmpd on s/freert/components/freertos/queue.c:1442 (xQueueGenericReceUse)- ess/fi fto/De!
Backtrace: 0x40090c50:0x3ffcf920 0x40090e81:0x3ffcf940 0x4008d77d:0x3ffcf960 0x400d35ff:0x3ffcf9a0 0x4008cb7d:0x3ffcf9c0

Rebooting...
ets Jun  8 2016 00:22:57
.....
```
Increasing this parameter is a bad idea. Because it sets the maximum number of items that can be stored in the queue at any time.
Need a dynamic buffer, I think. I do not know how to implement it. Not enough knowledge, alas.

Edzelf, 2019-05-18:
A queue IS a synamic buffer.  But I agree that a big buffersize will not solve sync problems.

SinglWolf, 2019-05-18:
Please see [psram in wrover kit](https://github.com/espressif/arduino-esp32/issues/1644) discussion. It may be an idea how to use external SPI RAM in your project.

SinglWolf, 2019-05-18:
[Arduino library to play MOD, WAV, FLAC, MIDI, RTTTL, MP3, and AAC files on I2S DACs or with a software emulated delta-sigma DAC on the ESP8266 and ESP32](https://github.com/earlephilhower/ESP8266Audio)
Please see this project. There is clearly external SPIRAM support for ESP32. Namely files
`src/AudioFileSourceSPIRAMBuffer.h`
 and
 `src/AudioFileSourceSPIRAMBuffer.cpp`
.

Edzelf, 2019-05-18:
Yes, but what is the point?  There's no real memory shortage.

SinglWolf, 2019-05-18:
I would like to have a large buffer for the audio stream with a bad Internet channel.
There is the Ka-Radio32 project, where external memory is used and determined automatically. But the framework ESP-IDF is not available for my understanding. With Arduino, I still somehow cope. I can not impose my opinion on you. Do not want to implement support for psram is your right. Thanks for answers.