Edzelf/ESP32-Radio #235: Daylight saving

compujan66, 2019-03-19:
Hi, running the espradio with succes. ;-)

clk_dst = 1  
clk_offset = 1
clk_server =  time.kpnnet.org   
Give's one hour later as the real time in the Netherlands... Is the dst correction manual? can't find the code to switch automatic....  Something for the todo list?

Grtz Jan (doit esp32, geeetech mp3 shield, 0.96inch oled, pam8403)

Edzelf, 2019-03-20:
clk_dst has maybe something to do with this.  I leave it to manual, because DST will probably disappear...

Sera26, 2019-03-22:
Hello,

same problem in Hungary also.
Time was correct at the beginning, but now I already have one hour plus.

Edzelf, 2019-03-22:
Change your settings for clk_*

Sera26, 2019-03-22:
OK, I'll check it. Great Thanks for your support.

Rainer-G, 2019-03-24:
Hi,
the ntp server only supplies UTC ( formerly Greenwich Mean Time ) and does not switch to DST. 
This is the responsibility of operating system / application program. e.g. Windows always saves the time in UTC and the regional settings tell what time to display and whether DST should be automatic or manual.

I wonder whether the TimeLib does something similar, so I set clk_dst to zero and will wait until the change happens! My ESP32-Radio is up 7x24, so it will not miss the transition to DST.

I'll keep you informed,
Rainer