Edzelf/ESP32-Radio #236: OLED1306-Missing characters in FONT and Draw Bitmap

DaTTcz, 2019-03-25:
I have a problem displaying META TAG from an internet radio MP3 stream. LCD OLED does not display characters because it is missing in FONT. (like šščřžýáíéúů ....) And next, I can't add any picture to the display at X,Y position. (Draw Bitmap array). Missing code in SSD1306.h for this. This can be added ?

Edzelf, 2019-03-25:
You may replace the SSD1306.h file.  My version can only display ASCII characters from 0x20 to 0x7E.

DaTTcz, 2019-03-25:
To add more characters, just add more characters in the array in SSD1306.h ? Or is there anything else to change in the code?

Edzelf, 2019-03-25:
Yes, you must add them to the array.  I do not know if the special characters are coded in one byte.  If so, it would not be too difficult to change the code.  Just increase the range.