Edzelf/ESP32-Radio #242: VS1053 does not work - no sound

jroggenbuck, 2019-04-18:


jroggenbuck, 2019-04-18:
My ESP32 (MCU 30 pin) will not connect with VS1053 with SD Card. Both hardware parts did work in other projects.
see config page, debug page and wiring attached.
__________________________________________________________________________

config

clk_dst = 1
clk_offset = 1
clk_server = pool.ntp.org
#
gpio_00 = uppreset = 1
gpio_12 = upvolume = 2
gpio_13 = downvolume = 2
#
ir_40BF = upvolume = 2
ir_C03F = downvolume = 2
#
mqttbroker = none
mqttpasswd = *******
mqttport = 1883
mqttprefix = none
mqttuser = none
#
pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder CLK
pin_enc_dt = 26                                     # GPIO Pin number for rotary encoder DT
pin_enc_sw = 27                                     # GPIO Pin number for rotary encoder SW
pin_ir = 35                                     # GPIO Pin number for IR receiver VS1838B
pin_sd_cs = 15                                     # GPIO Pin number for SD card CS
pin_spi_miso = 19                                      # GPIO Pin number for SPI Miso
pin_spi_mosi = 23                                      # GPIO Pin number for SPI Mosi
pin_spi_sck = 18                                       # GPIO Pin number for SPI Clock
pin_tft_scl = 22                                     # GPIO Pin number for TFT CS
pin_tft_sda = 21                                      # GPIO Pin number for TFT DC
pin_vs_cs = 5                                      # GPIO Pin number for VS1053 CS
pin_vs_dcs = 16                                     # GPIO Pin number for VS1053 DCS
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 DREQ
#
preset = 6
preset_00 = mp3channels.webradio.antenne.de/antenne  #  0 - Bayern Antenne
preset_01 = airspectrum.cdnstream1.com:8114/1648_128 #  1 - Easy Hits Florida 128k
preset_02 = us2.internet-radio.com:8050              #  2 - CLASSIC ROCK MIA WWW.SHERADIO.COM
preset_03 = airspectrum.cdnstream1.com:8000/1261_192 #  3 - Magic Oldies Florida
preset_04 = airspectrum.cdnstream1.com:8008/1604_128 #  4 - Magic 60s Florida 60s Classic Rock
preset_05 = us1.internet-radio.com:8105              #  5 - Classic Rock Florida - SHE Radio
preset_06 = icecast.omroep.nl:80/radio1-bb-mp3       #  6 - Radio 1, NL
preset_07 = 205.164.62.15:10032                      #  7 - 1.FM - GAIA, 64k
preset_08 = skonto.ls.lv:8002/mp3                    #  8 - Skonto 128k
preset_09 = 94.23.66.155:8106                        #  9 - *ILR CHILL and GROOVE
preset_10 = ihr/IHR_IEDM                             # 10 - iHeartRadio IHR_IEDM
preset_11 = ihr/IHR_TRAN                             # 11 - iHeartRadio IHR_TRAN
#
toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
#
volume = 84
#
wifi_00 = WLAN-******/*******
wifi_01 = SSID2/*******

----------------------------------------------------------------------------------------------------------------------------
Debug

 11:26:11.101 -> ets Jun  8 2016 00:22:57
11:26:11.101 -> 
11:26:11.101 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
11:26:11.101 -> configsip: 0, SPIWP:0xee
11:26:11.101 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:26:11.101 -> mode:DIO, clock div:1
[... 222 lines cut ...]
Wire

ESP32	Signal	Display Oled	VS1053	Note	SDCard	Note
GPIO32	__	__	Pin 7	DCS	__	__
GPIO5	__	__	Pin 6	CS	__	__
GPIO4	__	__	Pin 2	DREQ	__	__
GPIO22	SCL	I2C Clock	__	__	__	__
GPIO21	SDA	I2C Data	__	__	__	
GPIO16	RDX2	__	__	__	__	TX Nexion
GPIO17	TXD2	__	__	__	__	RX Nexion
GPIO18	SCK	__	Pin 13	SCK	__	__
GPIO19	MISO	__	Pin 12 	MISO	__	__
GPIO23	MOSI	__	Pin 11	MOSI	__	__
GPI15	CS	__	Pin 9	CS	CS	 für SD Card
GPIO3	RXD0	__	__	__	__	Res Seriel Inp
GPIO1	TXD0	__	__	__	__	Res Seriel Outp
GPIO35	IR	__	__	__	__	Infrared Receiver VS1838B
GPIO25	CLK	__	__	__	__	Rotary Encoder CLK
GPIO26	DT	__	__	__	__	Rotary Encoder DT
GPIO27	SW	__	__	__	__	Rotary Encoder SW
_____________________________________________________________________________________________						
GND	x	x		x		
VCC 5 V	x	x		x		
VCC 3 V						
EN	XRST		Pin 8

boerti71, 2019-04-18:
I have the same problem, but it's radio is working when I remove the SD card and rebbot the ESP

jroggenbuck, 2019-04-18:
Thank you, but that did not help....

Edzelf, 2019-04-18:
Will the SD card mount without the VS1053?

jroggenbuck, 2019-04-18:
The SD Card is part of the VS1053

Edzelf, 2019-04-18:
But uses different pins.

Edzelf, 2019-04-18:
I have a version without sd.  I use a separate sd card interface.

jroggenbuck, 2019-04-19:
I tried two different shields. One is from Geeetech with onboard SD, the other is from LC Tech without SD. No change. 
Do I understand you right, that you have a software sketch without SD support? How can I test that?

Edzelf, 2019-04-19:
No, there is just one version.  I meant that I have the VS1053 without SD slot.
In you case it looks like the SPI bus is corrupted by the SD card.  There is probably something wrong with the GPIO 15.

jroggenbuck, 2019-04-19:
Change config from pin 15 to pin 2
see attached picture pin 2 during startup, looks strange...

[image]

sarahjaneuk, 2019-04-23:
Not sure if it is of any help and I was getting the exact same range of VS1053 errors at start up (SB 0000 to 1644 returning the same values) and by a process of elimination (changing each part one by one, including the wires) discovered it was a faulty ESP32 board though just swapped it out rather than spent time finding out what the specific issue was.
Hope this helps.

jroggenbuck, 2019-04-23:
I changed shield vs1053 and MCU twice, checked cables, check config. I load the MCU with different internet radio software. 
https://www.instructables.com/id/Internet-Radio-Using-an-ESP32/
But this software is not userfriendly, so I like to run the software that ED created,,,

sarahjaneuk, 2019-04-23:
One thing I did do, separate from this project, is to download the VS1053 library (https://github.com/baldram/ESP_VS1053_Library). This contains an example of a web radio that I was able to get working. It is very basic though it is an easy way to check the ESP32 and VS1053 boards.

jroggenbuck, 2019-04-28:
Thank you, but I think I will give up the project Internet Radio ....

closed 2019-04-28 by jroggenbuck