Edzelf/ESP32-Radio #243: SPI semaphore errors when fail to get time from NTP server

sarahjaneuk, 2019-04-23:
Hi Ed and everyone

I am in the process of building the radio using the following parts:

ESP32 board - https://www.aliexpress.com/item/ESP-32-ESP-32S-Development-Board-4M-Memory-ESP32-Board-WiFi-Bluetooth-Ultra-Low-Power/32964407902.html
VS1053 board - https://www.aliexpress.com/item/1PCS-New-VS1053-MP3-Module-Development-Bard-on-board-recording-function/32354143509.html
Blue TFT - https://www.aliexpress.com/item/Smart-Electronics-1-8-Inch-128-160-Serial-SPI-TFT-LCD-Module-Display-PCB-Adapter-Power/32967441734.html

My connections are as standard in the radio manual.

Initially I was getting a lot of the VS1053 errors at boot up though discovered that this was a faulty ESP32 board so replaced this with another (same model) and this is now sorted. These were for SB values 0000 to 1644 and each of R1 and R2 showing FFFF.

The radio now works intermittently, which seems to depend on the ability to get the time from the NTP server. Namely, if at boot up, the time is obtained ok, the radio starts correctly and plays the last station.  If however the time cannot be obtained, it plays for a few seconds and then drops out with the 'SPI semaphore not taken within xxx ticks by CPU 0, id chunk'.  It seems a bit strange that this should be the case and I appreciate that there may be a link between not obtaining the time and the issues with SPI, though a struggling to see how this can be as I assume it will not been the SPI interface to get data from a server using the wifi connectivity on the board.

I have found that if I stop the radio using the web interface, the errors stop which is what I would expect.  The code continues to attempt to get the time as can be seen from the log below and if I start the radio again from the web interface after the time has been obtained, the radio then plays the stream ok.

......
D: STOP requested
D: Stopping client
D: SPI semaphore not taken within 3220 ticks by CPU 0, id chunk
D: SPI semaphore not taken within 3230 ticks by CPU 0, id chunk
D: SPI semaphore not taken within 3240 ticks by CPU 0, id chunk
D: SPI semaphore not taken within 3250 ticks by CPU 0, id chunk
D: SPI semaphore not taken within 3260 ticks by CPU 0, id chunk
D: SPI semaphore not taken within 3270 ticks by CPU 0, id chunk
D: SPI semaphore not taken within 3280 ticks by CPU 0, id chunk
--- deleted ---
D: SPI semaphore not taken within 4890 ticks by CPU 0, id chunk
D: SPI semaphore not taken within 4900 ticks by CPU 0, id chunk
D: SPI semaphore not taken within 4910 ticks by CPU 0, id chunk
D: SPI semaphore not taken within 4920 ticks by CPU 0, id chunk
D: SPI semaphore not taken within 4930 ticks by CPU 0, id chunk
D: SPI semaphore not taken within 4940 ticks by CPU 0, id chunk
D: SPI semaphore not taken within 4950 ticks by CPU 0, id chunk
D: SPI semaphore not taken within 4960 ticks by CPU 0, id chunk
D: Failed to obtain time!
D: Sync TOD
D: Failed to obtain time!
D: Sync TOD
D: Failed to obtain time!
D: Sync TOD
D: Failed to obtain time!
D: Sync TOD
D: Failed to obtain time!
D: Sync TOD
D: Failed to obtain time!
D: Sync TOD
D: Failed to obtain time!
D: Sync TOD
D: Sync TOD, new value is 14:13:10
D: Command client available
D: handlehttp started
.....

I have spent some time looking at the code (as I am a developer by profession) though am struggling to see why this would be related to the time server.

I have included a log from this process.

So bit of a loss.  Any ideas would be very welcome.

[20190422_ESPRadioLog1.txt](https://github.com/Edzelf/ESP32-Radio/files/3107811/20190422_ESPRadioLog1.txt)

Edzelf, 2019-04-23:
The timeserver should have no major effects on the radio.  Could you try a different station, because I saw a very high rate of metadata in the logging that may cause the tft to overrun.

sarahjaneuk, 2019-04-23:
Thanks for your reply, Ed. This log was from starting up with preset 01. I have tested it with startup on many of the 11 stations. It still behaves the same.
I agree, the timeserver poll is probably a symptom rather than the likely cause, though cannot see what would be affecting it.
By the way, I assume the stop command being issued when going to the config page on the web interface, is deliberate?
Thanks.

Edzelf, 2019-04-23:
Yes, playing stops in the config page.
You may try to change the GPIO used for the CS of the display.  Sometimes certain GPIO's act strange on certain boards.  You may even try to use the dummy TFT.

sarahjaneuk, 2019-04-23:
Hi Ed
I spent most of yesterday looking at this. I went through other issues here and tried things like taking out the TFT.
When I changed the screen to DUMMYTFT, which did initially solve the problem, which, once I commented out the TFT check in the gettime() method (to force it to get the time even if there is no screen to display it), the original semaphore issue came back again on the occasions when the radio booted and failed to get the time from the NTP server.

sarahjaneuk, 2019-04-23:
Hi again Ed.
One thing I thought of that I didn't give a go yesterday, was to see what happens if the clock is not used. I notice that there is a global variable time_req that is triggered in the timer100() interrupt method. So I commented out the line setting this to true so the clock is not used. Everything else (including the original screen - BLUETFT) is as before. And it works fine.
So crazy as it seems, as I know it does not make sense, it does look like the call to the NTP server is causing the issue. (I am using the standard pool.ntp.org.)

Edzelf, 2019-04-23:
It certainly looks like a problem with the NTP server.  But I cannot reproduce it here.  I'm also using the pool.ntp.org for months.

sarahjaneuk, 2019-04-23:
Hi yet again, Ed!
I have taken the testing deeper and got to a point where I have commented out the if statement in the gettime() method that calls getLocalTime() and replaced it with if (1 == 1) - that will force the gettime() method to fail to get the time without actually calling getLocalTime().
When I do this, the radio works fine with no semaphore errors.
So it seems if it is something to do with the actual getting of the time from the NTP servers.
I cannot see how this would be an issue for the SPI interface. Any ideas would be welcome.
Thanks.

Edzelf, 2019-04-23:
Maybe getLocalTime () takes a long time or will not return.

sarahjaneuk, 2019-04-23:
Could it really be the time it takes the NTP server to respond that is causing the SPI semaphore issues?
I cannot see a connection between the wifi connectivity and the SPI interface.

sarahjaneuk, 2019-04-23:
Hi Ed
Have studied the code a bit more and how the semaphores are used, I think I now understand what the issue is and what is causing it.
The call to gettime() is within the higher priority semaphore task (spftask) (through the handle_spec() method) and when the get of the time from the NTP server takes longer than expected (when it fails), it is blocking calls to the lower priority semaphore (playtask) (through xSemaphoreTake call in the claimSPI() method), hence the SPI semaphore errors because it is unable to get the next chunks of data from the stream, causing the player to effectively stop until the get time attempt in the spftask completes and allows the calls to the playtask semaphore to succeed again.
I will have a think about how to get around this.
Yes, I could just stop the call to the NTP server (as it does not seem to be essential) and I would rather a more elegant solution that allows the NTP server calls to continue in the background without affecting preventing data for the stream to be obtained.
Would be interested in your thoughts on this.

Edzelf, 2019-04-24:
Good work!  I've made a change to the software.  Now "gettime()" is called outside the SPI semaphore.  Thank you for your investigation.

sarahjaneuk, 2019-04-24:
Thanks Ed.
I actually discovered that the issue was caused because the pool.ntp.org domain only has a few servers available for here in Ireland, which is why they are so slow sometimes polling it,  I switch to specifically use the UK servers and the issue improved.  Eventually, I realised that I will probably only use the radio I am building at the moment here, so am now using my own local NTP server setup on my Linux server here.
However, it was a good thing to discover and work around, as it is always possible someone else may encounter the same issue in the future.

This is a fab project and I am so grateful to you for building it and maintaining it.

I would like to build quite a few features into the software, if that is ok.
First up, I would like the option to be able to use a static local IP address when connecting to the wifi (mainly as it makes it easier to know what IP to use for the web interface).

Edzelf, 2019-04-24:
Okay.  You may use your router to make a fixed IP-address.  DHCP will then recognize the MAC address and set the fixed IP for the device.
You may also use the hostname "esp32radio.local" in your browser.  For this feature you may have to install additional software on your PC ("Bonjour", Apple's implementation of zero-configuration networking).

sarahjaneuk, 2019-04-24:
Thanks Ed.
DNS or just editing the hosts file will do the trick on the client end.
I'll close the issue now. Thanks for your help.