Edzelf/ESP32-Radio #245: NFC support for MP3 urls

rdehuyss, 2019-04-28:
Is it possible to have an NFC reader which then maps NFC codes to certain MP3 files stored on a NAS?

Edzelf, 2019-04-28:
I think you can build that easily.  Specially if you can store the mp3 files on a 32 GB SD card.  You need a rfid reader, an Arduino nano, a dfplayer and 2 speakers.  For files on a NAS you need an ESP32 with MP3 decoder.

koskee, 2019-04-28:
I actually really like that idea. Maybe you got it elsewherre, maybe not, but that seems like a really elegant way of communicating.

Perhaps you could even find a way to make it bidirectional (maybe depending if the phone has something playing or not?)

One of the extra features I added to my radio is I made the artist display box on my web interface clickable, which would take the artist/song name and automatically pop up a new browser window and run a Google search of the song to see if it is available for d/L. This is SUPER convenient if I hear something I like and want to save it for later. 
This also potentially opens the possibility of drive-by song changes, say if you had a song you were streaming, and liked. You could simply place your phone against the speaker to instantly start playback on the radio of the same song by pushing the url onto the device over NFC and then having the player access the stream (assuming its available online and assuming the player is set to allow such actions) 

Perhaps we could also be able to instantly transfer wifi credentials to the radio as well, but I think you'd have to contend with some android/apple security API's to access that info probably. 

Either way, that's a really good sounding set of features that could really help to set the project apart from others, I think.