Edzelf/ESP32-Radio #246: Another SPI/VS1053B problem - no audio output

t-c-o-d-e, 2019-04-28:
First of all, many thanks to the author(s), this project is badass!! I'm and will learn tons from it as I'm quite new in the ESP and micro controller world but with some history in programming.

Okay, into the problem. Already read all issues related to SPI errors on startup, none of solutions worked for me and I'm very sorry for opening another one, but I'm trying to get sound out of VS1053B for couple of days with partial luck. 

This is [ESP32 ](https://www.ebay.com/itm/Espressif-ESP32-WLAN-Dev-Kit-Board-Development-Bluetooth-Wifi-v1-WROOM32-NodeMCU/253059783728?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649) I'm having with this [VS1053B](https://www.banggood.com/VS1053-MP3-Module-Development-Board-On-Board-Recording-Function-p-1244003.html?rmmds=myorder&cur_warehouse=CN) board. To rule out things no displays or any other peripheral attached on ESP32 except VS1053B (using/uncommented dummy screen in code). Sketch compiled and uploaded to EPS32, managed to configure it through WEB, it successfully connect to local Wi Fi network, but during startup getting SPI errors while debugging trough Serial Monitor and no sound on output.

This is what I get:

> 18:42:06.268 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
18:42:06.268 -> configsip: 0, SPIWP:0xee
18:42:06.268 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:42:06.302 -> mode:DIO, clock div:1
18:42:06.302 -> load:0x3fff0018,len:4
18:42:06.302 -> load:0x3fff001c,len:928
18:42:06.302 -> ho 0 tail 12 room 4
18:42:06.302 -> load:0x40078000,len:8424
18:42:06.302 -> ho 0 tail 12 room 4
18:42:06.302 -> load:0x40080400,len:5868
18:42:06.302 -> entry 0x4008069c
18:42:06.642 -> 
18:42:06.642 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Wed, 24 Apr 2019 08:20:00 GMT.  Free memory 287272
18:42:06.642 -> D: Display type is DUMMYTFT
18:42:06.642 -> D: Partition nvs found, 20480 bytes
18:42:06.676 -> D: Read 46 keys from NVS
18:42:06.676 -> D: pin_ir set to 35
18:42:06.676 -> D: pin_enc_clk set to 25
18:42:06.676 -> D: pin_enc_dt set to 26
18:42:06.676 -> D: pin_enc_sw set to 27
18:42:06.676 -> D: pin_tft_cs set to 15
18:42:06.676 -> D: pin_tft_dc set to 2
18:42:06.676 -> D: pin_tft_scl set to -1
18:42:06.676 -> D: pin_tft_sda set to -1
18:42:06.676 -> D: pin_tft_bl set to -1
18:42:06.676 -> D: pin_tft_blx set to -1
18:42:06.676 -> D: pin_sd_cs set to 21
18:42:06.676 -> D: pin_vs_cs set to 5
18:42:06.676 -> D: pin_vs_dcs set to 32
18:42:06.676 -> D: pin_vs_dreq set to 4
18:42:06.709 -> D: pin_shutdown set to -1
18:42:06.709 -> D: pin_shutdownx set to -1
18:42:06.709 -> D: pin_spi_sck set to 18
18:42:06.709 -> D: pin_spi_miso set to 19
18:42:06.709 -> D: pin_spi_mosi set to 23
18:42:06.709 -> D: GPIO0 is HIGH
18:42:06.709 -> D: GPIO2 is LOW, probably no PULL-UP
18:42:06.709 -> D: GPIO4 is HIGH
18:42:06.743 -> D: GPIO5 is HIGH
18:42:06.743 -> D: GPIO12 is HIGH
18:42:06.743 -> D: GPIO13 is HIGH
18:42:06.776 -> D: GPIO14 is HIGH
18:42:06.776 -> D: GPIO15 is HIGH
18:42:06.776 -> D: GPIO16 is HIGH
18:42:06.809 -> D: GPIO17 is HIGH
18:42:06.809 -> D: GPIO18 is HIGH
18:42:06.809 -> D: GPIO19 is HIGH
18:42:06.809 -> D: GPIO21 is HIGH
18:42:06.843 -> D: GPIO22 is HIGH
18:42:06.843 -> D: GPIO23 is HIGH
18:42:06.843 -> D: GPIO25 is HIGH
18:42:06.876 -> D: GPIO26 is HIGH
18:42:06.876 -> D: GPIO27 is HIGH
18:42:06.876 -> D: GPIO32 is HIGH
18:42:06.911 -> D: GPIO33 is HIGH
18:42:06.911 -> D: GPIO34 is LOW, probably no PULL-UP
18:42:06.911 -> D: GPIO35 is LOW, probably no PULL-UP
18:42:06.911 -> D: GPIO39 is LOW, probably no PULL-UP
18:42:06.911 -> D: gpio_00 will execute uppreset = 1
18:42:06.945 -> D: gpio_12 will execute upvolume = 2
18:42:06.945 -> D: gpio_13 will execute downvolume = 2
18:42:06.945 -> D: gpio_14 will execute stop
[... 260 lines cut ...]
>     client.print(String("GET ") + path[station_no] + " HTTP/1.1\r\n" +
>                "Host: " + host[station_no] + "\r\n" + 
>                "Connection: close\r\n\r\n");                
>   }
> 
>   void connectToWIFI()
>   {
>     WiFi.begin(ssid, pass);
>       while (WiFi.status() != WL_CONNECTED) {
>         delay(500);
>         Serial.print(".");
>    }
>    Serial.println("WiFi connected");
>    delay(1500);
>   }
> 
>   void initMP3Decoder()
>   {
>     player.begin();
>     player.switchToMp3Mode(); // optional, some boards require this
>     player.setVolume(VOLUME);
>     delay(3000);
>   }

And that's where I'm all lost. What else can I check? Thanks in advance and keep up good work!

Edzelf, 2019-04-28:
There is something wrong with the SPI bus.  Maybe you have a logic analyzer.  The cheap chinese ones understand the SPI protocol.

t-c-o-d-e, 2019-04-28:
Thanks for a quick reply!

Unfortunately I don't have. But how it works with simple code from VS1053 library example, if it is any trouble with SPI or hardware related?

t-c-o-d-e, 2019-04-28:
Also tried with ESP32 Node MCU 32S, it won't work with neither code :/. So strange.. faulty VS1053 board maybe?
Should get two new VS1053 boards tomorrow and try again. This is so frustrating :(.

Rainer-G, 2019-04-29:
Hi,

I had a look at you log and the sample code you attached.
The difference I can see that your code uses SPI.begin() with no parameters, whereas ESP32Radio uses :

  SPI.begin ( ini_block.spi_sck_pin,                     // Init VSPI bus with default or modified pins
                     ini_block.spi_miso_pin,
                     ini_block.spi_mosi_pin ) ;

Running SPI.begin without parameters uses the default setting from the board file, the ESP32Radio software may override thes values. E.g. on my Adafruit Feather I’m using
D: Command: pin_spi_miso with parameter 19
D: Command: pin_spi_mosi with parameter 18
D: Command: pin_spi_sck with parameter 5

Can you check which pins of the ESP32 the VS1053 SPI pins are connected ? 
You can find the default pins in the board file or do a print with the SPI names MISO, MOSI, SCLK 
    like   Serial.println( MOSI ); etc

Best regards, Rainer

Von: t-c-o-d-e
Gesendet: Sonntag, 28. April 2019 22:03
An: Edzelf/ESP32-Radio
Cc: Subscribed
Betreff: Re: [Edzelf/ESP32-Radio] Another SPI/VS1053B problem - no audiooutput (#246)

Also tried with ESP32 Node MCU 32S, it won't work with neither code :/. So strange.. faulty VS1053 board maybe?
Should get two new VS1053 boards tomorrow and try again. This is so frustrating :(.

t-c-o-d-e, 2019-04-29:
Thank you for your insight!

Checked, serial print says: MOSI 23, MISO 19, SCK 18, 

Well, tried one more time with sample code, it plays but looks like it loses some packets/data, I'm more and more towards defective VS1053B board. Will have two new boards tonight and try.

t-c-o-d-e, 2019-04-29:
@Edzelf I've tried with two new VS1053B boards, with same result. Can you or anybody else confirm that your code works with [this ](https://www.ebay.co.uk/itm/VS1053-MP3-Module-Development-Board-w-On-Board-Recording-Function-SPI-Interface/310645105092?_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D56924%26meid%3D43a3252da51f4dda858264ba89413672%26pid%3D100675%26rk%3D3%26rkt%3D15%26mehot%3Dpp%26sd%3D323789676275%26itm%3D310645105092&_trksid=p2481888.c100675.m4236&_trkparms=pageci%3Abb02c111-6ab7-11e9-8e21-74dbd1803149%7Cparentrq%3A6aa407f516a0ac3c3bc902aafff03d25%7Ciid%3A1) VS1053 board? What I'm doing wrong :(((?

sarahjaneuk, 2019-04-29:
The green (LC Technology) VS1053 board in your link looks identical to the one I am using and I can confirm it works fine in my set up.
Hope you get yours working.

t-c-o-d-e, 2019-04-29:
@sarahjaneuk thank you!

FOUND IT!!!!  Jumper wires (15 cm) were too long! Using 5 cm ones and I get occasionally one SPI error on startup, but most of the times it went through without errors and I finally having sound on output.

But, now having totally different issue, volume going down automatically for no reason!?

t-c-o-d-e, 2019-04-30:
Figured that one too, rotary encoder should be connected or disabled in config if not connected. Otherwise volume is going down to 0 every time.

closed 2019-04-30 by t-c-o-d-e

sarahjaneuk, 2019-04-30:
Excellent. The IR route is probably a better option. I found the rotary encoder a bit too fiddly.

In case you haven't seen it, check out Andreas Speiss's video on this project: https://www.youtube.com/watch?v=hz65vfvbXMs

Enjoy your new radio.

gregoryperrin, 2019-08-09:
Thanks to your comment https://github.com/Edzelf/ESP32-Radio/issues/246#issuecomment-487953465 i found the answer to my problem (volume going down automatically)