Edzelf/ESP32-Radio #247: Permanent reboot

Milka-Cow, 2019-04-28:
Hi,
my radio works fine since 2 month or so now.
It will start and reboot permanetly.
My son programmed the ESP in the past. He said he could not find any problems.
Also the different SSD are seen if he connect it to his PC.

Starting...
Version January 05 19
Connect to Wifi
Then 192.168.4.1
Starting...

Any idea what I can do?

Edzelf, 2019-04-29:
Reboot is often caused by weak powersupply.
Please try to run the radio with a different powersupply.

Milka-Cow, 2019-04-29:
Hi
before I had a 1A psu, same problem with a quality 2A psu.
Both show 5V with voltmeter, also wiring on pcb is tick enough.

emiklen, 2019-04-29:
i had the same problem, and in my case the cause was the preset station, may be some offending icy metadata, changed preset to another station  and the reboot stopped.

Milka-Cow, 2019-05-03:
@emiklen: The solution do not helped.
My son had time now and he re programmed the chip. Then it worked fine.
So no idea what happend.

closed 2019-05-03 by Milka-Cow