Edzelf/ESP32-Radio #250: OLED Display 3.12" 256*64

t-c-o-d-e, 2019-05-03:
Found this nice OLED [display ](https://www.aliexpress.com/item/Real-OLED-Display-3-12-256-64-25664-Dots-Graphic-LCD-Module-Display-Screen-LCM-Screen/32991357565.html?spm=2114.search0604.3.112.244b2bafmSkJfw&ws_ab_test=searchweb0_0,searchweb201602_10_10065_10068_319_10059_10884_317_10887_10696_321_322_10084_453_10083_454_10103_10618_10307_10820_10301_10821_10303_537_536,searchweb201603_52,ppcSwitch_0&algo_expid=b0138b1d-2584-470a-9517-c4196839da9a-20&algo_pvid=b0138b1d-2584-470a-9517-c4196839da9a), it looks almost perfect for this project. How difficult wold be to add support for it :)?

Edzelf, 2019-05-03:
Not so difficult.  However, the number of lines is limited.

t-c-o-d-e, 2019-05-03:
That's ok, I would be happy with same number of lines as is on 128x64 OLED screen.