Edzelf/ESP32-Radio #253: how to connect esp32 to ampliffer without VS1053 ?

EvgenPavlyuchek, 2019-05-08:
I have no sound when I connect the instructions here
https://github.com/Edzelf/ESP32-Radio/issues/112

I want to try without VS1053
I connect ampliffer to GPIO 25, 26 and gnd -  no sound again
have i to change something in arduino script ?

my HELTEC WiFi Kit 32
https://ae01.alicdn.com/kf/HTB1awOch2NNTKJjSspeq6ySwpXa9.jpg
VS1053
https://ru.aliexpress.com/item/1384719110.html?productId=1384719110&productSubject=Free-Shipping-1PC-New-VS1053-MP3-Module-Development-Bard-on-board-recording-function&spm=a2g0s.9042311.0.0.274233edDRXp6J
and ampliffer PAM8403
https://ru.aliexpress.com/item/32712741429.html?productId=32712741429&productSubject=PAM8403-mini-5V-digital-amplifier-board-with-switch-potentiometer-can-be-USB-powered&spm=a2g0s.9042311.0.0.274233edK9VEL7

Edzelf, 2019-05-08:
The VS1053 is used to convert the mp3 stream to sound.  ESP32-Radio cannot convert the mp3 stream by software.

EvgenPavlyuchek, 2019-05-08:
aac must work
http://dir.xiph.org/by_format/AAC

EvgenPavlyuchek, 2019-05-08:
or can i ask to check my connection to vs1053

you said here https://github.com/Edzelf/ESP32-Radio/issues/112#issuecomment-385610186
Heltec   https://ae01.alicdn.com/kf/HTB1awOch2NNTKJjSspeq6ySwpXa9.jpg
VS1053  https://ru.aliexpress.com/item/1384719110.html?productId=1384719110&productSubject=Free-Shipping-1PC-New-VS1053-MP3-Module-Development-Bard-on-board-recording-function&spm=a2g0s.9042311.0.0.274233edDRXp6J   
 
> // Heltec   Signal        Wired to VS1053      
> // -------- ------  --------------      ----------
> // GPIO5            -                   pin 2 XCS   
> // GPIO34   -       -                 pin 1 XDCS   
> // GPIO2            -                   pin 4 DREQ   
    
> // GPIO18   SCK     -               pin 5 SCK         
> // GPIO19   MISO    -             pin 7 MISO        
> // GPIO23   MOSI    -             pin 6 MOSI            
> // EN       -       -                     pin 3 XRST    

default arduino
pin_vs_cs = 5                                          # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 32                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"

so have i to change arduino to 
pin_vs_cs = 5                                          # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 34                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 2                                      # GPIO Pin number for VS1053 "DREQ"

or change connections to 
> // GPIO5            -                   pin 2 XCS   
> // GPIO32   -       -                 pin 1 XDCS   
> // GPIO4            -                   pin 4 DREQ 

ps honestly tried on everyone - never works