Edzelf/ESP32-Radio #254: More preset stations

uwe-pt, 2019-05-14:
Hi, is there any chance to augment the number of preset stations? I would need for a project about +200. Have had a look at the code, but got lost... Help would be appreciated!

Edzelf, 2019-05-14:
The NVS data is stored in a partion called "nvs".  There is a way to change the size of this partition (default 20 kB).  Look for a configuration file somewhere in the esp32 toolchain, I think it has a ".csv" extension.

uwe-pt, 2019-05-15:
Ed, thank you very much, but you overestimate my programming skills. I do, however, wonder

uwe-pt, 2019-05-15:
Ed, you overestimate my programming skills. I do, however, wonder why the limit is set to 100 stations as the data written into the file are less than 18kB with 200 stations (checked this as txt file). Isn't there a simple possibility to increase the limit so that the 20kB can be fully used?

Edzelf, 2019-05-15:
You are right.  I did not realize that there is a limit on the number of presets.  I will look into that.

Edzelf, 2019-05-15:
Could you try the new version I just uploaded?  There is now a definition "MAXPRESETS" for the number of presets.

uwe-pt, 2019-05-15:
Thank you, in principle it works. If I have stations from preset_80 to preset_120 I can scroll through them using the GPIO buttons, thus having presets > 99 should be ok. Trying to upload a configuration file with 200 stations (having as txt only 13.4 kB) only 99 stations get copied into NVS and configuration following the stations, such as tone* and wifi* get lost.

Edzelf, 2019-05-15:
I could only load 79 presets in 20 kB NVS.  So you have to increase the NVS partition.
Look for "default.csv".  On my computer it is on "C:\Users\Ed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\tools\partitions".  It looks like this:
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x140000,
app1,     app,  ota_1,   0x150000,0x140000,
eeprom,   data, 0x99,    0x290000,0x1000,
spiffs,   data, spiffs,  0x291000,0x16F000,
```
Using some SPIFFS space for nvs, you may try this (untested!):
```
# Name,   Type, SubType, Offset,  Size, Flags
nvsold,   data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x140000,
app1,     app,  ota_1,   0x150000,0x140000,
eeprom,   data, 0x99,    0x290000,0x1000,
nvs,      data, nvs,     0x291000,0xB000,
spiffs,   data, spiffs,  0x29C000,0x164000,
```
Note that you loose the current NVS settings (webinterface config page), so save the setting before uploading the sketch.

uwe-pt, 2019-05-15:
Everything works fine! I had to trim the stations name to squeeze all 200 stations in, but that is ok. The only thing that is not working is the web interface, the drop-down menu of the presets remains empty. Selection of stations >99 works also with the 'next' button.

Edzelf, 2019-05-16:
Do you have this version?:
#define VERSION     "Wed, 15 May 2019 15:20:00 GMT"

uwe-pt, 2019-05-16:
Yes, I do have this version. After force cache refresh (Crtl+F5) the drop-down is filled with all stations. However, a click on any station brings me to station '0'. Is the format 'preset_000' to 'preset_199' correct?

Rainer-G, 2019-05-16:
seems to me that the formatting string uses only two decimals ?
`String readhostfrompref ( int8_t preset )
{
  char           tkey[12] ;                            // Key as an array of chars

  sprintf ( tkey, "preset_%02d", preset ) ;            // Form the search key
  if ( nvssearch ( tkey ) )                            // Does it exists?
  {
`

Rainer

Edzelf, 2019-05-16:
@Rainer: the newest version should accept 1, 2 or 3 decimals.

Edzelf, 2019-05-16:
@uwe: fixed in the new version.

uwe-pt, 2019-05-16:
Yes, all stations are accessible by the drop-down menu. Unfortunately now the 'prev' function is not working correctly. Clicking 'prev' when the current station is 3 or 103 does not work. The message is e.g. 'Preset is now 102', but station stays on 103. Clicking again message remains the same. It In all other positions (stations) it seems to work flawless.

Edzelf, 2019-05-16:
Are you sure that the stations under preset 2 and preset 102 are working?

uwe-pt, 2019-05-16:
Checked 000-004 and 100-104 all working.

Edzelf, 2019-05-16:
Sorry, I cannot reproduce it here.  Try to analyze the logging on the Serial output.

uwe-pt, 2019-05-16:
It has to do with this station:
www.listenlive.eu/rfimonde.m3u # 3 - RadioFranceInternationale(RFI)Monde
plays in most cases normal, but occasionally not. Inhibits 'prev' always.
Output on 'prev' from 4 to 3:
D: Command client available
D: handlehttp started
D: Get command is: downpreset=1
D: Send reply for downpreset=1
D: Command: downpreset with parameter 1
D: STOP requested
D: Stopping client
D: New preset/file requested (3/0) from www.listenlive.eu/rfimonde.m3u
D: Connect to new host www.listenlive.eu/rfimonde.m3u
D: Playlist request, entry 1
D: Connect to www.listenlive.eu on port 80, extension /rfimonde.m3u
D: Connected to server
D: nvs_get_str failed 1102 for key basicauth, keylen is 9, len is 150!
D: Contents: 
D: Read from playlist
D: Playlistheader: HTTP/1.1 200 OK
D: Playlistheader: ETag: "24-582b717e-31db8640c38cf525;;;"
D: Playlistheader: Last-Modified: Tue, 15 Nov 2016 20:35:10 GMT
D: Playlistheader: Content-Type: audio/x-mpegurl
D: Playlistheader: Content-Length: 36
D: Content-Length is 36
D: Playlistheader: Accept-Ranges: bytes
D: Playlistheader: Date: Thu, 16 May 2019 13:47:56 GMT
D: Playlistheader: Server: LiteSpeed
D: Playlistheader: Vary: User-Agent
D: Playlistheader: Connection: close
D: Playlistheader: 
D: Switch to PLAYLISTDATA, search for entry 1
D: Playlistdata: http://live02.rfi.fr/rfimonde-64.mp3
D: Entry 1 in playlist found: http://live02.rfi.fr/rfimonde-64.mp3
D: Connect to new host live02.rfi.fr/rfimonde-64.mp3
D: Connect to live02.rfi.fr on port 80, extension /rfimonde-64.mp3
D: Connected to server
D: nvs_get_str failed 1102 for key basicauth, keylen is 9, len is 150!
D: Contents: 
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: ice-audio-info: samplerate=22050;channels=2
D: Headerline: icy-description:RFI monde 64
D: Headerline: icy-genre:Misc
D: Headerline: icy-name:RFI monde 64
D: Headerline: icy-pub:1
D: Headerline: icy-url:http://www.rfi.fr
D: Headerline: Server: Icecast 2.4.0-kh6
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Connection: Close
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 0, metaint is 16000
D: Metadata block 16 bytes
D: Streamtitle found, 15 bytes
D: StreamTitle='';
Output on 'prev' from 3 to 2 (not working, stays on 3):
D: Command client available
D: handlehttp started
D: Get command is: downpreset=1
D: Send reply for downpreset=1
D: Command: downpreset with parameter 1
D: STOP requested
D: Stopping client
D: New preset/file requested (3/0) from www.listenlive.eu/rfimonde.m3u
D: Connect to new host www.listenlive.eu/rfimonde.m3u
D: Playlist request, entry 1
D: Connect to www.listenlive.eu on port 80, extension /rfimonde.m3u
D: Connected to server
D: nvs_get_str failed 1102 for key basicauth, keylen is 9, len is 150!
D: Contents: 
D: Read from playlist
D: Playlistheader: HTTP/1.1 200 OK
D: Playlistheader: ETag: "24-582b717e-31db8640c38cf525;;;"
D: Playlistheader: Last-Modified: Tue, 15 Nov 2016 20:35:10 GMT
D: Playlistheader: Content-Type: audio/x-mpegurl
D: Playlistheader: Content-Length: 36
D: Content-Length is 36
D: Playlistheader: Accept-Ranges: bytes
D: Playlistheader: Date: Thu, 16 May 2019 13:50:39 GMT
D: Playlistheader: Server: LiteSpeed
D: Playlistheader: Vary: User-Agent
D: Playlistheader: Connection: close
D: Playlistheader: 
D: Switch to PLAYLISTDATA, search for entry 1
D: Playlistdata: http://live02.rfi.fr/rfimonde-64.mp3
D: Entry 1 in playlist found: http://live02.rfi.fr/rfimonde-64.mp3
D: Connect to new host live02.rfi.fr/rfimonde-64.mp3
D: Connect to live02.rfi.fr on port 80, extension /rfimonde-64.mp3
D: Connected to server
D: nvs_get_str failed 1102 for key basicauth, keylen is 9, len is 150!
D: Contents: 
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: ice-audio-info: samplerate=22050;channels=2
D: Headerline: icy-description:RFI monde 64
D: Headerline: icy-genre:Misc
D: Headerline: icy-name:RFI monde 64
D: Headerline: icy-pub:1
D: Headerline: icy-url:http://www.rfi.fr
D: Headerline: Server: Icecast 2.4.0-kh6
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Connection: Close
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 0, metaint is 16000
D: SPI semaphore not taken within 120 ticks by CPU 1, id hspectft
D: SPI semaphore not taken within 130 ticks by CPU 1, id hspectft
D: SPI semaphore not taken within 140 ticks by CPU 1, id hspectft
D: SPI semaphore not taken within 150 ticks by CPU 1, id hspectft
D: SPI semaphore not taken within 160 ticks by CPU 1, id hspectft
D: SPI semaphore not taken within 170 ticks by CPU 1, id hspectft
D: SPI semaphore not taken within 180 ticks by CPU 1, id hspectft
D: SPI semaphore not taken within 190 ticks by CPU 1, id hspectft
D: SPI semaphore not taken within 200 ticks by CPU 0, id hspectft
D: SPI semaphore not taken within 210 ticks by CPU 0, id hspectft
D: SPI semaphore not taken within 220 ticks by CPU 0, id hspectft
D: Metadata block 16 bytes
D: Streamtitle found, 15 bytes
D: StreamTitle='';
Also if a station is not suitable for any reason it probably should not affect scrolling.

uwe-pt, 2019-05-17:
Ed, I have 'solved' the problem by deleting the station in question. If you want to look into the cause, I stand by for testing. Otherwise we can close this issue, as the number of stations permitted has been increased. Thank you for that!

uwe-pt, 2020-02-01:
Hi Ed, I think above problem has to do with the radio station ending .m3u. If I remove all stations ending with .m3u the 'next' and 'prev' buttons in the web interface work flawless. Skipping with 'next' and 'prev' also works if I disable  'if ( host.endsWith ( ".m3u" ) )  // Is it an m3u playlist?' by replacing '.m3u' by '.m3ux'. I think the program tries to jump to a next song within the same station instead to jump to the next or previous radio station. As the number of stations using the .m3u ending seems to be increasing it would be great if this could be fixed. Thank you.