Edzelf/ESP32-Radio #256: Separate defines for NAME (NVS) and hostname

chrkau75, 2019-05-16:
The 
#define NAME 
is currently used for both the NVS namespace and also for the hostname of the device in the network.
As some might use more than one ESP32Radios-Instances in one subnet, perhaps it would be possible to get/set hostname from a string in NVS preferences to distinguish between instances by hostname.
BTW, great project!

Edzelf, 2019-05-16:
Did you try it with 2 radio's?  I would not be suprised if the router adds a postfix in case of duplicate names.

chrkau75, 2019-05-16:
Thanks for quick reply - it´s not a matter of function, nor a bug - just as a suggestion to be able to set meaningful names, e.g. RadioLivingRoom

Rainer-G, 2019-05-16:
I'm running 2 Radios at the same time, the router adds the IP address, so I can open 2 Web Interfaces, OTA works as well: 
[image]