Edzelf/ESP32-Radio #257: Libraries

cosminiq, 2019-05-20:
Hi Ed. You have some links to used libs? I have problems with NVS, i don't find it.

Edzelf, 2019-05-20:
I think this is part of the ESP32 core.  So it should always be available.

cosminiq, 2019-05-20:
hi. got this error. what it meens???

C:\Users\Lenovo\Desktop\ESP32-Radio-master (1)\ESP32-Radio-master\Esp32_radio\Esp32_radio.ino: In function 'void nvsopen()':

Esp32_radio:1125:31: error: 'NVS_READWRITE' was not declared in this scope

     nvserr = nvs_open ( NAME, NVS_READWRITE, &nvshandle ) ;  // No, open nvs

Edzelf, 2019-05-20:
This is fefined in nvs.h.  Are you sure that you have selected an ESP32 board?

cosminiq, 2019-05-20:
I think saw. I want to reinstal Arduino IDE and the ESP 32 lib. 

I will try this tomorow.

I really whant to make this project.

Thx for your fast reply!

cosminiq, 2019-05-21:
Arduino: 1.8.9 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

C:\Users\Lenovo\Desktop\ESP32-Radio-master (1)\ESP32-Radio-master\Esp32_radio\Esp32_radio.ino: In function 'void nvsopen()':

Esp32_radio:1125:31: error: 'NVS_READWRITE' was not declared in this scope

     nvserr = nvs_open ( NAME, NVS_READWRITE, &nvshandle ) ;  // No, open nvs

                               ^

Esp32_radio:1125:57: error: 'nvs_open' was not declared in this scope

     nvserr = nvs_open ( NAME, NVS_READWRITE, &nvshandle ) ;  // No, open nvs

                                                         ^

C:\Users\Lenovo\Desktop\ESP32-Radio-master (1)\ESP32-Radio-master\Esp32_radio\Esp32_radio.ino: In function 'esp_err_t nvsclear()':

Esp32_radio:1142:36: error: 'nvs_erase_all' was not declared in this scope

   return nvs_erase_all ( nvshandle ) ;                // Clear all keys

                                    ^

C:\Users\Lenovo\Desktop\ESP32-Radio-master (1)\ESP32-Radio-master\Esp32_radio\Esp32_radio.ino: In function 'String nvsgetstr(const char*)':

Esp32_radio:1158:56: error: 'nvs_get_str' was not declared in this scope

   nvserr = nvs_get_str ( nvshandle, key, nvs_buf, &len ) ;

                                                        ^

C:\Users\Lenovo\Desktop\ESP32-Radio-master (1)\ESP32-Radio-master\Esp32_radio\Esp32_radio.ino: In function 'esp_err_t nvssetstr(const char*, String)':

Esp32_radio:1184:12: error: 'ESP_ERR_NVS_NOT_ENOUGH_SPACE' was not declared in this scope

     return ESP_ERR_NVS_NOT_ENOUGH_SPACE ;

            ^

Esp32_radio:1194:56: error: 'nvs_set_str' was not declared in this scope

     nvserr = nvs_set_str ( nvshandle, key, val.c_str() ) ; // Store key and value

                                                        ^

C:\Users\Lenovo\Desktop\ESP32-Radio-master (1)\ESP32-Radio-master\Esp32_radio\Esp32_radio.ino: In function 'void nvschkey(const char*, const char*)':

Esp32_radio:1216:37: error: 'nvs_erase_key' was not declared in this scope

     nvs_erase_key ( nvshandle, oldk ) ;                    // Remove key

                                     ^

C:\Users\Lenovo\Desktop\ESP32-Radio-master (1)\ESP32-Radio-master\Esp32_radio\Esp32_radio.ino: In function 'bool nvssearch(const char*)':

Esp32_radio:1280:53: error: 'nvs_get_str' was not declared in this scope

   nvserr = nvs_get_str ( nvshandle, key, NULL, &len ) ; // Get length of contents

                                                     ^

Multiple libraries were found for "SD.h"
 Used: C:\Users\Lenovo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
'NVS_READWRITE' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

cosminiq, 2019-05-21:
no progres. same error. new reinstal.

cosminiq, 2019-05-21:
i use this lib for escape from the first error

https://github.com/knolleary/pubsubclient

cosminiq, 2019-05-21:
so. i think the problem in mqtt lib. Can you recomand one good lib?

cosminiq, 2019-05-21:
new update.
i manage to compile the schetch in linux Mint!!! but...
i make the connection to ssid, i link the Nextion display and ... booom! the esp 32 is restarting forever.
If Nextion is not link stil esp32 is restarting.

i compile the schetch again, with dummy display and the debug and web page is stable.

any ideea for this behavior?

cosminiq, 2019-05-22:
And finaly it works😁.

closed 2019-05-22 by cosminiq