Edzelf/ESP32-Radio #258: Pin configuration

Tony763, 2019-05-23:
Hello,
   is there a way to read saved settings for pins sooner?

I saved a configuration via web interface.

```
pin_vs_cs = 5    
pin_vs_dcs = 32  
pin_vs_dreq = 4 
#
sd_cs_pin = 22
#
spi_miso_pin = 19
spi_mosi_pin = 23
spi_sck_pin = 18
#
tft_cs_pin = 27
tft_dc_pin = 25
tft_rst_pin = 33
```

Console:
```
22:24:27.652 -> 
22:24:27.652 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Wed, 24 Apr 2019 08:20:00 GMT.  Free memory 293432
22:24:27.652 -> D: Display type is ILI9341
22:24:27.652 -> D: Partition nvs found, 20480 bytes
22:24:27.685 -> D: Read 22 keys from NVS
22:24:27.685 -> D: pin_ir set to -1
22:24:27.685 -> D: pin_enc_clk set to -1
22:24:27.685 -> D: pin_enc_dt set to -1
22:24:27.685 -> D: pin_enc_sw set to -1
22:24:27.685 -> D: pin_tft_cs set to -1
22:24:27.685 -> D: pin_tft_dc set to -1
22:24:27.685 -> D: pin_tft_scl set to -1
22:24:27.685 -> D: pin_tft_sda set to -1
22:24:27.685 -> D: pin_tft_bl set to -1
22:24:27.685 -> D: pin_tft_blx set to -1
22:24:27.685 -> D: pin_tft_rst set to -1
22:24:27.685 -> D: pin_sd_cs set to -1
22:24:27.685 -> D: pin_vs_cs set to 5
22:24:27.685 -> D: pin_vs_dcs set to 32
22:24:27.685 -> D: pin_vs_dreq set to 4
22:24:27.719 -> D: pin_shutdown set to -1
22:24:27.719 -> D: pin_shutdownx set to -1
22:24:27.719 -> D: pin_spi_sck set to 18
22:24:27.719 -> D: pin_spi_miso set to 19
22:24:27.719 -> D: pin_spi_mosi set to 23
22:24:27.719 -> D: GPIO0 is HIGH
22:24:27.719 -> D: GPIO2 is HIGH
22:24:27.719 -> D: GPIO4 is HIGH
22:24:27.753 -> D: GPIO5 is HIGH
22:24:27.753 -> D: GPIO12 is HIGH
22:24:27.753 -> D: GPIO13 is HIGH
22:24:27.787 -> D: GPIO14 is HIGH
22:24:27.787 -> D: GPIO15 is HIGH
22:24:27.787 -> D: GPIO16 is HIGH
22:24:27.787 -> D: GPIO17 is HIGH
22:24:27.822 -> D: GPIO18 is HIGH
22:24:27.822 -> D: GPIO19 is HIGH
22:24:27.822 -> D: GPIO21 is HIGH
22:24:27.857 -> D: GPIO22 is HIGH
22:24:27.857 -> D: GPIO23 is HIGH
22:24:27.857 -> D: GPIO25 is HIGH
22:24:27.857 -> D: GPIO26 is HIGH
22:24:27.893 -> D: GPIO27 is HIGH
22:24:27.893 -> D: GPIO32 is HIGH
22:24:27.893 -> D: GPIO33 is HIGH
22:24:27.927 -> D: GPIO34 is LOW, probably no PULL-UP
22:24:27.927 -> D: GPIO35 is LOW, probably no PULL-UP
22:24:27.927 -> D: GPIO39 is LOW, probably no PULL-UP
22:24:27.927 -> D: Create list with acceptable WiFi networks
22:24:27.927 -> D: Added mirror-guard2 to list of networks
22:24:27.962 -> [I][WiFiMulti.cpp:202] APlistAdd(): [WIFI][APlistAdd] add SSID: mirror-guard2
22:24:27.962 -> D: End adding networks
22:24:28.032 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
22:24:28.032 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
22:24:28.137 -> D: Scan Networks
22:24:29.631 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 1 - SCAN_DONE
22:24:29.631 -> D: Scan completed
22:24:29.669 -> D: Number of available networks: 1
22:24:29.669 -> D:  1 - mirror-guard2             Signal: -58 dBm, Encryption WPA_WPA2_PSK, Acceptable
22:24:29.669 -> D: End of list
22:24:29.669 -> D: Command: clk_dst with parameter 1
22:24:29.669 -> D: Command: clk_offset with parameter 1
22:24:29.669 -> D: Command: clk_server with parameter 10.0.1.1
22:24:29.669 -> D: Command: pin_vs_cs with parameter 5
22:24:29.669 -> D: Command: pin_vs_dcs with parameter 32
22:24:29.669 -> D: Command: pin_vs_dreq with parameter 4
22:24:29.669 -> D: Command: preset with parameter 0
22:24:29.669 -> D: Command: preset_00 with parameter 10.0.1.1/kissproton128.mp3.m3u
22:24:29.705 -> D: Command: preset_01 with parameter icecast3.play.cz/evropa2-128.mp3
22:24:29.705 -> D: Command: sd_cs_pin with parameter 22
22:24:29.705 -> D: Command: spi_miso_pin with parameter 19
22:24:29.705 -> D: Command: spi_mosi_pin with parameter 23
22:24:29.705 -> D: Command: spi_sck_pin with parameter 18
22:24:29.705 -> D: Command: tft_cs_pin with parameter 27
22:24:29.705 -> D: Command: tft_dc_pin with parameter 25
22:24:29.705 -> D: Command: tft_rst_pin with parameter 33
22:24:29.740 -> D: Command: toneha with parameter 0
22:24:29.740 -> D: Command: tonehf with parameter 1
22:24:29.740 -> D: Command: tonela with parameter 0
22:24:29.740 -> D: Command: tonelf with parameter 2
22:24:29.740 -> D: Command: volume with parameter 70
22:24:29.844 -> D: Slow SPI, Testing VS1053 read/write registers...
22:24:30.124 -> D: Fast SPI, Testing VS1053 read/write registers again...
```
This way Display and SD is not working.

But if I change pins numbers in arduino sketch:
```
  { "pin_tft_cs",    &ini_block.tft_cs_pin,       27 }, // Display SPI version
  { "pin_tft_dc",    &ini_block.tft_dc_pin,       25 }, , // Display SPI version
  ...
```
Display and SD start to work.
```
23:01:30.484 -> 
23:01:30.484 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Wed, 24 Apr 2019 08:20:00 GMT.  Free memory 293432
23:01:30.519 -> D: Display type is ILI9341
23:01:30.519 -> D: Partition nvs found, 20480 bytes
23:01:30.519 -> D: Read 22 keys from NVS
23:01:30.519 -> D: pin_ir set to -1
23:01:30.519 -> D: pin_enc_clk set to -1
23:01:30.519 -> D: pin_enc_dt set to -1
23:01:30.519 -> D: pin_enc_sw set to -1
23:01:30.519 -> D: pin_tft_cs set to 27
23:01:30.519 -> D: pin_tft_dc set to 25
23:01:30.519 -> D: pin_tft_scl set to -1
23:01:30.519 -> D: pin_tft_sda set to -1
23:01:30.519 -> D: pin_tft_bl set to -1
23:01:30.519 -> D: pin_tft_blx set to -1
23:01:30.519 -> D: pin_tft_rst set to 33
23:01:30.553 -> D: pin_sd_cs set to 22
23:01:30.553 -> D: pin_vs_cs set to 5
23:01:30.553 -> D: pin_vs_dcs set to 32
23:01:30.553 -> D: pin_vs_dreq set to 4
23:01:30.553 -> D: pin_shutdown set to -1
23:01:30.553 -> D: pin_shutdownx set to -1
23:01:30.553 -> D: pin_spi_sck set to 18
23:01:30.553 -> D: pin_spi_miso set to 19
23:01:30.553 -> D: pin_spi_mosi set to 23
23:01:30.553 -> D: GPIO0 is HIGH
23:01:30.553 -> D: GPIO2 is HIGH
23:01:30.588 -> D: GPIO4 is HIGH
23:01:30.588 -> D: GPIO5 is HIGH
23:01:30.588 -> D: GPIO12 is HIGH
23:01:30.623 -> D: GPIO13 is HIGH
23:01:30.623 -> D: GPIO14 is HIGH
23:01:30.623 -> D: GPIO15 is HIGH
23:01:30.623 -> D: GPIO16 is HIGH
23:01:30.658 -> D: GPIO17 is HIGH
23:01:30.658 -> D: GPIO18 is HIGH
23:01:30.658 -> D: GPIO19 is HIGH
23:01:30.693 -> D: GPIO21 is HIGH
23:01:30.693 -> D: GPIO22 is HIGH
23:01:30.693 -> D: GPIO23 is HIGH
23:01:30.693 -> D: GPIO25 is HIGH
23:01:30.728 -> D: GPIO26 is HIGH
23:01:30.728 -> D: GPIO27 is HIGH
23:01:30.728 -> D: GPIO32 is HIGH
23:01:30.762 -> D: GPIO33 is HIGH
23:01:30.762 -> D: GPIO34 is LOW, probably no PULL-UP
23:01:30.762 -> D: GPIO35 is LOW, probably no PULL-UP
23:01:30.762 -> D: GPIO39 is LOW, probably no PULL-UP
23:01:30.797 -> D: Start display
23:01:31.527 -> D: Locate mp3 files on SD, may take a while...
23:01:31.631 -> D: 2 tracks on SD
23:01:31.631 -> D: Create list with acceptable WiFi networks
23:01:31.631 -> D: Added mirror-guard2 to list of networks
23:01:31.631 -> [I][WiFiMulti.cpp:202] APlistAdd(): [WIFI][APlistAdd] add SSID: mirror-guard2
23:01:31.666 -> D: End adding networks
23:01:31.735 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
23:01:31.735 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
23:01:31.839 -> D: Scan Networks
23:01:33.336 -> [D][WiFiGeneric.cpp:336] _eventCallback(): Event: 1 - SCAN_DONE
23:01:33.336 -> D: Scan completed
23:01:33.336 -> D: Number of available networks: 1
23:01:33.336 -> D:  1 - mirror-guard2             Signal: -57 dBm, Encryption WPA_WPA2_PSK, Acceptable
23:01:33.371 -> D: End of list
23:01:33.371 -> D: Command: clk_dst with parameter 1
23:01:33.371 -> D: Command: clk_offset with parameter 1
23:01:33.371 -> D: Command: clk_server with parameter 10.0.1.1
23:01:33.371 -> D: Command: pin_vs_cs with parameter 5
23:01:33.371 -> D: Command: pin_vs_dcs with parameter 32
23:01:33.371 -> D: Command: pin_vs_dreq with parameter 4
23:01:33.371 -> D: Command: preset with parameter 0
23:01:33.371 -> D: Command: preset_00 with parameter 10.0.1.1/kissproton128.mp3.m3u
23:01:33.406 -> D: Command: preset_01 with parameter icecast3.play.cz/evropa2-128.mp3
23:01:33.406 -> D: Command: sd_cs_pin with parameter 22
23:01:33.406 -> D: Command: spi_miso_pin with parameter 19
23:01:33.406 -> D: Command: spi_mosi_pin with parameter 23
23:01:33.406 -> D: Command: spi_sck_pin with parameter 18
23:01:33.406 -> D: Command: tft_cs_pin with parameter 27
23:01:33.406 -> D: Command: tft_dc_pin with parameter 25
23:01:33.406 -> D: Command: tft_rst_pin with parameter 33
23:01:33.406 -> D: Command: toneha with parameter 0
23:01:33.441 -> D: Command: tonehf with parameter 1
23:01:33.441 -> D: Command: tonela with parameter 0
23:01:33.441 -> D: Command: tonelf with parameter 2
23:01:33.441 -> D: Command: volume with parameter 70
23:01:33.546 -> D: Slow SPI, Testing VS1053 read/write registers...
23:01:33.823 -> D: Fast SPI, Testing VS1053 read/write registers again...
```

Edzelf, 2019-05-24:
In your configuration you have:
```
tft_cs_pin = 27
tft_dc_pin = 25
tft_rst_pin = 33
```
But it should be:
```
pin_tft_cs = 27
pin_tft_dc = 25
pin_tft_rst = 33
```

Tony763, 2019-05-26:
Sorry, my bad. Changed as You mentioned and it is working.

closed 2019-05-26 by Tony763