Edzelf/ESP32-Radio #259: Is it possible ESP32 to connect to a  5G connection...a.k.a..802.11ac?

SurenBono, 2019-05-24:


PepeESP32, 2019-06-13:
The ESP32 is 2.4GHz only...

SurenBono, 2019-06-20:
..cant we  modifiy the core to accept a 5G /802.11a since it was the 1st types of wifi band ever existed before anything that we are using now...

SurenBono, 2019-06-20:
[image]

SurenBono, 2019-06-20:
[image]

SurenBono, 2019-06-20:
[image]

SurenBono, 2019-06-26:
[image]