Edzelf/ESP32-Radio #261: LCD1602 communication error

ido1990, 2019-06-02:
Hi,
Thank you for your project!
Just got my parts and started to assemble it.
I'm using small LCD display that is working with Arduino over I2C.
The chip is ST7066 (HD44780 compatible) so it should work like 1602...
[image]

[image]

The LCD is working with this code:
```
/* I2C LCD 16x2 Arduino Tutorial
 * More info http://www.ardumotive.com/i2clcd(en).html
 * Dev: Michalis Vasilakis Date: 19/11/2016 Ver. 1.0 */

//Libraries 
#include <Wire.h>  
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);  // Set the LCD I2C address, if it's not working try 0x27.

void setup(){
  lcd.begin(16,2);   // iInit the LCD for 16 chars 2 lines
  lcd.backlight();   // Turn on the backligt (try lcd.noBaklight() to turn it off)
  lcd.setCursor(0,0); //First line
  lcd.print("Hi!");

}

void loop(){
}
```

But when I connect it to the ESP32 I get an error:
```
E (22037) i2c: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2c.c:1244 (i2c_master_cmd_begin):i2c driver not installed
D: LCD1602 communication error!
```
And then it crashes:

```

D: LCD1602 communication error!
E (1692) i2c: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2c.c:1244 (i2c_master_cmd_begin):i2c driver not installed
D: LCD1602 communication error!
E (1707) i2c: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2c.c:1244 (i2c_master_cmd_begin):i2c driver not installed
D: LCD1602 communication error!
E (1722) i2c: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2c.c:1244 (i2c_master_cmd_begin):i2c driver not installed
D: LCD1602 communication error!
E (1737) i2c: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2c.c:1244 (i2c_master_cmd_begin):i2c driver not installed
D: LCD1602 communication error!
E (1752) i2c: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2c.c:1244 (i2c_master_cmd_begin):i2c driver not installed
D: LCD1602 communication error!
E (1767) i2c: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2c.c:1244 (i2c_master_cmd_begin):i2c driver not installed
D: LCD1602 communication error!
E (1782) i2c: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2c.c:1244 (i2c_master_cmd_begin):i2c driver not installed
D: LCD1602 communication error!
E (1797) i2c: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2c.c:1244 (i2c_master_cmd_begin):i2c driver not installed
D: LCD1602 communication error!
E (23820) i2c: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2c.c:1244 (i2c_master_cmd_begin):i2c driver not installed
D: LCD1602 communication error!
E (23836) i2c: /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/i2c.c:1244 (i2c_master_cmd_begin):i2c driver not installed
D: LCD1602 communication error!
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (Spftask) 
Core 1 register dump:
PC      : 0x4008d537  PS      : 0x00060636  A0      : 0x40082bc4  A1      : 0x3ffd4b00  
A2      : 0x00000000  A3      : 0x00060623  A4      : 0x00060620  A5      : 0x00000001  
A6      : 0x00060423  A7      : 0x00000000  A8      : 0x3ffd5344  A9      : 0x0000abab  
A10     : 0x00000000  A11     : 0x4015d9e4  A12     : 0x3ffb8678  A13     : 0x00000080  
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff9  

Backtrace: 0x4008d537:0x3ffd4b00 0x40082bc1:0x3ffd4bd0 0x4008b43b:0x3ffd4bf0 0x4008b4f9:0x3ffd4c30 0x40081cf9:0x3ffd4c50 0x40081ded:0x3ffd4c80 0x4015da04:0x3ffd4ca0 0x400f4e0b:0x3ffd4cc0 0x4000bd83:0x3ffd4ce0 0x4000117d:0x3ffd4d00 0x400592fe:0x3ffd4d20 0x4005937a:0x3ffd4d40 0x40058b69:0x3ffd4d60 0x40102e31:0x3ffd4d90 0x401085ce:0x3ffd4dc0 0x4010875d:0x3ffd50d0 0x40085eba:0x3ffd5100 0x400ea496:0x3ffd5150 0x400d2c58:0x3ffd5190 0x400d2c86:0x3ffd51b0 0x400d2cb3:0x3ffd51d0 0x400d2cc4:0x3ffd51f0 0x400d2de5:0x3ffd5210 0x400d2ef5:0x3ffd5240 0x400d4abe:0x3ffd5270 0x400d4b42:0x3ffd5290 0x4008a775:0x3ffd52b0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6400
entry 0x400806a8
```
(had to cut lines from the output, it was too long to paste here)

Looking for clues to debug this issue...
Any help would be appreciated :)
Thank you!

Edzelf, 2019-06-02:
The I2C address of your display interface is 0x27, so change the value in line 22 of lcd1602.h from 0x3f to 0x27.

ido1990, 2019-06-02:
> 
> 
> The I2C address of your display interface is 0x27, so change the value in line 22 of lcd1602.h from 0x3f to 0x27.

Hi Ed!
Already changed it there...

Edzelf, 2019-06-02:
Are the gpio pins correctly set in your preferences?

Edzelf, 2019-06-02:
The i2c adress is shifted one place to the left before it is ised, so try 0x13 and 0x1f.

ido1990, 2019-06-02:
> 
> 
> Are the gpio pins correctly set in your preferences?

Yes.
This is my config:
```
pin_enc_clk = -1                                     # GPIO Pin number for rotary encoder "CLK"
pin_enc_dt = -1                                     # GPIO Pin number for rotary encoder "DT"
pin_enc_sw = -1                                      # GPIO Pin number for rotary encoder "SW"
pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
pin_sd_cs = 21                                       # GPIO Pin number for SD card "CS"
pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"
pin_tft_scl = 33 # GPIO Pin number for SCL
pin_tft_sda = 34 # GPIO Pin number for SDA
pin_vs_cs = 5                                        # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 32                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"
```
(disabled the rotary encoder for now)

> 
> 
> The i2c adress is shifted one place to the left before it is ised, so try 0x13 and 0x1f.

Tried now.
Same error.

ido1990, 2019-06-02:
I just noticed that pins 34-36 and 39 are input only.
Changed to 25 and 26.
Now getting this:
```
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!
D: LCD1602 communication error!

```

Edzelf, 2019-06-02:
Try to runthe i2c scan sample project.

ido1990, 2019-06-02:
Ok, so the pins are 22 and 21.
Can't use other pins.
And I had to change the address to 0x27
Thank you so much!

BTW, is there any way to change the text on the screen to the preset name?
Some of my streams doesn't have title...

Edzelf, 2019-06-03:
It would not be so hard to change the screen texts.  It is a good opportunity to learn programming.

ido1990, 2019-06-03:
> 
> 
> It would not be so hard to change the screen texts. It is a good opportunity to learn programming.

I'll dig into it.
Thank you!

closed 2019-06-03 by ido1990

Thoky81, 2020-02-16:
Hi,

can you please post your working code? I am not able to figure out why the lcd is not working even if I dont get any errors.
Thanks

ido1990, 2020-02-16:
> 
> 
> Hi,
> 
> can you please post your working code? I am not able to figure out why the lcd is not working even if I dont get any errors.
> Thanks

Hi,
I recommend you to open new issue, yours might be different from what I had.

DanDimnet, 2024-04-07:
Píše mi to stejnou chybu a nevím proč i když mám piny nastavený 21, a 22.