Edzelf/ESP32-Radio #263: Rotary Encoder 1 click left and 1 click right problem.

ve2cfb, 2019-06-04:
Hello Ed,

Can you tell me if this is a normal operation or not..

While using the Encoder for either Volume or Preset, the first encoder click does not show up.
Sometime it does but its a russian roulette. The other example is, if i continiously click the encoder 1 notch to the left and 1 notch to the right, i do not get an increment.

Here is a small trace of the  locrotcount variable

                              Taken from   void IRAM_ATTR isr_enc_turn() 

  ```
// Read current state of CLK, DT pin. Result is a 2 bit binary number: 00, 01, 10 or 11.
  act_state = ( digitalRead ( ini_block.enc_clk_pin ) << 1 ) +
              digitalRead ( ini_block.enc_dt_pin ) ;
  inx = ( old_state << 2 ) + act_state ;                        // Form index in enc_states
  locrotcount += enc_states[inx] ;                              // Get delta: 0, +1 or -1
dbgprint ( "locrotcount, %d bytes", locrotcount ) ;   // ================ Added by me
  if ( locrotcount == 4 )
  {
    rotationcount++ ;                                           // Divide by 4
    locrotcount = 0 ;
  }
  else if ( locrotcount == -4 )
  {
    rotationcount-- ;                                           // Divide by 4
    locrotcount = 0 ;
  }
  old_state = act_state ;                                       // Remember current status
  enc_inactivity = 0 ;
}
============================================================

```
As you can see, there is no Increment as i rotate the encoder back and forth.

D: locrotcount, 0 bytes
D: locrotcount, 1 bytes
D: locrotcount, 2 bytes
D: locrotcount, 2 bytes
D: locrotcount, 3 bytes

D: locrotcount, 2 bytes
D: locrotcount, 1 bytes
D: locrotcount, 1 bytes
D: locrotcount, 0 bytes
D: locrotcount, -1 bytes

D: locrotcount, 0 bytes
D: locrotcount, 1 bytes
D: locrotcount, 2 bytes
D: locrotcount, 2 bytes
D: locrotcount, 3 bytes

D: locrotcount, 2 bytes
D: locrotcount, 1 bytes
D: locrotcount, 0 bytes
D: locrotcount, -1 bytes
D: locrotcount, 0 bytes

D: locrotcount-1, 1 bytes
D: locrotcount-1, 1 bytes
D: locrotcount-1, 2 bytes
D: locrotcount-1, 2 bytes
D: locrotcount-1, 3 bytes

Pierre VE2CFB

Edzelf, 2019-06-04:
It is not allowed to use dbgprint in an interrupt routine.  As far as I know, there is a debugline for the rotary encoder at non-interrupt level.

Edzelf, 2019-06-04:
It could be that 2 clicks is one count.

ve2cfb, 2019-06-04:
Hi Ed,

Is the **"2 clicks is one count"** the normal way.
Is it possible to change that...?

and by the way, i forgot about the Print within the ISR.

Thanks again for your time.

Pierre VE2CFB

Edzelf, 2019-06-05:
The click is mechanical, the pulses are electrical.  You have to redesign the encoder for that.

ve2cfb, 2019-06-05:
Maybe i did not explain the problem correctly.

for example, when i turn the encoder one position for the volume, nothing happen (but sometime is does ).
I have to do 2 turn in order to augment the volume or diminish the volume.
Can we modify the code  so that one turn of the encoder will give me a count of +1 or -1 depending of the rotation. 

Same thing happen when i do a doubleclick for the preset, on message "Turn to select station Press to confirm" i have to turn the encoder  twice in order to get a rotationcount = 1 

Is this a bug in my setup or that the way the code work. Can it be change.

Thanks.

Edzelf, 2019-06-05:
This looks like a bad rotary encoder.

ve2cfb, 2019-06-06:
I did try 4 different encoder but same thing.

Take look at issue #45  quite interesting.
i took that code and tried it and everything works except the **"rotationcount"** goes up by a value of 4 whenever i turn the encoder one click. but at least its steady.
Is it possible to modify the code in order to have  a rotationcount = 1 when turning the encoder 1 click.

Edzelf, 2019-06-06:
Yes, issue 45 is interesting.  That's why I incorporate this part in the radio (27-nov-2019).  I did a test today with the rotary encoder.  My set-up (configuration):
```
pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder "CLK"
pin_enc_dt = 26                                      # GPIO Pin number for rotary encoder "DT"
pin_enc_sw = 27                                      # GPIO Pin number for rotary encoder "SW"
```
Rest of the wiring:
Encoder "GND" to ground.
Encoder "+" to 3.3V
When I turn the knob slowly like: 2 clicks left, 2 clicks right, 2 clicks left, 2 clicks right, ...... I see in de logging:
```
D: Rotation count -1
D: Rotation count 1
D: Rotation count -1
D: Rotation count 1
D: Rotation count -1
D: Rotation count 1
D: Rotation count -1
D: Rotation count 1
D: Rotation count -1
D: Rotation count 1
D: Rotation count -1
D: Rotation count 1
D: Rotation count -1
D: Rotation count 1
D: Rotation count -1
D: Rotation count 1
D: Rotation count -1
D: Rotation count 1
D: Rotation count -1
D: Rotation count 1
```
So, I'm sorry, but I cannot reproduce your problem.

ve2cfb, 2019-06-06:
Exactly... i get the same result. No problem there.

Now, (wait 1 to 2 seconds between clicks) try it with 1 click left and then 1 click right.  Keep doing that and see if the rotationcount increment or decrement. In my case, no rotationcount. 

and try this one: 
= wait for 4-5 sedonds, 
= turn right one turn, 
= rotationcount = 1, 
= turn right one turn,
= rotationcount = 1
= if you keep turning right, rotationcount = 1
= turn left one turn,
= ************ no rotationcount,
= turn left one turn,
= rotationcount = -1,

it seems that when you chage direction, nothing happen. 
you have to do an extra click in the same direction.

If yours is not doing that, then i have a problem with the breadboard and the connection.

Thanks.

Edzelf, 2019-06-06:
I did TWO right clicks an then TWO left clicks every time.  I know that 1 click is not enough if you change direction.  That has to do with the encoding.

closed 2020-09-02 by ve2cfb