Edzelf/ESP32-Radio #264: problem when connecting to wifi.

cherob67, 2019-06-14:
Hellio !
After updating my Esp32-radio, some time, i often have an error when connecting to wifi and i have to do a hard reset on the esp32.
any idea ?

"
D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 16 May 2019 12:10:00 GMT.  Free memory 238248
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 60 keys from NVS
D: pin_ir set to 13
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to 18
D: pin_tft_sda set to 23
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 21
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: Enable pin 13 for IR
D: Start display
D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added Prc to list of networks
D: Added domicile_rc to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 8
D:  1 - Bbox-B476B850             Signal: -78 dBm, Encryption WPA_WPA2_PSK, 
D:  2 - domicile_rc               Signal: -84 dBm, Encryption WPA_WPA2_PSK, Acceptable
D:  3 - XXXX-1698                 Signal: -85 dBm, Encryption WPA_WPA2_PSK, 
D:  4 - Freebox-651FE3            Signal: -89 dBm, Encryption WPA2_PSK, 
D:  5 - FreeWifi_secure           Signal: -90 dBm, Encryption ????, 
D:  6 - Bbox-0CFE9D25             Signal: -91 dBm, Encryption WPA_WPA2_PSK, 
D:  7 - freebox_CeliaSXB          Signal: -91 dBm, Encryption WPA_PSK, 
D:  8 - FreeWifi_secure           Signal: -91 dBm, Encryption ????, 
D: End of list
D: Command: ir_02fd with parameter stop = 0
D: Command: ir_10ef with parameter newpreset = 7
D: Command: ir_18e7 with parameter newpreset = 5
D: Command: ir_22dd with parameter uppreset = 1
D: Command: ir_30cf with parameter newpreset = 4
D: Command: ir_38c7 with parameter newpreset = 8
D: Command: ir_42bd with parameter reset = 1
D: Command: ir_4ab5 with parameter newpreset = 0
D: Command: ir_52ad with parameter stop = 1
D: Command: ir_5aa5 with parameter newpreset = 9
D: Command: ir_629d with parameter upvolume = 2
D: Command: ir_6897 with parameter newpreset = 1
D: Command: ir_7a85 with parameter newpreset = 6
D: Command: ir_9867 with parameter newpreset = 2
D: Command: ir_a857 with parameter downvolume = 2
D: Command: ir_b04f with parameter newpreset = 3
D: Command: ir_c23d with parameter downpreset = 1
D: Command: lstmods with parameter Wed, 24 Apr 2019 07:37:27 GMT
D: Command: mqttbroker with parameter broker.hivemq.com
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttpubtopic with parameter P_espradioIP
D: Command: mqtttopic with parameter P_espradio
D: Command: mqtttpreset with parameter none
D: Command: mqttuser with parameter none
D: Command: preset with parameter 3
D: Command: preset_00 with parameter chai5she.cdn.dvmr.fr:80/franceinter-midfi.mp3
D: Command: preset_01 with parameter chai5she.cdn.dvmr.fr:80/francemusique-midfi.mp3
D: Command: preset_02 with parameter chai5she.cdn.dvmr.fr:80/franceculture-midfi.mp3
D: Command: preset_03 with parameter rfm-live-mp3-128.scdn.arkena.com/rfm.mp3
....
....
D: Command: preset_25 with parameter 163.172.94.169:8080/radiolib
D: Command: preset_26 with parameter cdn.nrjaudio.fm/adwz1/fr/30227/mp3_128.mp3?origine=fluxradios
D: Command: toneha with parameter 13
D: Command: tonehf with parameter 15
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 84
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
E (11301) boot: Assert failed in esp_err_t nvs::PageManager::requestNewPage(), /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_pagemanager.cpp:175 (usedEntries == newPage->getUsedEntryCount())
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
...
"

erichegit, 2019-07-21:
Sadly I suddenly also face this error :(
Any suggestion would be nice

`D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 16 May 2019 12:10:00 GMT.  Free memory 233316
D: Display type is BLUETFT
D: Partition nvs found, 20480 bytes
D: Read 52 keys from NVS
D: pin_ir set to 39
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 25
D: pin_vs_cs set to 32
D: pin_vs_dcs set to 33
D: pin_vs_dreq set to 35
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is HIGH
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = -1
D: gpio_13 will execute downvolume = -1
D: gpio_14 will execute stop
D: gpio_17 will execute resume
D: gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
D: Enable pin 39 for IR
D: Locate mp3 files on SD, may take a while...
D: 8 tracks on SD
D: Create list with acceptable WiFi networks
D: Added HomeWLAN to list of networks
D: Added SSID2 to list of networks
D: End adding networks
E (1433) boot: Assert failed in esp_err_t nvs::PageManager::requestNewPage(), /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_pagemanager.cpp:175 (usedEntries == newPage->getUsedEntryCount())
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE`

Edzelf, 2019-07-21:
Looks like NVS is corrupted.  Try to zero it.

erichegit, 2019-07-25:
you are right - earasing the NVS helped - No idea how this happened - you can close the issue.

Milka-Cow, 2020-04-14:
> Looks like NVS is corrupted. Try to zero it.

Hi Ed,
I have the same problem, but I'm not a programmer.
My son is not willing to spend more time to solve the problem :-(
He must learn for school. He will finish high school soon.

Any tipp or help how to zero NVS?

Tks
Joe

fellek, 2020-04-19:
This may help you:
https://github.com/espressif/esptool/issues/348