Edzelf/ESP32-Radio #265: Implementing the display of Cyrillic fonts without using the utf8ascii() function.

SinglWolf, 2019-06-19:
I Implemented the display of Cyrillic fonts without using the utf8ascii() function in a tft display 2.4 inch 320x240 pixel.
Я реализовал отображение кириллических шрифтов без использования функции utf8ascii() на TFT-дисплее 2,4 дюйма с разрешением 320x240 пикселей.
[image]
On the screen is an example with the .Helvetica NeueUI font. (The quality of the photo is very poor. In reality, the image on the display is just wonderful!)
На экране пример со шрифтом .Helvetica NeueUI. (Качество фото очень плохое. На самом деле изображение на дисплее просто замечательное!)
If the author is interested in the internationalization of the project, I will be happy to share the method used.
Если автор заинтересован в интернационализации проекта, я буду рад поделиться использованным методом.

Egen-M, 2019-12-26:
Здравствуйте SinglWolf! Я собрал Edzelf ESP32 radio и подключил к нему TFT дисплей 4" 320x480 st7796s / ILI9488. Радио хорошо работает, но в нём очень не хватает русского шрифта. Хочу узнать можете ли Вы поделиться секретом русификации данного радио? Спасибо!
[image]

SinglWolf, 2019-12-26:
> Здравствуйте SinglWolf! Я собрал Edzelf ESP32 radio и подключил к нему TFT дисплей 4" 320x480 st7796s / ILI9488. Радио хорошо работает, но в нём очень не хватает русского шрифта. Хочу узнать можете ли Вы поделиться секретом русификации данного радио? Спасибо!

Здравствуйте!
Никакого секрета нет. Просто я использовал библиотеку [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI)
Возни со шрифтами было много, но оно того стоило.
К сожалению, автор не отреагировал этот пост, как отмахнулся от предложения реализовать поддержку [SPIRAM](https://github.com/Edzelf/ESP32-Radio/issues/230) для ESP32-WROVER. Не дождавшись ответа, я стал искть другое решение. Ибо имея на борту девайса кучу неиспользованной RAM, как-то стрёмно.
На данный момент я использую собственный форк [Ka-Radio32](https://github.com/karawin/Ka-Radio32) для IDE VsCode+PlatformIO [KaRadio32-pio](https://github.com/SinglWolf/KaRadio32-pio) (Это мой форк. Там ничего не модифицировано. Только адаптировано под PlatformIO).
Из этого форка я развиваю проект [ESP32-Media](https://github.com/SinglWolf/ESP32-Media). Что из него выйдет пока не знаю. Но особых трудностей не возникает. Все задумки будут реализованы. Может быть в будущем удастся портировать замечательную библиотеку [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI) для своего проекта.
Пишите мне в telegram @ServerDoma. Может быть смогу помочь с внедрением библиотеки. Или найду исходники этого проекта.
Удачи.

Egen-M, 2019-12-26:
На этой библиотеке TFT_eSPI я делал метеостанцию. Хорошая библиотека. Сначала я хотел собрать  Ka-Radio32, но так и не сообразил как подключить туда этот экран, ведь там только bin файлы.

SinglWolf, 2019-12-26:
> На этой библиотеке TFT_eSPI я делал метеостанцию. Хорошая библиотека. Сначала я хотел собрать Ka-Radio32, но так и не сообразил как подключить туда этот экран, ведь там только bin файлы. Подскажите пожалуйста как подключить TFT_eSPI к Edzelf Esp32 radio, тогда может и я смогу русифицировать радио.

Читайте внимательно.

> Пишите мне в telegram @ServerDoma. Может быть смогу помочь с внедрением библиотеки. Или найду исходники этого проекта.