Edzelf/ESP32-Radio #270: Strage buzzing noise

Zip-o-mat, 2019-07-10:
Hi,
I have an VS1053 with an SD slot and a ESP32 dev board. I have connected everything es described. When I play music I get a strage buzzing noise beside the music. I recorded the noise in the following video:
https://youtu.be/nyHSdeg06z0

Does anyone know where this noise is coming from? The system is powered from a Powerbank and the buzzing appears with an amplifiers and also when I use headphones.

Zip-o-mat, 2019-07-22:
I just found out that the buzzing noise only appears when the volume is low. When I increase the Volume up to 100 the buzzing is not audible any more.

ido1990, 2019-08-22:
I've the same issue.
It seems like a grounding issue but I didn't find how to fix it yet.
When I increase the volume to 100 it solves the issue on music stations but when there's silence or when people talk, it still in the background.