Edzelf/ESP32-Radio #271: mqtt - mp3track

erichegit, 2019-07-25:
From the code I have the feeling that it should be possible to call it following way

mp3track=localhost/folder/file.mp3

But I have a wired behoviour :(
Please take a look at my log
First I am playing an internet stream - then I try to switch to a mp3 file via mqtt command - BUT it plays not the song I want.
Hint my song is 1,1,0,0
it plays 2,1,0,0

`D: Command client available
D: handlehttp started
D: Get command is: preset=0
D: Send reply for preset=0
D: Command: preset with parameter 0
D: STOP requested
D: New preset/file requested (0/0) from 109.206.96.34:8100
D: Connect to new host 109.206.96.34:8100
D: Connect to 109.206.96.34 on port 8100, extension /
D: Publish to topic P0AA4AE30/preset : 0
D: Connected to server
D: nvs_get_str failed 1102 for key basicauth, keylen is 9, len is 150!
D: Contents: 
D: Switch to HEADER
D: Headerline: icy-name:NAXI LOVE RADIO (NAXI,Belgrade,Serbia, NAXI,Beograd,Srbija) - 128k
D: Headerline: icy-genre:Love and Romance
D: Headerline: icy-url:http://www.naxi.rs
D: Headerline: content-type:audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-pub:0
D: Headerline: icy-metaint:8192
D: Headerline: icy-bP0AA4D: Publish to topic P0AA4AE30/icy/name : NAXI LOVE RADIO (NAXI,Belgrade,Serbia, NAXI,Beograd,Srbija) - 128k

D: Switch to DATA, bitrate is 128, metaint is 8192
D: Metadata block 64 bytes
D: Streamtitle found, 62 bytes
D: StreamTitle=' Double - The Captain Of Her Heart';StreamUrl='';
D: Publish to topic P0AA4AE30/icy/streamtitle :  Double - The Captain Of Her Heart
D: Publish to topic P0AA4AE30/playing : 0
D: Publish to topic P0AA4AE30/playing : 1
D: MQTT message arrived [P0AA4AE30/command], lenght = 55, mp3track=localhost/2/15 - Mellow Trax - Outta Space.mp3
D: Command: mp3track with parameter localhost/2/15 - Mellow Trax - Outta Space.mp3
D: getSDfilename requested node ID is localhost/2/15 - Mellow Trax - Outta Space.mp3
D: getSDfilename Full MP3 = localhost/
D: Playing localhost/
D: STOP requested
D: Stopping client
D: connecttofile - host = ⸮⸮⸮?_
D: handle_ID3 - p = 
D: handle_ID3 - path = ⸮⸮⸮?
D: SD_currentnode is localhost/2/15 - Mellow Trax - Outta Space.mp3, curnod is localhost/2/15 - Mellow Trax - Outta Space.mp3, delta is 1
D: getSDfilename random choice
D: getSDfilename requested node ID is 2,1,0,0
D: Publish to topic P0AA4AE30/icy/streamtitle : 
D: getSDfilename Full MP3 = localhost/A/01.MP3
D: connecttofile - host = ⸮⸮⸮?_
D: handle_ID3 - p = A/01.MP3
D: handle_ID3 - path = ⸮⸮⸮?
D: Found ID3 info
D: ID3 TPOS = 3/3
D: ID3 TRCK = 1/16
D: ID3 TYER = 2013
D: Publish to topic P0AA4AE30/icy/streamtitle : A/01.MP3
D: SPI semaphore not taken within 120 ticks by CPU 0, id hspec
D: SPI semaphore not taken within 130 ticks by CPU 0, id hspec
D: SPI semaphore not taken within 140 ticks by CPU 1, id hspec
D: SPI semaphore not taken within 150 ticks by CPU 0, id hspec
D: SPI semaphore not taken within 160 ticks by CPU 0, id hspec
D: SPI semaphore not taken within 170 ticks by CPU 0, id hspec
D: SPI semaphore not taken within 180 ticks by CPU 0, id hspec
D: SPI semaphore not taken within 190 ticks by CPU 0, id hspec
D: SPI semaphore not taken within 200 ticks by CPU 0, id hspec
D: Publish to topic P0AA4AE30/preset : 0
D: Publish to topic P0AA4AE30/playing : 0
`

erichegit, 2019-07-25:
I already fixed it ;)
you only have in analyzeCmd one way --> mp3track
BUT when using the webinterface you send as value 1,2,0,0.
When you send via mqtt a mp3 filename - you do

> value = getSDfilename ( value ) ;               // like "localhost/........"

which is not working for a filepath.

So I changed the keyword for mqtt to get it work and skip the getSDfilename in this case.

I have to say that I enjoy your code and the project :+1: , but I don`t understand why you didn`t also forward the MP3 filepath from the combobox of the webinterface.

this 1,2,0,0 would be for me something I would use in background - maybe to ask which file to play next - but not for calling MP3 files :(

I want to extend the code to choose Folders to play and it took me hours to understand how you implemented this call.

Edzelf, 2019-07-25:
Sending filenames to the http client will probably take too much time or space.

erichegit, 2019-07-26:
so what is your suggestion?
Do you confirm this Bug - or are I am doing something wrong?

My tests with mqtt and long filenames are fine so far.